prusa3d/Prusa-Firmware #3969: Thermal Anomoly Message- MK3S, Mosquito Hotend, Mosquito High Temp Thermistor

Extra-Fox, 2023-01-28:
**Printer type** - MK3S
**Printer firmware version** -  3.12.1

**SD card or USB/Octoprint**
OctoPrint

**MMU**
N/A

**Describe the bug**
Installed the new 3.12.1 firmware and it went through the calibration routine. Afterward the printer was idling at about 20c, a beep was heard and "Thermal Anomaly" is now on the screen. The message appears to have cleared on it's own after some amount of time.

Printer is in an enclosure, but the internal temperatures are consistent with room temperature at the current time.
  
**To Reproduce**
1) Have printer sit idle after calibration.
2) Wait for message.
  
**Expected behavior**
I should not have expected to get a thermal anomaly from an idle printer. I definitely don't want to get this message during a print.
  
**G-code**
N/A

**Video**
N/A

wavexx, 2023-01-28:
Does the slice high temp thermistor require a different thermistor table or is it fully compatible?
Is this stock 3.12.1 or you're building a custom repository?

Is this still a regular 40W heater?

Extra-Fox, 2023-01-28:
> Does the slice high temp thermistor require a different thermistor table or is it fully compatible? Is this stock 3.12.1 or you're building a custom repository?
> 
> Is this still a regular 40W heater?

I do believe a different table is required and that's why I'm using the Bondtech branch of this firmware. However, it's my understanding that this and some adaptions for their BMG extruder are really the only differences between the two versions. (for reference: https://github.com/BondtechAB/Bondtech-Prusa-Firmware/branches/all).

I'm not sure of the wattage of the Mosquito, but it runs off the OEM powersupply without an issue.

wavexx, 2023-01-28:
The thermal model needs to know the power of the heater, and maybe some extra details. I see only 50W standard heaters on the slice engineering website, and a 30W industrial version.

Can you check which one you've got?

Extra-Fox, 2023-01-28:
Thankfully my old order was till up, looks like I've got the 50W version of the heater.

https://www.bondtech.se/product/mosquito-24v-50w-heater/

Extra-Fox, 2023-01-28:
So just in the last few minutes here, it's beeped and indicated Thermal Anomaly twice now. Then it goes away.

wavexx, 2023-01-28:
Do you have a way to record the serial output?

Run the following gcode:

```
M310 S0
M155 S1 C3
D70 S1
M310 P50
M310 A F0
M310 S1
M500
```

This disables the model, enables extended logging, sets the heater power to 50W then runs calibration again and re-enables it.
If you can, please attach the entire log (from beginning to end) here as a zip. We can use this to see if there is any adjustment to do for the mosquito.

I'm curious to know also if setting the power to 50W is already enough to make the warning to away.

Extra-Fox, 2023-01-29:
I am absolutely willing to do that, but would you be able to point me to an article or explain how to accomplish that? It's not something I've done before.

Extra-Fox, 2023-01-29:
So first off, apologies, because I wasn't really sure what to expect here. I think I let this run too long as it hadn't occurred to me that by enabling extended logging that it would continue to send messages. So I'll point out a few things in the log to hopefully help navigate it easier.

1. I turned this on then took a call, during that time it captures several anomalies at lines 18-30.
2. I started the command at line 31.
3. At line 4897 it says "Settings Stored" which I believe is likely the end of whatever routine it was running.
4. I finally realize what's going on and mistakenly hit the off button at line 17663.
5. And finally I hit disconnect to stop the auto-scroll.

Zip file attached.
[20230128_Extra-Fox_Mk3s_Log.zip](https://github.com/prusa3d/Prusa-Firmware/files/10529098/20230128_Extra-Fox_Mk3s_Log.zip)

wavexx, 2023-01-29:
Thanks! Looks like the log is complete. Give me a few days to have a look at it. Meanwhile, since it doesn't seem to be enough, turn it off by sending `M310 S0` and `M500` so that you can keep on printing.

While looking at this, I'm wondering if you ever did a PID calibration, or you're still using the stock values? I'm saying this because it looks like it's still oscillating quite a bit:

[image]

This shouldn't have any impact for this specific issue, but you'll benefit for more stable printing temperatures if you do that. Doing a PID calibration should be recommended for any non-stock hotends.

Extra-Fox, 2023-01-29:
There's a really go chance I've not done a PID calibration at all. So I'm running that now. I'll see if that make a difference and if not, send the Gcode as suggested.

Extra-Fox, 2023-01-29:
Here's the PID autotune results after the second round.

PID Autotune start
 bias: 88 d: 88 min: 236.25 max: 248.13
 bias: 82 d: 82 min: 236.18 max: 244.54
 bias: 79 d: 79 min: 236.46 max: 244.03
 Ku: 26.58 Tu: 26.46
 Classic PID
 Kp: 15.95
 Ki: 1.21
 Kd: 52.75
 bias: 79 d: 79 min: 236.43 max: 243.85
 Ku: 27.10 Tu: 26.46
 Classic PID
 Kp: 16.26
 Ki: 1.23
 Kd: 53.79
 bias: 78 d: 78 min: 236.64 max: 243.69
 Ku: 28.19 Tu: 25.92
 Classic PID
 Kp: 16.91
 Ki: 1.30
 Kd: 54.80
PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h

I'm going to run the Gcode routine again that you initially suggested. I'll attach the log when finished.

Extra-Fox, 2023-01-29:
Oh wow, I think I've done this all wrong. 

My initial results was likely completely wrong based on how I ran the Gcode. I tried to run it all on a single line which it sounds like since Gcode is interpreted line-by-line this was likely an error on my part (I'm a novice to this so I'm still learning). So for this result I ran it one line at a time. Waiting for it to finish between each step. So the new log is attached.

This is also after running the PID tune twice. I did notice how much more stable the temperature line was after I ran the PID tune. I remember running this the first time and thinking how much noise in the temperature reading there was and not it seems really steady.

[image]

[20230129_Extra-Fox_Mk3s_Log.txt](https://github.com/prusa3d/Prusa-Firmware/files/10530461/20230129_Extra-Fox_Mk3s_Log.txt)

3d-gussner, 2023-01-30:
@Extra-Fox A Mosquito can have a "standard" thermistor or a Slice Engineering PT1000 thermistor, question is which do yo have?

I can see that you flashed a Bondtech based firmware from your log file. 
```
echo: 3.12.1-6197
echo: Last Updated: Jan 25 2023 17:29:33 | Author: (none, default config)
Compiled: Jan 25 2023
```

Also in the latest log file we can see that `Proterface` doesn't understand `D70` and doesn't log the temperatures only the fans from `M155 S1 C3`

You have OctoPrint right?
Can you please follow the instructions for Octoprint https://help.prusa3d.com/article/crash-dump_364959 to set up the logging and provide the `serial.log` file. Of course ignore the `D21` part of this help article.

And re-run the gcodes using octoprint?

Extra-Fox, 2023-01-30:
Well then, I've learned something new about Octoprint as well. One observation I have made is that it seems like I'm only getting the Thermal Anomalies when the printer is idle and at room temperature. I've done a few prints at this point and at no point did this message appear. Anecdotal I'm sure, but still seemed worth mentioning.

I'm using the following Mosquito Thermistor, although it seems it's been end-of-life'd. 
- https://www.bondtech.se/product/mosquito-thermistor/
- https://www.sliceengineering.com/products/thermistor-high-temperature?variant=36948577452194

I've now re-run the series of commands and the serial output file is attached. I do see that it's now recording the temperature.

[20230130_Extra-Fox_Mk3s.zip](https://github.com/prusa3d/Prusa-Firmware/files/10537383/20230130_Extra-Fox_Mk3s.zip)

3d-gussner, 2023-01-30:
@Extra-Fox Thanks, well done.
Always nice that we learn something, isn't it?

```
2023-01-30 07:27:14,158 - Send: M155 S1 C3
2023-01-30 07:27:14,165 - Recv: ok
2023-01-30 07:27:15,166 - Recv: T:36.4 /0.0 B:23.0 /0.0 T0:36.4 /0.0 @:0 B@:0 P:0.0 A:23.5
2023-01-30 07:27:15,181 - Recv: E0:0 RPM PRN1:0 RPM E0@:0 PRN1@:0
...
2023-01-30 07:27:37,399 - Send: D70 S1
2023-01-30 07:27:37,414 - Recv: ok
2023-01-30 07:27:37,629 - Recv: TML 0 -47 0 42020000 41c64000
2023-01-30 07:27:37,899 - Recv: TML 0 0 0 420e0000 41bf8000
2023-01-30 07:27:38,169 - Recv: TML 0 1 0 420b8000 41bd8000
2023-01-30 07:27:38,333 - Recv: T:34.9 /0.0 B:22.3 /0.0 T0:34.9 /0.0 @:0 B@:0 P:0.0 A:23.7
2023-01-30 07:27:38,341 - Recv: E0:0 RPM PRN1:0 RPM E0@:0 PRN1@:0
...
2023-01-30 07:27:52,465 - Send: M310 P50
...
2023-01-30 07:28:09,981 - Send: M310 A F0
2023-01-30 07:28:10,012 - Recv: TM: calibration start
2023-01-30 07:28:10,015 - Recv: LCD status changed
2023-01-30 07:28:10,018 - Recv: TM: initial C est.
2023-01-30 07:28:10,032 - Recv: TML 0 0 0 42078000 41cb8aab
2023-01-30 07:28:10,299 - Recv: TML 0 2 0 420b0000 41c08000
2023-01-30 07:28:10,571 - Recv: TML 0 1 7f 42068000 41bf4000
...
2023-01-30 07:43:38,334 - Recv: echo:  M310 I0 R13.93
2023-01-30 07:43:38,338 - Recv: echo:  M310 I1 R13.58
2023-01-30 07:43:38,341 - Recv: echo:  M310 I2 R13.22
2023-01-30 07:43:38,345 - Recv: echo:  M310 I3 R12.51
2023-01-30 07:43:38,348 - Recv: echo:  M310 I4 R12.34
2023-01-30 07:43:38,356 - Recv: echo:  M310 I5 R12.17
2023-01-30 07:43:38,361 - Recv: echo:  M310 I6 R12.00
2023-01-30 07:43:38,373 - Recv: echo:  M310 I7 R11.82
2023-01-30 07:43:38,377 - Recv: echo:  M310 I8 R11.70
2023-01-30 07:43:38,381 - Recv: echo:  M310 I9 R11.58
2023-01-30 07:43:38,384 - Recv: echo:  M310 I10 R11.45
2023-01-30 07:43:38,387 - Recv: echo:  M310 I11 R11.33
2023-01-30 07:43:38,391 - Recv: echo:  M310 I12 R11.25
2023-01-30 07:43:38,394 - Recv: echo:  M310 I13 R11.17
2023-01-30 07:43:38,398 - Recv: echo:  M310 I14 R11.10
2023-01-30 07:43:38,401 - Recv: echo:  M310 I15 R11.02
2023-01-30 07:43:38,406 - Recv: echo:  M310 P50.00 C17.49 S0 B1 E1.74 W1.20 T-7.00
...
2023-01-30 07:49:11,997 - Send: M310 S1
2023-01-30 07:49:12,011 - Recv: ok
...
2023-01-30 07:49:27,556 - Send: M500
2023-01-30 07:49:27,748 - Recv: echo:Settings Stored
```
Looks good, please leave the OctoPrint serial logging on for the next few prints and if you have THERMAL ANOMALY on LCD please post them including the serial log here.
You can reset the printer and double check that the `M310` still reports these values and also with `M503` you can see your Magnum HT PID values as well.

Here your plot

[image]

3d-gussner, 2023-01-30:
Please check the terminal output and/or serial.log for `TM error:` messages at room temp.
You can try to tweak the Thermal model warning by in-/decreasing the value with `M310 W1.20` as default.
Please try slight steps off `0.05` or `0.10` to set get rid of the warnings.
Do you use a sock? If not it would be interesting to see how the Magnum HT behaves when you dump heat to a wrench during preheat. This is a good test to see that the `E` error value works well.

Extra-Fox, 2023-01-30:
Looks like after reset it's still listing the values, I'm reasonably confident at least.

I am using a silicone cover on the hot end if that's what you're referring to? And if I'm understanding correctly, you'd like me to remove the cover and attach a wrench to the nozzle to act as a heat sink of sorts? Would I run the same routine as before?

I will report back if/when I have an error.

Extra-Fox, 2023-01-30:
Well then, it's caught a couple and at least one occurred while preparing the file to send over.

Errors occur at lines:
[image]
 
[20230130_Extra-Fox_Mk3s_02.zip](https://github.com/prusa3d/Prusa-Firmware/files/10539678/20230130_Extra-Fox_Mk3s_02.zip)

3d-gussner, 2023-01-31:
> I am using a silicone cover on the hot end if that's what you're referring to? And if I'm understanding correctly, you'd like me to remove the cover and attach a wrench to the nozzle to act as a heat sink of sorts? Would I run the same routine as before?

Yes, if you have time to remove the silicone sock and use wrench as a heat sink it would be a "nice" test that also the Thermal Model on a Magnum HT behaves similar to an E3D v6.

3d-gussner, 2023-01-31:
@wavexx Do you have any idea why the TM would report TM errors at low temps and without heating up here?
Example:
Hotend temp is 20.9°C without a target temperature set and no PWM to heater.
```
2023-01-30 12:09:36,137 - Recv: T:20.9 /0.0 B:21.1 /0.0 T0:20.9 /0.0 @:0 B@:0 P:0.0 A:28.9
2023-01-30 12:09:36,867 - Recv: TM: error |-1.329722|>1.200000
2023-01-30 12:09:36,870 - Recv: LCD status changed
2023-01-30 12:09:37,138 - Recv: TM: error |-1.329722|>1.200000
2023-01-30 12:09:37,163 - Send: M117 H=-/-
2023-01-30 12:09:37,171 - Recv: ok
2023-01-30 12:09:38,141 - Recv: T:20.8 /0.0 B:21.1 /0.0 T0:20.8 /0.0 @:0 B@:0 P:0.0 A:28.8
```

3d-gussner, 2023-01-31:
@Extra-Fox Which OctoPrint plugin sends so many `M117 H=-/-` and `M117 0% L=-/-` LCD messages? Asking just out of  curiosity.

wavexx, 2023-01-31:
One potential issue is that the einsy enclosure is not getting _as_ hot when idling. With +/- 5C from both sensors and a -7 adjustment you might get some >10C difference from the real ambient temperature and we might be seeing extra "fake" heat from the hotend that doesn't exist.

I've never seen this happen on my MK3/s so far though, and I routinely leave them idling at 18-20C.

@Extra-Fox when leaving the printer idle, go to Support -> Temperatures. Can you post all readings?

3d-gussner, 2023-01-31:
@wavexx We have all temperatures with `M155` the hotend, bed, PINDA2(old) and  ambient.
`T:20.9 /0.0 B:21.1 /0.0 T0:20.9 /0.0 @:0 B@:0 P:0.0 A:28.9`
Hotend 20.9°C
Bed 21.1°C
Ambient 28.9°C

Extra-Fox, 2023-01-31:
> @Extra-Fox Which OctoPrint plugin sends so many `M117 H=-/-` and `M117 0% L=-/-` LCD messages? Asking just out of curiosity.

It's the "Display Layers Progress" plugin. https://github.com/OllisGit/OctoPrint-DisplayLayerProgress

And I'm thinking of turning off the screen display because it's kind of annoying. Thankfully I believe you can control that through the plugin options.

@wavexx 
Nozzle is fluctuation around 19-24C
Bed is 18C
Ambient is 20-21C
PINDA is 0C (!?)

Since you mentioned the Einsy heat I thought I'd attach this picture. It's of the Einsy in a different enclosure that opens outward. I installed this since the printer was going into an enclosure and I thought it would be good to have better access. In addition, my enclosure has a vent directly below the Einsy to draw up cooler air. There's also a fan in the vent that turns on when the case reaches 28C. I didn't mention this earlier because it didn't seem relevant at the time. 

[image]

Prusa-Support, 2023-03-02:
Not sure if related, but the recently released [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) version improves the Thermal Model compatibility over serial connection and with some third-party hardware.

Please give it a try.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-03-27:
Did the recent firmware upgrade help?
Is this issue still relevant?

Michele Moramarco
Prusa Research

Extra-Fox, 2023-03-29:
No, unfortunately the problem is definitely persistent.

n3xwrld, 2023-08-18:
same issue at 215 degree for pla, 240 for petg runs 40 hours without problems.

3d-gussner, 2023-11-14:
@n3xwrld have you tried to have PID tuning AND Thermal model values for 215°C and 240°C, which you can add to the PrusaSlicer Filament gcode or to the printer profile.

Try to log your prints and look for the TM errors reported.

Please also describe your printer, is it stock or modified if so which modifications do you have?

closed 2024-01-22 by github-actions[bot]