prusa3d/Prusa-Firmware #3970: [FEATURE REQUEST] Store multiple thermal profiles: one per nozzle diameter

dst74, 2023-01-29:
I searched and contacted the HD to be sure this feature is not already available.
I have just one printer and I swap nozzle regularly.
I have not yet uploaded the new firmware because of how the new thermal model protection behaves. 
It would appreciate the printer will store calibrations regardless to the nozzle diameter and auto-load the correct one before to print  by reading the nozzle diameter from the G-CODE and comparing it with the loaded one.
If it is the same --> No action. Otherwise load the required calibration.
If this automatic procedure is too resource-demanding at least give the user the ability to select the correct profile.
In this later scenario it will be appreciated to have a warning when starting a print if the nozzle calibration profile differs from the nozzle in the G-CODE: the printer pauses the printing process and asks the user if stop it or ignore it.
The calibration process should ask for what nozzle diameter the calibration is intended for (between the (presently four) standard supported)  and store it accordingly.

This will shorten significantly the nozzle swap process and will save a lot of energy and money, especially in these times where a kWh reached sky-high prices: here is at 0.38 € / kWh.

Thank you for you attention

Best regads

wavexx, 2023-01-30:
You shouldn't need to recalibrate the model when switching between nozzle sizes. Do you have already some experience that swapping a nozzle causes issues?

dst74, 2023-02-05:
> You shouldn't need to recalibrate the model when switching between nozzle sizes. Do you have already some experience that swapping a nozzle causes issues?

It is explicitly stated in the Prusa description and the HD confirmed it. 

https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0

Please re-run the temp model calibration to ensure the best performance and minimize the temp model warnings,
when you change:
        the nozzle
        heater block
        the heater cartridge
        the hotend thermistor
        the print fan
        fan shroud
        complete hotend assembly
    When you add a silicone sock
    When you move the printer to a different environment (enclosure, etc.).

So, how do you interpret it: Is it just and advise or, as I understand, is mandatory?

wavexx, 2023-02-05:
On a regular V6 heater, swapping nozzle _usually_ doesn't require recalibration. There is some margin for variations.

I say usually, because there are some thing that come into effect:

- When you swap the nozzle on a V6 heater you might change the heatblock height. If this changes too much it might become influenced by the fan less/more. This is the #1 reason why recalibration is recommended. When changing the nozzle on an assembled printer you can't usually rotate the heatblock, so the height doesn't change significantly in this situation. But if you reassemble the hotend this can change enough.

- You might change the material of the nozzle. If you calibrate on brass you might have a different response compared to a steel nozzle. My experience so far has been to switch between different nozzle sizes of the same material type (brass to brass, or nickel-coated copper to copper) on which  recalibration isn't necessary.

We have no data on the Revo to say anything precise on this yet. However the Revo nozzle is shielded by default, and the swap procedures makes it so the height doesn't change when swapping nozzles. So the fan influence, which is one of the biggest factors, shouldn't be a problem.

On the other side, the nozzle integration has more thermal mass. So it really depends on how much tolerance E3D is allowing on the nozzle assembly itself. If nozzle changing doesn't require PID calibration for optimal results, my feeling is that you won't require recalibration either.

At this stage, my recommendation would be: try, and report what happens. We have a beta build which has Revo support (see the last comments in PR #3957 - it doesn't work out of the box yet) and we're looking for feedback.

In the worst-case scenario the model data can also be saved/restored though the `M310` G-code. Running `M310` will output the entire data which can be used to re-set it later, so if the printer is attached to octoprint you can already save and restore this info across different nozzles. We can also tune the warning thresholds so that we account for the tolerances of the nozzles on the Revo. However we need more data to do this.

closed 2023-11-14 by github-actions[bot]