prusa3d/Prusa-Firmware #3972: [BUG] PINDA Calibration fails with HOTEND FAN ERROR

fornellas, 2023-01-29:
**Printer type** - MK2.5S
**Printer firmware version** -  3.12.1

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**

- PINDA calibration fails with hotend fan error.
  - Hotend fan works fine when printing thtough.
  - Self test passes.
- Bed is "forgotten" at 110C, until it is manually set to 0.
- Environment temperature is ~22C.
- PINDA Calibration worked fine with previous firmwares.
- When the failure happens, I noticed the hotend was passivelly heated up to 50C, and itts fan seems to start / stop in a loop every ~2s or so. I wonder if this is somehow triggering the hotend fan error.

**To Reproduce**

- Calibration > PINDA Cal.
  
**Expected behavior**
For calibration to conclude susccessfully.
  
**G-code**
N/A

**Video**

[image]

gudnimg, 2023-01-29:
> When the failure happens, I noticed the hotend was passivelly heated up to 50C, and itts fan seems to start / stop in a loop every ~2s or so. I wonder if this is somehow triggering the hotend fan error.

@wavexx I wonder if this issue is related https://github.com/prusa3d/Prusa-Firmware/issues/3513 🤔

gudnimg, 2023-01-29:
@fornellas does the error appear immediately? Or does it happen at a specific step in the calibration?

fornellas, 2023-01-29:
@gudnimg the error happend 2-3 times, IIRC, all at step 6/6, which yes, has the hotend at 50C and bed at 110C.

I'm trying again now. 1/6 to 5/6 went somewhat quick, but it tseems to be stuck at step 6/6 for ~2h at this point. It seem whatever the calibration algorithm is doing, isn't converging to a result. All I get from serial at this point is:

```
echo:busy: processing
T:50.01 E:0 B:110.0
T:50.06 E:0 B:110.0
echo:busy: processing
T:50.09 E:0 B:110.0
T:50.00 E:0 B:110.0
echo:busy: processing
T:50.02 E:0 B:110.0
T:50.12 E:0 B:110.0
echo:busy: processing
T:50.02 E:0 B:110.0
T:50.04 E:0 B:109.9
```

gudnimg, 2023-01-29:
I suspect this is where the firmware is getting stuck: https://github.com/prusa3d/Prusa-Firmware/blob/c9722c2b63b66e32c55f6fb477d952116ca21c62/Firmware/Marlin_main.cpp#L4987

gudnimg, 2023-01-29:
@fornellas There should be at the start a log which has this string: `"start temperature: "` it should include `start_temp` could you check what it shows?

fornellas, 2023-03-21:
@gudnimg sorry for the late reply, just updated to 3.12.2-5713 and tried again. It is one more time stuck on 6/6 for over 1h:

```
zcat prusa-PINDA-Calibration.log.gz | grep -Ev '^(echo:busy|T:)'
MMUstart
echo: 3.12.2-5713
echo: Last Updated: Feb 28 2023 08:57:48 | Author: (none, default config)
Compiled: Feb 28 2023
echo: Free Memory: 2858  PlannerBufferBytes: 1792
echo:Stored settings retrieved
adc_init
FSensor ENABLED
echo:SD init fail
echo:SD init fail
echo:enqueing "G76"
echo:Enqueing to the front: "G28 W0"
PINDA probe calibration start
start temperature: 35.0
LCD status changed
MMU not responding - DISABLED

ZERO: 0.172

Step: 1/6 (skipped)
PINDA temperature: 35 Z shift (mm):0

Step: 2/6

PINDA temperature: 40.0 Z shift (mm): 0.010
Step: 3/6

PINDA temperature: 45.2 Z shift (mm): 0.025
Step: 4/6

PINDA temperature: 50.0 Z shift (mm): 0.050
Step: 5/6

PINDA temperature: 55.0 Z shift (mm): 0.070
Step: 6/6
```

[prusa-PINDA-Calibration.log.gz](https://github.com/prusa3d/Prusa-Firmware/files/11033014/prusa-PINDA-Calibration.log.gz)

gudnimg, 2023-03-21:
Thank you for the logs! :) Looking at it, it seems to confirm my suspicion that the firmware is getting stuck here https://github.com/prusa3d/Prusa-Firmware/blob/c9722c2b63b66e32c55f6fb477d952116ca21c62/Firmware/Marlin_main.cpp#L4987

Unfortunately the firmware is not reporting the current PINDA temperature which seems very strange to me. The PINDA temperature reading is apparently not reaching 110°C But we don't know what the current temperature the sensor sees since its not in the logs.

gudnimg, 2023-03-21:
@fornellas is the PINDA maybe too far away from the bed? How does your PINDA position compare to the image below?

[image]

fornellas, 2023-03-21:
It was like this (about "1 zip tie" high):

[image]

I've pushed it down a bit:

[image]

and tried again and... it worked.

It seems there are 3 things to consider here:

- The fan error maaay have been related to the temperature being at the threshold of the fan starting / stopping, and at low PWM values, the fan wouldn't turn.
- The fact that there's no time limit to this calibration is a bit concerning: IMHO it should have one.
- If the time limit is reached, potentially adding a suggestion to check the distance between the PINDA probe and the bed would be helpful.

Thanks for the help @gudnimg !

I'll leave this ticket open in case someone decides to work on any of the items above (or discuss further), at least for my case, "case closed" :-D

3d-gussner, 2023-03-22:
Please close this issue as users can search also for closed one

closed 2023-03-23 by fornellas