prusa3d/Prusa-Firmware #3977: [BUG] Colon ":" interpreted as command separator

RoboMagus, 2023-01-30:
**Printer type** - MK3S
**Printer firmware version** -  3.13 DEVELOP

**SD card or USB/Octoprint**
SD Card

**Describe the bug**
Legacy bug ported over from ye olde Marlin.
Original bug report filed in their repo: [Marlin 1568](https://github.com/MarlinFirmware/Marlin/issues/1568).
And the subsequent [bug fix](https://github.com/MarlinFirmware/Marlin/pull/1570)

Currently, `:` (colon) is regarded as command separator, but was regarded a bug in Marlin back in 2015 and patched since.

Found when working on PR #3946.

**To Reproduce**
Print from SD card ad Serial does not reproduce.
File should contain e.g. `M118 A1 action:pause`. This will be split in 2 commands: `M118 A1 action` and `pause`.
  
**Expected behavior**
Similar behavior as is the case when printing over Serial, and how Octoprint interprets these commands. I.e. no command splitting on `:`.
  
**G-code**
`M118 A1 action:pause`

wavexx, 2023-01-30:
So looks like this was really intended to match ';'

RoboMagus, 2023-01-30:
@wavexx, not sure if that's the case because looking at the changes in MarlinFirmware/Marlin#1570 that was already taken into account in a separate statement.

leptun, 2023-01-30:
A very superficial google search yielded that the `:` are relics from the CNC era. It is meant to be used only as the first thing in the line and it also means that what follows it is a comment, but with the twist that the comment should be displayed to the user on screen as a log message. I feel like we could probably safely remove this exception and just not handle `:` at all. It's not like anyone would be crazy enough to run CNC gcode that uses this feature on their MK3 printers...

wavexx, 2023-01-30:
> It's not like anyone would be crazy enough to run CNC gcode

I'm sure @alexiri might have something to say about that ;), but if Marlin already fixed this we can surely follow

RoboMagus, 2023-01-30:
> I feel like we could probably safely remove this exception and just not handle : at all.

Had the same feeling 😄

commit c3cfc860 (2023-02-10): Fix #3977
closed 2023-02-10 by DRracer