prusa3d/Prusa-Firmware #3980: [BUG]Stuck with pinda not triggered error message

sarf2k4, 2023-02-01:
**MK3S**
**3.12.1**

**MMU2S**
**1.0.6-372**

**Describe the bug**
When homing, the printer able to detect if the pinda is triggered or not during the homing process which mainly the nozzle already pressing against the bed. The printer will automatically stop the procedure and displays 

```
pinda didn't trigger while homing procedure, is there debris on the nozzle? waiting....
```
On the lcd screen. Clicking/rotating the knob did nothing. I had to press reset on the lcd panel
  
**To Reproduce**
Don't let the pinda probe trigger during the z-homing related movement
  
**Expected behavior**
I am able to click the knob so I can rectify the issue why my pinda probe didn't trigger

Prusa-Support, 2023-03-09:
For clarity, the following picture shows what a similar error message looks like.

[image]

Please notice that the message is truncated, and the arrows mean that the message continues on the next screen.
I don't have pictures of the following screen in my archive but the remaining part says _<[...] Waiting for reset.>_ so there is no other way to resume than resetting the printer.

Can you please confirm this is what you have seen, and that the message would continue on the following screen, either after pressing the knob (only once) or, automatically, after a few seconds?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-03-10:
The problem of the message not being entirely displayed is confirmed and an issue about it already exists.

The fact that the printer requires a reset is the intended behavior.
The problem that the remaining part of the message is not showing up is a duplicate of issue https://github.com/prusa3d/Prusa-Firmware/issues/3402.

Please allow me to close this issue.

Michele Moramarco
Prusa Research

closed 2023-03-10 by Prusa-Support

sarf2k4, 2023-03-10:
Sorry for the long delay, I was thinking what best way to induce such error without nozzle crashing to my bed again.

I also forgot to mention that I do have octoprint hooked. Therefore octoprint sent `M112` immediately to the printer. Which I never thought this would be 'fatal error' on octoprint's side. Here is the video and terminal from octoprint

https://user-images.githubusercontent.com/9604217/224330263-3c0c436f-7c2c-4528-91bc-983266b553f5.mp4

```
G28 W
tmc2130_home_enter(axes_mask=0x01)
  0 step=62 mscnt= 994
tmc2130_goto_step 0 45 2 1000
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
tmc2130_home_enter(axes_mask=0x02)
  0 step= 2 mscnt=  32
tmc2130_goto_step 1 1 2 1000
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
KILL: 0
Error:Printer halted. kill() called!
M112
N2 M112*35
N3 M104 T0 S0*34
N4 M104 T1 S0*36
N5 M104 T2 S0*38
N6 M104 T3 S0*36
N7 M104 T4 S0*34
N8 M140 S0*109
```
I believe it is because the printer/firmware reports this as a fatal error, hence octoprint will issue `M112`. I think this area need some rework. 

An improvement I could think about this feature is, instead of sending `KILL` to octoprint/host, raise the z-axis first by 10-20mm, then send `warning` to octoprint/host to `pause` any ongoing buffered g-codes/commands like during a print job.

Prusa-Support, 2023-03-10:
Thanks for your feedback.
This is a known already existing issue indeed (https://github.com/prusa3d/Prusa-Firmware/issues/3402).

Michele Moramarco
Prusa Research