prusa3d/Prusa-Firmware #3981: [FEATURE REQUEST] RELAX Fan Error Print Pauses please!!

bolsoncerrado, 2023-02-02:
For MKxx Family of Prusa Printers:

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when...

...I sit away from the print room and I hear (SOMETIMES!) a single BEEP, just to find out a print has been paused, and nozzle burning the piece it stopped onto, and the FAN ERROR message appears on screen. Upon visual inspection both fans still spin so I guess it was a miriad of causes that could trigger the error (a filament "hair" stopped the fan for a moment, dust, electric glitch, start of bad wire thing...)
  
**Describe the solution you'd like**
SO

I thought it would be great if

a) the printer does NOT pause
b) the printer KEEPS BEEPING FOR ATTENTION (like when filament is gone) even if the print goes on...
c) upon pausing, move the nozzle up and away like the actual pause/change filament thing does.

This... OR:

a) make the firmware check the RPMs for a while just in case it autorecovers from the error BEFORE PAUSING, and if it has to pause,
b) upon pausing, move the nozzle up and away like the actual pause/change filament thing does
then
c) the printer KEEPS BEEPING FOR ATTENTION (like when filament is gone).

Pretty PLEASE!

wavexx, 2023-02-06:
If you have bad cooling in overhangs you can have a wrecked print if you don't pause so it's not clear cut what's best.
How realistic is that this a bad fan or connector issue? (try just to reseat the fan connector in the einsy).

bolsoncerrado, 2023-02-06:
I think the issue applies to both fans and while I appreciate the FAN monitoring system, all I ask is a bit of RELAX because if it's just a milisecond glitch on the connector, wire or print hair that "autosolves", machine can start to beep as loud as it can while PRINT CONTINUES.

Up to the customer to enable or disable this relax mode maybe?

In the end 3D prints should never be completely unattended... I just hate that micropauses can cause damage to the ongoing print....

wavexx, 2023-02-06:
The fan check measures the fan speed for a few milliseconds, so it can't be just a single micro-interruption.

bolsoncerrado, 2023-02-06:
microinterruption = few milliseconds = less than a second is already avoidable.

wavexx, 2023-02-06:
The the micro a millisecond is like an eternity. If there's a contact fault for that long it's an issue to look into IMHO, not to ignore.

bolsoncerrado, 2023-02-06:
Well I keep printing without any issues with the fan check off, but as I said that does not mean a more severe fault should be beeping...

The whole point is we're humans and miliseconds are too fast for us. LOL

wavexx, 2023-02-06:
No, what I'm saying is that true microinterruptions are already handled (or let's put it this way: they should already be).

If a disconnection is happening for longer, the fan will not keep a stable reliable speed so it should be looked into, not ignored, even if the fan still seems to work.

This is why I was asking if you try to reseat the connectors, or even better if you tried to swap the fan. It would be useful to know if there's a bug in there, or it's a contact issue.

bolsoncerrado, 2023-02-10:
They may still be a few seconds, the time I take to hear the beep and realize that even if the LCD reads fan error,  both fans are spinning fine and at good RPM :??

Once again, if this is a wire or connector starting to fail, some more seconds of relax before pause would be great; while printing continues keep beeping, etc.

bolsoncerrado, 2023-03-09:
TBH I dont know what the restore point is, I just know that when i hear the BEEP in the printing room my ass jumps the chair as quickly as I can just to find out the FAN ERROR but both fans operating fine :?

All in all, this OP was to "relax" the miliseconds, seconds or time in general the printer pauses on fan pseudo-failure, just that.

Can't tell for sure how many seconds does it take for the fan to restore power but the issue "is there".

Prusa-Support, 2023-03-28:
I think we really need to understand the real cause of the fan error.
We can't exclude any of the most common causes [listed above](https://github.com/prusa3d/Prusa-Firmware/issues/3981#issuecomment-1460728814).
Would a careful cleaning of the fan help?
Would a fan replacement solve the problem once and for all?

We shouldn't really modify a safety feature to make it work with a "reasonably working fan despite a few problems".
The safety feature is likely triggered because the fan was not working reliably.

We need more feedback or specific instructions to intentionally reproduce the problem and confirm a bug.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-19:
Not sure if our developers will be keen to lower protection on a safety feature, especially since there is a concern about a possibly not correctly working fan. I hope that a replacement or one of the suggestions above was helpful.
Is this issue still relevant?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-09:
This issue may not be relevant anymore or simply obsolete.
Lowering the protection barriers could be risky and doesn't sound legit in this case.
This issue will therefore be closed, however, feedback would surely be appreciated.

Michele Moramarco
Prusa Research

closed 2023-05-09 by Prusa-Support