prusa3d/Prusa-Firmware #3983: Pause (M601) firmware does not clear out old move data

kweb-, 2023-02-04:
**Printer type** - MK3S
**Printer firmware version** -  3.12.1-5686
**Program source** - SD card.
**Material** - PETG

**Describe the bug**
Custom gcode after pause is pre-emtped by old move before custom gcode runs.  There is a <cmdqueue_pop_front()> command issued in firmware to avoid having a loop on the M601 command but it seems to only back the state up to the previous move data state.
  
**To Reproduce**
The following annotated gcode program fragment outlines a very repeatable issue where a previously queued move is run before custom gcode inserted after the M601 command is.  This is very reproducible and can be observed in any program by adding custom gcode command(s) after the M601 command line as per the example below and witnessing behavior.  I inserted the M601 command in prusaslicer.
```
;PAUSE_PRINT
M117 Insert Test Code After M601
M601
;Next line happens after Resume Print is selected on LCD
;A move back to the part by all axes.  Only after this moves completes does the next line get processed.
;Custom_GCode
G1 X80.5 Z65.0 E3 F720
...
```
  
**Expected behavior**
The custom gcode should run after the resume command is issued via the LCD.   The pre-queued move that occurs is both unexpected and disconcerting because it comes down very low to the bed/part and it contributes to quality issues because it more often than not deposits a blob of material on the part at the end of the move before the moves specified by the custom gcode are run.
  
**G-code**
see above

gudnimg, 2023-02-04:
> The pre-queued move that occurs is both unexpected

When the print is paused the firmware will automatically park the extruder to the side regardless of what the gcode file says. It's normal for the extruder to move to the last position known position when you send resume command.

Do I understand correctly this is the move you are seeing before the custom gcode?

kweb-, 2023-02-05:
Yes, the move to the last known position after the resume runs before any custom gcode inserted after the M601 line does.  In the example gcode I provided, the line "G1 X80.5 Z65.0 E3 F720" only executes after the machine moves to the previous position.  

Is there a way in gcode to clear the old move date?

gudnimg, 2023-02-05:
> Is there a way in gcode to clear the old move date?

I'm not aware of any gcode which does this. The pause/resume commands are using the "same" code as the firmware uses to recover from power failure during a print. 

If the printer would run `G1 X80.5 Z65.0 E3 F720` then the last known position would be invalidated and currently the printer doesn't have any way to know whether or not a custom gcode should be run while paused. Would a workaround work: run the gcode just before M601?

Why would you want to run this gcode during pause anyway? I'm curious about the use case :)

kweb-, 2023-02-05:
Thank you for your interest in the use case driving my submitted report.  
I print almost exclusively with petg.  I use pause for part insertion (nuts, magnets, stuff) and some of these activities can take a while if for example, things need to be glued in place or I'm placing pre-made support structure for overhanging features.  Even if the resume is requested in less than 60 seconds, preloaded material in the nozzle has been oozing out during cool-down and the subsequent re-heat phases.  There's always some initial portion of material deposition that is missing or partially extruded when the part program resumes as a result.  Sometimes, the slicer makes paths such that I can use the slicer seam painter to have these defects occur somewhere inconsequential or the first move of the next layer is support material.  However, sometimes the first path is on a perimeter that is in a location where the defect is noticeable and in some cases, such as wells for fastener nuts, creates a weakness where there will be forces trying to separate the layers and that's undesirable whether the defect is visible or not.
The missing material may be inconsequential functionally but for certain parts, my clients perceive it as an indication that the part is poorly made. 

By running custom gcodes after resume, I can move the hotend to a position high enough to get at it with my brush, get everything back to temperature,  advance the extruder to completely reload the nozzle, retract the extruder a little bit to stop oozing briefly, issue a short pause for quickly wiping the nozzle clean, approach the first position in the new layer, and continue the program.   It's a tower wipe with a brush.   So far, this seems to have solved my quality issue with the exception of the move forced on this process by the current implementation of resume.  Per my initial message, this forced move usually drops a blob on the part.  This process works whether it's a 60 second pause or a 10 minute pause 

The big advantage of using Prusa Slicer to insert the pause is that you get the countdown timer to the next pause displayed on the LCD when the program is running.  This is an amazing feature!  This is why I'm asking you about how to make this use case work with the existing slicer pause feature.  Keeping the bed at temperature during the pause is very important so that can't be broken and not sure if flushing queue would cause a problem there.

Could M601 accept a parameter that would ask for the queue to be purged?  
What would happen if the case statement for M601 was modified to loop on cmdqueue_pop_front() until it returned false?

kweb-, 2023-02-05:
BTW: it's starting to become clear that the post resume move is intentional 
and a considered "feature" in certain circumstances.  I guess that it's kind of a semantics game if this is a "bug" or not but in the use case presented, it is at best a very undesirable "feature".

Prusa-Support, 2023-03-09:
Oozing can't be avoided during a pause. The melted plastic will naturally drop due to gravity.
Please remove the oozed material before resuming or it may leave blobs and debris on the print object.

If oozing is significant though, the same amount of oozed plastic would be missing in the nozzle. A tiny empty gap could be noticed in the following extrusion movement as a consequence.

This normally wouldn't be noticeable as long as there are multiple perimeters, because internal perimeters are usually printed first. In case of single perimeters prints (not advisable), you can find an option in Prusa Slicer to print "_infill before perimeters_".

The movements that you see after a pause are a set of movements preset in the firmware.
Unfortunately, Prusa Slicer doesn't support "before pause" and "after pause" custom g-codes, nor an option to override the normal pause operations which are preset in the firmware. 
All of this may not be necessary but could be discussed on the repository relevant to Prusa Slicer.

I hope this clarifies.
Can this issue be closed?

Michele Moramarco
Prusa Research

kweb-, 2023-03-29:
Sorry that it has taken so long to respond.  

You acknowledge all of the issues that I enumerated with the exception of down playing defects that happen when the pause is a few minutes long which have been reproduced here many times; certainly more reproducible than perfect results for the use cases we encounter here with petg materials.  If no changes can be made to the canned pause code to purge the queued approach move(s) then there's no point keeping this open.

I appreciate the interaction.  
Best regards,

closed 2023-03-29 by kweb-