prusa3d/Prusa-Firmware #3986: [BUG] Altfan detection doesn't work as expected

DominikPalo, 2023-02-05:
**Printer type** - MK3S+
**Printer firmware version** -  3.12

**Describe the bug**
 I bought my Prusa MK3S+ last year, so I got the printer with [the Delta hotend fan](https://www.prusa3d.com/product/hotend-fan/) installed. As I understand, the Delta fan should be treated as an alternative fan, therefore the reduced power (50% PWM) should be applied. I'd like to replace my Delta fan with [a Noctua one](https://noctua.at/en/products/fan/nf-a4x10-5v) (because my printer with the Delta one is much noisier compared to older MK3S with Noctua fans and I want to silent it). Due to this change/upgrade, I want to disable the "Altfan detection" functionality to run the newly installed Noctua fan at the full speed and ignore the stored setting for the original Delta fan from the initial printer setup. According to the documentation, this could be done either by:

1. Doing a factory reset after installing the new fan, so the firmware can measure the speed again and properly detect the Noctua fan
2. Disabling the ["Altfan detection"](https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213#altfan-detection) in the "Experimental menu" (the documentation says: "By turning the Altfan detection off, the printer will run the fan at 100% power")

I don't want to do a factory reset, so I decided to just disable the fan detection. To my surprise, when I did it, the hotend fan still runs at the exactly same speed as before. So, it seems my Delta fan wasn't properly detected during the initial setup and it's been running for the whole time incorrectly at the full speed (`EXTRUDER_ALTFAN_SPEED_SILENT = 0`) OR that "Altfan detection" = OFF setting is ignored and the fan always runs at 50% (`EXTRUDER_ALTFAN_SPEED_SILENT = 1`).

  
**To Reproduce**
 1. Make sure your printer uses a factory Delta fan
 2. Enable the experimental menu and try to disable the "Altfan detection"
 3. Observe the fan is running at the same speed as with the "Altfan detection" enabled
  
**Expected behavior**
Disabling the "Altfan detection" on the MK3S+ printer with a Delta fan ("alternative fan") should increase the speed of the fan.

gudnimg, 2023-02-05:
> Disabling the ["Altfan detection"](https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213#altfan-detection) in the "Experimental menu" (the documentation says: "By turning the Altfan detection off, the printer will run the fan at 100% power")

Hi @DominikPalo 👋 I spotted issue with option 2 in 3.12 and fixed it for 3.13 here https://github.com/prusa3d/Prusa-Firmware/pull/3939 At this time we're not planning on including this in 3.12.2 due to risks regarding the temperature model.

I can send you a modified 3.12 firmware with the fix if you would like to try it and see if the fix works OK on your end? For me personally the fix resolves the issue with the LCD menu not working.

For reference my main printer is MK3S+ too.

DominikPalo, 2023-02-05:
@gudnimg thank you for your quick response :) If the issue is present only in the FW version 3.12, I'll try to downgrade it back to version 3.11 and check if it helps (because I use the Revo hotend, I had to disable the new "Thermal model" functionality, so the FW 3.12 doesn't bring any real benefits to me and I can stay on the 3.11 for now)

DominikPalo, 2023-02-05:
UPDATE: Downgraded to v3.11 and everything works as expected - when the "Altfan detection" is enabled, the fan runs at 50% and when it is disabled, the fan runs at the full speed 👍

DominikPalo, 2023-02-05:
BTW, it seems that not only the "Altfan detection" menu item is broken in 3.12, but as reported here https://github.com/prusa3d/Prusa-Firmware/issues/3933, after updating to 3.12 (I did it a few weeks ago), my hotend fan went to the full speed even without messing with that hidden menu (I didn't use that menu before)... That's why observed high noise levels and decided to replace the fan :/ and now I know the issue was caused by the FW update. So the issue may affect many users with stock Delta fans, which is much much worse.

gudnimg, 2023-02-05:
> my hotend fan went to the full speed even without messing with that hidden menu

I haven't seen this on my end. Do you need to run a print to trigger this?

DominikPalo, 2023-02-05:
> I haven't seen this on my end. Do you need to run a print to trigger this?
No, just preheat the hotend

closed 2024-02-07 by 3d-gussner