prusa3d/Prusa-Firmware #3987: M112 Alarm Sounds Constantly Until Dismissed

Extra-Fox, 2023-02-05:
**Printer type** - MK3S
**Printer firmware version** - 3.12.1 (using Bondtech branch for HT Thermistor and Mosquito hot end)

**Modifications**
1. Slice Engineering Mosquito Hot End.
2. Slice Engineering High Temp Thermistor.
3. BondTech BMG Extruder
4. BondTech SLS Print Head Housing and X-carriage.
5. Printer is installed in an enclosure.

**SD card or USB/Octoprint**
OctoPrint

**MMU**
N/A

**Describe the bug**
A quick note before I start here, I have fire suppression installed inside my enclosure which is why I'm comfortable doing an overnight print.

The printer had been sitting idle after a print. Early in the morning around 02:57 an M112 warning was triggered and the alarm sounded CONSTANTLY. Thinking it was a noise from a squeaky fan bearing this alarm was ignored for quite some time, but around 06:30 I was informed this was my printer and went down to check. Sure enough it was playing the song of its people loudly.

M112 errors didn't previously do this though. Typically there was a beep and that was it. So I'm not sure if this is something new or if there was a glitch. 

The attached log was clipped to just the end of the print to when I cancelled the alarm and then subsequently shut the printer off. As an added bonus you'll see several thermal anomaly events as it's cooling down and eventually at idle temperatures.

**To Reproduce**
1. Have printer sit idle.
2. Wait for M112 Error.
3. Alarm will sound.
4. Alarm will not stop.

**Expected behavior**
I did not expect the alarm to continuously sound.

**G-code**
N/A

**Video**
N/A

[serial.zip](https://github.com/prusa3d/Prusa-Firmware/files/10611301/serial.zip)

wavexx, 2023-02-06:
Does 3.12.0 do this, or do you know the last version that worked as intended?

Extra-Fox, 2023-02-06:
I've seen the M112 error many times before, usually after the machine has sat idle overnight, but never has it sounded a constant alarm. So this is new since the 3.12 firmware.

Extra-Fox, 2023-02-12:
Quick update.

This constant alarm is very consistent. I get it most often when the printer is sat idle and the room is a little colder than usual. It's massively annoying since being under temp while idle is hardly an emergency.

Extra-Fox, 2023-02-20:
I disabled the thermal model to get rid of the intermittent Thermal Anomaly messages, but this did not eliminate these M112 errors. The machine was sat completely idle for some time at around room temperature (20C) and the alarm went off.

Extra-Fox, 2023-02-27:
I have downgraded to the 3.11 firmware. 

This alarm effectively meant I could not leave my printer on overnight unless I wanted to be woken up by the loud, continuous beep.

Prusa-Support, 2023-03-10:
Thanks for the report and the updates.

As you may know, the code `M112` is triggered as a generic error.
I wonder if the constant beep only happens on certain specific error occurrences.

Also, I wonder if the same constant beep would occur if you print without Octoprint for one night.

1. Were you able to identify a specific error message in Octoprint's log?
2. Could you please try again printing on the standalone printer? Of course, I mean without Octoprint and third-party devices connected.
3. Do you ever happen to leave the printer idler for many hours after a print, during the day? Maybe you will be able to reproduce the issue during the day, catch the moment when the loud beep starts, and observe if the printer's behavior revels a specific occurrence before the constant beep.
4. Assuming it was a _thermal anomaly_ error, I wonder how the problem persisted after disabling the _thermal model_. Did you specifically send codes `M310 S0` to disable the Thermal Model, followed by `M500` to save the change?

Michele Moramarco
Prusa Research

Extra-Fox, 2023-03-10:
If I recall correctly, when I looked in the log it simply said that the error had been triggered. There was nothing before or after it that looked unusual or suggested a possible reason the error was triggered.

On the 3.12.x firmware the alarm sounded every time I got the `M112` error and persisted even after disabling the _thermal model_. I did send the two GCode statements which had the effect of reducing the number of Thermal Anomaly messages, but did nothing for the `M112` message or alarm. 

On the 3.11.x firmware, I still get the `M112` error, but there's **no** alarm associated with it.

For _both_ versions of the firmware, the error presents itself under the same conditions;
1. The Printer is completely idle and at room temperature.
2. Temperatures in the room are below 20C/68F.

Some things of note;
1. The error will occur every time these conditions occur so they're completely reproducible.
2. I've never experienced the error during a print.

danchar4, 2023-03-19:
### Constant beep when printer idle under 15°C with 3.12.2 firmware

I am having this constant beeping issue as well.
Mk3S with factory hotend printing from SDCard.

The printer is in an unheated garage and temperatures drop below 10°C overnight. If the printer is idle when the temperature drops, then a constant beep is heard with the 3.12.2 firmware. The only way to resolve it is to use a heat gun to warm up the hotend to 15°C and reset the printer.

With 3.11.x firmware, there is no beep, but there is a mintemp error as expected.

This beeping is a major regression for cold climate folks - I would like to disable this beep without **fully** disabling audible warnings.
 

I also have a Mk2.5S - it does not have this beeping issue with 3.12.2 firmware when cold - but its thermistors are not accurate and never register anything below 20°C.

Prusa-Support, 2023-03-28:
So far the concerning aspect seems to be a **cold ambient temperature**.

These thermistors are accurate enough for their purpose.
They are designed to work in a large range of temperatures, especially at high temperatures. It goes without saying that they may not be very precise at low temperatures.
I assume the thermistor resistance in a cold ambient is so high that the printer will virtually read infinite resistance (= no current = it simulates a disconnected thermistor).

Please be observant of the working conditions specified in the [Handbook](https://cdn.prusa3d.com/downloads/manual/prusa3d_manual_mk3s_en.pdf#_ga=2.253921664.94815203.1679920444-1262546293.1669820483).
>Working temperature range: 18 °C (PLA)-38 °C, indoor use only

The [MINTEMP](https://help.prusa3d.com/article/mintemp-error-and-mintemp-bed_2169) error should theoretically be triggered in a temperature below 10°C (in case of MK3-series) but the printer is not supposed to operate below 18°C.
Because the temperature reading can't be very precise at low temperatures though, I guess it is not unlikely that a MINTEMP error could be triggered even if the real ambient temperature is above 10°C.

This said, when the temperature is well below 18°C, it is recommended to heat up the ambient and the printer before use.

Assuming it is the case of a MINTEMP error, I don't know what specifically changed about MINTEMP error alerts in the new firmware version.
I will try to find this information.

Michele Moramarco
Prusa Research