prusa3d/Prusa-Firmware #3989: [BUG] 3.12.1 MK3S Firmware Crash on MMU2 Recovery

ltlowe, 2023-02-05:
**Printer type** - MK3S
**Printer firmware version** -  3.12.1-5686

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6-372

**SD card or USB/Octoprint**
  SD Print with Octoprint monitoring via USB

**Describe the bug**
Since upgrading to 3.12.1 from 3.10, (I did not use 3.11), each time I have had an MMU error the printer has had a firmware crash immediately following MMU recovery and the print aborted.  Both times this was several hours into an 11 hour or longer print, but I do not have timing details.    This has happened 3 times and I am reverting to 3.10 for a final attempt at the print.  

The other change is that the gcode was sliced with prusaslicer 2.6.0 alpha2.

I have captured a D21 dump however.
  
**To Reproduce**
Unknown, but I have not had a successful MMU recovery with the new firmware.  Unfortunately I am unable to help reporduce immediately as I need to try to get a successful print this weekend. Once done, I will reflash 3.12.1 and try to repro with minimal steps if the crash dump does not show something obvious. 
  
**Expected behavior**
 Once MMU recovery is complete, the print should continue
  
**G-code**
[helicopter_4color_split_edit-2+sprus-treesupport.2_0.6n_0.3mm_PLA_MK3SMMU2S_11h37m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/10611353/helicopter_4color_split_edit-2%2Bsprus-treesupport.2_0.6n_0.3mm_PLA_MK3SMMU2S_11h37m.gcode.zip)

I was unable to issue D21 with pronterface, but have attached a log of printer booting and D21 from Octoprint.

[3.12.1_crashdump.txt](https://github.com/prusa3d/Prusa-Firmware/files/10611313/3.12.1_crashdump.txt)

atom-b, 2023-02-06:
I'm experiencing the same issue with a nearly identical setup.

Printer type - MK3S+
Printer firmware version - 3.12.1-5686

MMU upgrade - MMU2S
MMU upgrade firmware version - 1.0.6-372

SD card or USB/Octoprint
Octoprint on a Pi Zero 2W via the GPIO connection

The very first swap on the first layer failed due to a mechanical issue with the filament spool, requiring intervention from me. Upon resuming the print the firmware crashed.

1. After identifying the failure, the printer left the toolhead parked over the wipe tower instead of raising it in the Z and moving to the the far left of the bed as it had done on previous firmware versions. The hotend cooled off as expected.
2. Roughly 30 minutes passed before I was able to intervene. 
3. I had to press the encoder knob to re-heat the hotend as usual, although unlike on previous firmware versions there was no message saying to do so. There was only the normal info screen with an "mmu load error" message at the bottom. I don't recall the exact text of the message.
4. Upon reaching the target temperature the error message screen that I'm used to (the one saying you must reheat the hotend to proceed) *did* appear, and the hotend began to cool itself again.
5. I pressed the encoder knob a second time and the hotend began to re-heat again, this time displaying the temperature and message that I'm used to in this scenario.
6. I fixed the issue with the filament and pressed the middle button on the MMU to load the filament into the MMU. This completed successfully and the red and green LEDs flashed as expected.
7. I pressed the right button on the MMU to resume the print.
8. The MMU began loading the filament to the toolhead. 
9. Roughly a second later, with the filament tip 4-5cm into the bowden tube, the firmware crashed.

[mmu2s-crash-dump-2023.02.06.txt](https://github.com/prusa3d/Prusa-Firmware/files/10661419/mmu2s-crash-dump-2023.02.06.txt)

nsheridan, 2023-02-08:
**Printer type** - MK3S
**Printer firmware version** - 3.12.1-5686

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6-372

SD card or USB/Octoprint
SD Print with Octoprint monitoring via USB

I had a similar experience, but it was at the start of a print when loading the filament after performing the mesh routine and the printer crashed during recovery.

[crash_dump_20230204.log](https://github.com/prusa3d/Prusa-Firmware/files/10684098/crash_dump_20230204.log)

ltlowe, 2023-02-11:
@gudnimg commented in #3990 that "Pausing MMU prints while the MMU is performing the filament change may cause FW crashes." is known and documented in the release notes.

I hadn't thought about that as a possible cause for my crash since the printer paused itself, but could it be the same ?

gudnimg, 2023-02-12:
@ltlowe I think its very likely the problem are the same. I'll do my best to try learn more about the problem today but my time is limited usually to the weekends.

Part of the problem is a watchdog timeout but there's seems to be more to it than that. Unfortunately this is very hard to debug since I can only rely on serial messages to debug so this may take some time :( This issue is not present at all in 3.13 with the new (unreleased MMU FW) so I hope it will be released sooner than later.

nsheridan, 2023-02-19:
Last night I compiled and flashed the tagged 3.12.2 source (3.12.2-5697), and today I had a load failure and the printer crashed in the same way after recovery.

[fw_3.12.2_crash.log](https://github.com/prusa3d/Prusa-Firmware/files/10776910/fw_3.12.2_crash.log)

nsolling, 2023-03-07:
This is still an issue on 3.12.2

P1xellat3d, 2023-03-28:
Downgraded to 3.12.0 and the issue still persists. 

Rolled back to 3.11.0 and the issue appears to be gone.

mwharton3, 2023-04-08:
I was having the same issues on 3.12.1 and a downgrade to 3.10.1 fixed the problem

gysnst2, 2023-04-09:
This a actually makes my 3 printer unusable right now prusa please fix this asap

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

There are no known crashes in FW 3.13.0 so a new Github issue should be created if that happens for some reason

closed 2023-07-31 by gudnimg