prusa3d/Prusa-Firmware #3997: [QUESTION] Cannot compile latest firmware

Moyrn, 2023-02-08:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**What is your question?**

Moyrn, 2023-02-08:
Using latest Arduino IDE (but was the same on 1.9), Einsy Rambo for MK3S, compilation brings an error on temperature.cpp

S:\__Firmware\Prusa-Firmware-3.12.1\Firmware\temperature.cpp: In function 'constexpr float temp_model::iir_mul(float, float, float, float)':
S:\__Firmware\Prusa-Firmware-3.12.1\Firmware\temperature.cpp:2339:1: error: body of constexpr function 'constexpr float temp_model::iir_mul(float, float, float, float)' not a return-statement
 }
 ^

exit status 1

Compilation error: body of constexpr function 'constexpr float temp_model::iir_mul(float, float, float, float)' not a return-statement
______________________________
I'm not a C++ coder, so please excuse my ignorance.
I found a fix like this:

static constexpr float iir_mul(const float a, const float b, const float f, const float nanv)
{
    //const float a_ = !isnan(a) ? a : nanv;
    //return (a_ * (1.f - f)) + (b * f);
    return ((!isnan(a) ? a : nanv) * (1.f - f)) + (b * f);
}

-- Compiles fine
Le croquis utilise 231130 octets (91%) de l'espace de stockage de programmes. Le maximum est de 253952 octets.
Les variables globales utilisent 5588 octets de mémoire dynamique.

thanks for your support.

leptun, 2023-02-08:
Please update the board definition in arduino ide. You are using an older version of the board definition which has different compile options, so it fails to build.

Moyrn, 2023-02-08:
Ok. I'll try to find how to do this  -- thanks

Moyrn, 2023-02-08:
Note for some others who may need it
- open arduino ide
- preferences, add board url window, paste https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json
- now go to board manager, type "prusa", unsintall 1.0.1 einsy rambo cards, install latest (1.0.6) from the proposals
- now select this board
- compile
- it works...

Thanks...

Note: i had tough time to find this URL  (even add to go to arduino unofficial page https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls saw that it was pointing to the raw.github... json pages, then saw one on github for einsy and reconstructed the URL)

There was most possibly simpler, but did not stumble on it.

closed 2023-02-08 by Moyrn