prusa3d/Prusa-Firmware #4000: [BUG] 3.12.1 MK3S Firmware Crash at specific ambient temperatures

v1p3r0u5, 2023-02-09:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.1

**MMU upgrade** - no
**MMU upgrade firmware version** - no

**Enclosure** - Prusa Original Enclosure
**Enclosure Add-Ons** - White LED-stripe, PSU Quick Release, Advanced Filter System

**SD card or USB/Octoprint**
 Print via SD-Card

**Describe the bug**
  When printing inside the enclosure with doors closed, the printer crashes suddenly at a specific ambient temperature (the one showed by the enclosure in the top left corner). It's a different temperature based on the filament. For Prusament PLA it crashes around 30.1° Celsius and for Prusament PETG the crash happens around 33.4° Celsisu. In 4 out of 5 times it crashes with absolutely no beep or message. In 1 out of 5 times there is a loud beep followed by the firmware crash error (see pictures in the attached ZIP-file).

  It doesn't matter if I use the original PSU cables or the PSU quick release. I guess there is a problem with Firmware 3.12.x because after a downgrade to 3.11.0 I'm not having those kind of problems at all. Also the fact that it happens at different ambient temperatures based on the filament supports me in this guess. Additionally I can say that I'm not having problems when I pre-heat the printer and wait until the ambient temperature goes higher than 35° Celsius. It just occurs during the print.
  
**To Reproduce**
  Just put the MK3S+ inside the Prusa Original Enclosure and use one of the g-code files for PLA or PETG from the attached ZIP-file and start printing with closed enclosure doors. You can force the crash a little bit when you first heat up the ambient temperature a bit. Anyway after around 20 to 30 minutes it will crash. At least in my environment. Ensure you are testing with 3.12.1 firmware to reproduce the crash and double check with firmware 3.11.0 if everything is ok.
  
**Expected behavior**
  Printer should not crash / reset without a message what happens.
  
**G-code**
  G-Code files for PLA and PETG are in the attached ZIP file

**Video**
  It's too big to attached => find it at we-transfer for about 5 days: https://we.tl/t-UoZesPpzFo 
  If you miss it to download, please. contact me and I will gladly provide you the video again.

**Memory-Dumps**
  Added 3 memory dumps as well to the attached ZIP file
  
**Pictures**
  Pictures of the error that happens sometimes are inside the attached ZIP file

** Attachment**
[bug_report.zip](https://github.com/prusa3d/Prusa-Firmware/files/10698218/bug_report.zip)

Prusa-Support, 2023-03-11:
Thanks for reporting.

This temperature is arguably easily reachable in an enclosure and doesn't seem to be problematic.
Interestingly enough this is _not_ a known issue, despite so many Prusa Enclosures being out there.
I suspect this is not a bug but a failure related to a specific component on your printer.

I can't seem to see anything specifically concerning in your G-codes but I think that you are using an obsolete configuration in Prusa Slicer.
(1) Would you mind "checking for updates" in the _configuration_ top bar menu of Prusa Slicer, and slicing and testing again a freshly sliced print file, with up-to-date configuration?

(2) Can you confirm that if the doors are opened, _no_ FW crash will occur?

(3) Would the printer trigger a FW crash if you "actively" heat up the enclosure at 35°C and perform simple operations like moving an axis via the settings menu?

(4) Would the problem persist with firmware 3.12.0 and 3.11.0?

(5) Did it ever happen in the past that the temperature passed 40°C?

Michele Moramarco
Prusa Research

v1p3r0u5, 2023-03-13:
Thanks for your answer.

Long story short, I updated the printer today to the newest firmware 3.12.2 and did some heavy testing. The issue is somehow solved. Don't know if it is the new firmware or if the downgrade to 3.11.0 a couple of weeks ago did the job.

Just for completion, my answers to your questions.

(1) Pruca Slicer and its configuration was and is up-to-date. I did a copy of the printer settings and modified the the start g-code a little bit (1st heat up to 180°, do the mesh bad leveling, heat up to the final temperature. Like this:
`...
...
M104 S180 ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S180 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Z20.0 F1000.0 ; wait a little bit higher
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
...
...`

(2) With the doors opened, the crash didn't occur at all.

(3) Was not the case. Tried it more than once also with axis movement, no crash.

(4) With 3.11.0 no problem at all. Didn't test it with 3.12.0.

(5) Nope, around 36°C was the highest temperature I reached so far inside the enclosure.

So from my point of view the problem does not exist anymore. Don't know why, but to be honest I'm happy it works like this :-)

Prusa-Support, 2023-05-02:
I may seem to understand that the problem doesn't persist anymore.
Can this issue be closed please?

Michele Moramarco
Prusa Research

v1p3r0u5, 2023-05-03:
I have never had it again since the new firmware version. Therefor I am fine with closing it. Thx a lot 👍🏻

Prusa-Support, 2023-05-31:
Glad to hear that. 🙂 

Michele Moramarco
Prusa Research

closed 2023-05-31 by Prusa-Support