prusa3d/Prusa-Firmware #4009: [BUG]<Enter comprehensive title>

piotr-kubica, 2023-02-11:
**Describe the bug**
wrong or missing build instructions for Windows
  
**To Reproduce**
python .\utils\bootstrap.py

**Actual behavior**
`On branch MK3_3.12
Your branch is up to date with 'origin/MK3_3.12'.

nothing to commit, working tree clean

C:\Users\pio\projects\Prusa-Firmware>python .\utils\bootstrap.py
python: can't open file 'C:\\Users\\pio\\projects\\Prusa-Firmware\\utils\\bootstrap.py': [Errno 2] No such file or directory`
  
**Expected behavior**
firmware build succedes

gudnimg, 2023-02-11:
@piotr-kubica You need to use `MK3` (3.13) branch instead of `MK3_3.12`

piotr-kubica, 2023-02-11:
@gudnimg so 3.12 cannot be built in Windows?

gudnimg, 2023-02-11:
Sorry, I edited your comment 😢 I was going to add this for you:

Yes of course :) Long story short we switched to using CMake in 3.13. This is why for 3.12 you need to use one of the other ways listed in the 3.12 README. https://github.com/prusa3d/Prusa-Firmware/blob/MK3_3.12/README.md the Arduino IDE instructions are a bit dated, you can see updated instructions below.

**NOTE**: The Arduino IDE instructions we have for 3.13 should work on 3.12 https://github.com/prusa3d/Prusa-Firmware#arduino-ide-deprecated

piotr-kubica, 2023-02-11:
@gudnimg thanks! In Linux subsystem `dos2unix PF-build.sh` and than running `./PF-build.sh` did the trick. Build succeded!

gudnimg, 2023-02-12:
@piotr-kubica please consider closing this issue, if all works OK :)

closed 2023-02-12 by piotr-kubica