prusa3d/Prusa-Firmware #4015: [ENHANCEMENT] Thermal Model Protection should accept 30W to 60W heaters

knightsljx, 2023-02-13:
This suggestion is tangentially related to issue #3636 but on a broader spectrum.

The current allowable range of the thermal model protection is too narrow to accommodate the most common 24V heater cartridges on the market. Even E3D themselves sell 24V 30W heaters. It should not be on the user to expect to know what is happening, especially when all other printers on the market simply need a PID tune to get the common heaters working.

A better implementation of the Thermal Model Protection should allow a broader range of heater wattages/resistance to accommodate common models. I believe 25-65W should be an ok range inside the software to account for manufacturing tolerances. Thank you.

knightsljx, 2023-02-14:
I just found out that volcano hotends go up to 80W, so the range of accepted heaters need to go up for thermal model protection

darraghbr, 2023-02-24:
According to the docs M310 accepts a 'P' term which is the power in watts, but from what I can see the documentation does not suggest an allowable range? Is there a finite range that is just not documented?

knightsljx, 2023-02-26:
Thermal protection currently does not work on anything but the stock 40W heater. Which is why people on the Revo has to disable thermal protection.
For an 'open source' project, this is a huge oversight.

Prusa-Support, 2023-04-12:
Is this issue still relevant?
If so, please provide thermal calibration logs and all hardware details.

Michele Moramarco
Prusa Research