prusa3d/Prusa-Firmware #4016: thermal anomaly

MG-BFL, 2023-02-13:


MG-BFL, 2023-02-13:
what do I do to continue?  Can I use an older version of the firmware and instll it?

knightsljx, 2023-02-14:
flash back to 3.11 until they fix this 'feature'

MG-BFL, 2023-02-14:
thank you
Rich Johnson
Los Banos, Ca.
Maker

www.Rjgardenrailroad.com
***@***.***

Prusa-Support, 2023-03-24:
Thanks for reaching us.

That's right, in case you use genuine stock Prusa parts, reaching out to the official Support for help could be a good idea.
https://help.prusa3d.com/article/customer-support_2287

Generally speaking, **(1) a careful hotend assembly checkup** + **(2) PID tuning** + **(3a) new calibration** (preferably first **(3b) lower the printhead on the bed** for a more resilient calibration) may solve the problem and can improve the _Termal Model_ accuracy.

Modded printers or printers using third-party components in general, may require tweaking for the _Thermal Model_ to work correctly.
**Please find more about the thermal model functionality and parameters in the relevant Pull Request https://github.com/prusa3d/Prusa-Firmware/pull/3552#issue-1324629390.**
Also please consider creating a log file as explained at https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1262926997 to be uploaded in the thread or a relevant issue: please check the existing issues under https://github.com/prusa3d/Prusa-Firmware/labels/thermal%20model label. It may be of use for our developers to understand the specific problem or for further future improvements.
 Of course, in the case of non-stock parts, please provide precise information about your printer's hardware too.
Meantime, you can disable the thermal calibration by running two simple codes (`M310 S0` followed by `M500`) via Pronterface or similar programs, and "normally" use the printer without the thermal model if necessary.

This issue will be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/3908.

Michele Moramarco
Prusa Research