prusa3d/Prusa-Firmware #4018: [BUG] Unable to set language into CZ after printer firmware update

wyxarcz, 2023-02-15:
**Printer type** - MK3S
**Printer firmware version** - 3.12.1 (5686)

**SD card or USB/Octoprint** - SD card

**Describe the bug**
  I have just updated firmware in my newly bought Prusa 3D printer MK3S to latest available version, after update the printer asked me to select preferred language and I selected Czech language. The printer restarted, information about language change appeared on a LCD during boot time, but after that the language selection menu shown agail and I wasn't be able to leave it until I selected English language. I noticed that the firmware update process is failing during 3rd phase of update (checksum error while reading), but the printer boot up with any other issue (I am not sure if this is importat).

Note 1: I am using Prusa Slicer 2.5.0 (WIN64)
Note 2: Selection of other languages hasn't been tested.
  
**To Reproduce**
  Firmware update has been done regarding to included manual with firmware file. Type of firmware has been mistaken with firmware for MK3 printer first, but after boot, while printer warned me that is wrong firmware I flashed correct one.

leptun, 2023-02-15:
Please post the log from the firmware updater window.

sarusani, 2023-02-15:
Please provide log according to [leptun](https://github.com/leptun).

I just wanted to give you some additional infos.
The firmware update for MK3 has two parts. (with each 2 phases). The first part is the firmware with english lang included. The second one is all other languages. Your update succeeded with the first part and failed during the installation of the additional language files.
You can safely use your printer with the latest features, just not select anything other than english lang until your problem is fixed.

wyxarcz, 2023-02-17:
Hi,

excuse me for delay, I missed notification. Anyway here is the log file:
[log.txt](https://github.com/prusa3d/Prusa-Firmware/files/10764421/log.txt)

Reading of memory and writing to memory done successfully, a data verification is failing every time, sometimes more times, sometimes less times.

leptun, 2023-02-17:
Does it throw the checksum error and verification mismatch every time at the same address in flash? If yes, then it is possible that the microcontroller on the board has a damaged flash memory and needs to be replaced.

Please also try using a different USB port, cable or computer for uploading the firmware. Maybe it's something specific to your current setup causing communication issues.

The reason you can't select the languages is because the upload process in your log stops during the first stage. It never gets to the second stage where it flashes the XFLASH contents (the language data). So you get only a partial update where the flash has (most) of the new firmware and the XLFASH contains outdated translations, hence why it fails to switch to another language.
Considering the content mismatch in the main flash memory, I don't recommend that you use the printer in this state as it might be unsafe. Crashes could happen because the code in flash is corrupted.

sarusani, 2023-02-17:
I was wrong, leptun is right. You should not use your printer since your firmware is corrupted.

wyxarcz, 2023-02-17:
Yes, it worked! :D

I found another shorter USB cable with ferrite cores electrical interference protection, also I placed it out of path of any else cables and flashing verification has been done succesfully including the second phase.

Thanks for your help!

Closing as fixed.

closed 2023-02-17 by wyxarcz