prusa3d/Prusa-Firmware #4019: [BUG] Random Z-axis moves on startup sequence in Silence Mode

jonnieZG, 2023-02-15:
**Printer type** - MK3S
**Printer firmware version** -  3.12.1 (and earlier)

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
  Both

**Describe the bug**
  Sometimes during the printer boot sequence, when Silence Mode is enabled, one or both Z motors make a step in random directions, causing the horizontal axis to slant and go off-sync.

**To Reproduce**
  1. Turn on Silent mode 
  2. Perform Z-axis calibration
  3. Perform bed leveling in Octoprint to ensure that the horizontal axis is actually horizontal (the bed is not slanted).
  4. Turn off the printer and wait for a few seconds
  5. Turn the printer back on
  6. *Sometimes* while there is still the startup screen shown, just before the status screen shows, you will hear one (or both) Z-motors click, while making a random movement.
  7. If you hear the click, go to step 3 to verify that the leveling guide shows the print bed slanted (in fact, it's the horizontal axis that is slanted).
  8. If you don't hear the click, repeat from step 4.
    
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**Video**
  https://youtu.be/zNcuMVp7WTs

At first I thought it was a hardware issue with my motors/controller, but:

1. The printer otherwise works perfectly. 
2. I noticed that this happens when Silent mode is enabled.

Perhaps it would help to always enable Normal mode at the very beginning of the startup to prevent motors from random moves, and when the system is fully up, to return to Silent mode if enabled.

jonnieZG, 2023-02-15:
> My printer is mostly in silent mode and I regularly check my bed leveling with the bed level visualizer in octoprint. I'm very sure I would have noticed if my printer behaves the same (I also never heard a click sound during startup), so I don't think this is widespread behavior. Seems to be something limited to your printer. I'd try contacting Prusa support!

I did contact them several times, and they did not know what to do with that, other than suggesting that it _might_ be a problem with Einsy or motors. In the meantime I noticed that the problem does not occur in Normal mode, so it might be a good idea to see if others noticed the same behavior, before dismissing the ticket as not reproducible.

sarusani, 2023-02-15:
The reason I suggested contacting support is to see if others noticed the same behavior. If that would be the case, the support team would be the ones to know.

Since you say it's one or both motors, it's unlikely to be a motor problem. In my opinion, it would be very unlikely that two motors fail with the same rare error. But maybe someone else has some idea what could cause this.

And don't get me wrong, I'm not saying the firmware should not be touched, but I don't think changing behavior without knowing the reason would be good practice.

gudnimg, 2023-02-15:
I’ve not heard about this issue before. 🤔I have a few questions:

* Can the issue be reproduced without Octoprint?
* When you say “sometimes”, how often are we talking about? Does this happen very rarely? I’m wondering how hard this is to debug. 
* Is the issue present in all recent firmware releases?

jonnieZG, 2023-02-16:
> * Can the issue be reproduced without Octoprint?
Yes, it can.

> * When you say “sometimes”, how often are we talking about? Does this happen very rarely? I’m wondering how hard this is to debug.
If I turn it off, wait approx. 10-20 seconds, and back on for 10 times, it is almost certain that it would happen.

> * Is the issue present in all recent firmware releases?
Yes, for all the releases some 1-2 years back.

gudnimg, 2023-02-16:
> If I turn it off, wait approx. 10-20 seconds, and back on for 10 times, it is almost certain that it would happen.

Thanks, I’ll test this on my machine over the weekend and see if I can reproduce it. (MK3S+) 👍

gudnimg, 2023-02-19:
@jonnieZG I can heard a click from the Z-motors on boot up when I use Stealth mode. But I am having trouble reproducing it again after I switched to Normal mode and back to Stealth.

I can sometimes trigger it with these steps:

1. Boot up printer
2. Once on the status screen, hold down the knob to get the Z-move menu
3. Move up +14mm
4. Turn off the printer
5. Turn on the printer after a few seconds

I still only hear the loud click 1 in 10 attempts perhaps. Haven't seen any movement so far with the naked eye (might be unlucky and watching the wrong motor).

jonnieZG, 2023-02-20:
Thank you for the effort to reproduce the issue! 

> @jonnieZG I can hear a click from the Z-motors on boot up when I use Stealth mode. But I am having trouble reproducing it again after I switched to Normal mode and back to Stealth.

Yes, it is quite random and therefore difficult to reproduce with a certain reliability, but I am glad it is possible to notice the issue after some tinkering.

> I still only hear the loud click 1 in 10 attempts perhaps. Haven't seen any movement so far with the naked eye (might be unlucky and watching the wrong motor).

That pretty much measures with my estimates too. You can feel exactly which axis moved and in which direction, if you place a finger to the bottom of the screws, where they exit their motors.

Did you perform the Prusa-bed-leveling in Octopriint before and afterwards? That will show you **exactly** how much did the motors move, as you will see that the slant between the left and the right side of the print bed.

jonnieZG, 2023-02-20:
The printer design relies on the belief that both Z motors would act the same, as there is only one controller for two motors. It is ok during the operation, but before the motors become initialized, they obviously can move in an uncontrolled way (I did not work with that specific type of step motors, nor got time to read the documentation, but that is my educated guess). 

I gave a look at the source and I see that the st_init() is called relatively late in the process, and it does perform different steps according to which mode is selected. The random moves happen before st_init() is called, so my suggestion is to change the code as follows:

1. Perform z-motors init before all other time-consuming inits, and do it the same way as if Normal mode is selected.
2. After all the devices have been initialized, perform the "proper" Z-motors init, accordingly to the selected mode.

X, Y and Extruder motors are not critical in that way, as they are "singletons", and they will be "homed" at the beginning of each print.

gudnimg, 2023-02-20:
> You can feel exactly which axis moved and in which direction, if you place a finger to the bottom of the screws, where they exit their motors.

I hadnt thought of doing this. Will try it next time 👍

Regarding Octoprint, I don’t a Pi at the moment :( I’ve been waitikg for the Raspbwrry Pi Zero 2 W to come into stock for a few months now. 

I might maybe just buy a Raspberry Pi 3 or 4, but I’d like to keep my setup as small as possible because I move my printer alot to work on the firmware when I’m not printing.

gudnimg, 2023-02-20:
Just a bnote: The “Loud” click I heard yesterday… I’ve heard it numerous times before. But I never thought much of it as I never had any serious issues to attribute to the click.

jonnieZG, 2023-02-20:
> Just a note: The “Loud” click I heard yesterday… I’ve heard it numerous times before. But I never thought much of it as I never had any serious issues to attribute to the click.

Frankly, neither did I, until I started using Octopi. Then I noticed in the Octopi's bed-leveling that the bed gets tilted to one side after a while, and after further time I noticed that the horizontal axis got noticeably slanted to one side. It especially became obvious when printing tall objects, as you could see that the horizonal rod and the frame are not parallel.

Auto-bed leveling is doing a great job for small offsets, but when the horizontal axis hangs too much to one side, that becomes impossible to compensate and to print a proper cube..

Prusa-Support, 2023-03-24:
I'm familiar with the noise too. It may rather regularly occur without being a cause of problems.

Specifically, I think you can intentionally reproduce the noise by changing the motor status from "disabled" to "engaged":
1. make sure the motors are disabled (via _settings_ or _reset button_);
2. hold down the knob to open the "_move Z_" screen and move the Z axis by 1 mm.

You would hear the noise right away.

If you want to avoid the 1 mm movement open the "Adjust Z" screen instead, then change the value by 0.001 mm or so. The Z axis will virtually stay still (one or a few microns movement only), but you will hear that noise and definitely feel the vibration with your finger against the lead screw.

I think the printer was holding the motor position for a moment. Not sure what may be the erratic cause of it during the reboot but the Z axis shouldn't move - or in a negligible manner.

On a side note, I noticed that the lead screw was rotated with one bare hand inn the video, while the Z-axis motor was engaged.
According to my experience, it is very hard or nearly impossible to rotate the lead screw while the Z stepper motors are engaged. Either the lead screw will slip or you will hurt yourself.

I'm wondering if the problem actually affects printability in any way and if we are looking at a problem with the specific motor or the specific port it is connected to.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-13:
I'd like to underline that rotating a lead screw while engaged it's quite difficult.
If rotating the lead screws while they are holding their position is relatively easy and they are not holding the position, there could be a concern about the motor itself (or added weight maybe).
You may want to check if you can feel consistent resistance caused by the "steps" when rotating a freed (unmounted and disconnected) stepper motor.

It is relatively common to hear that noise.
I can speculate with enough confidence that the noise is due to the stepper motor "anchoring itself" to the closest full step.
This happens when the motor is found in an intermediate position (between steps) before being engaged.
Steps are 200/revolution = 1.8°/step.
When you hear the click, it means that the stepper motor moved by approx +/-0.9° to the closest step.

The easiest way to reproduce the noise is:
1. press X reset button,
2. manually rotate the lead screws by <1.8° while refreshing.

At the end of the startup process, the Z motors will be engaged and attracted into the close full-step position.
If the stepper motor is rotated by approx 1.8° (or a multiple) you may not clearly hear the sound. If it was rotated by approx 0.9°C (mid-way between steps) you will hear it louder.

The Z stepper motors may have a tendency to move towards the bottom even so slightly, due to gravity.
When you power off the printer while it is holding a micro-step (intermediate position between steps), it may happen again that the printer will move to the subsequent closes step when engaged too.

This should not be relevant for the printer functionality but one day the printer may beg for a "Calibrate Z" if the X-axis will lose a certain degree of parallelity. No biggies I would say, the printer may be subject to vibration and spontaneous slight geometry changes anyway - usually, this is totally negligible.

In other words, this looks like a hardware limitation to me.

Is this issue relevant to the everyday use and reliability of the printer?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-04:
It seems safe to call it a normal behavior for all stepper motors, and not concerning in any way.
While the Z position may even slightly change for several reasons, the printer's features would normally make it up for that.

This issue apparently describes the regular behavior of the printer, and I hope the above clarifies.
Moreover, the conversation seems to be over.
As a result, the issue should be closed as irrelevant and possibly stale.
Feel free to further keep posting below in case of concerning aspects of such behavior, that may result in anomalies during the everyday use of the printer.

Michele Moramarco
Prusa Research

closed 2023-05-04 by Prusa-Support