prusa3d/Prusa-Firmware #4022: [BUG] Resuming MMU loading after an MMU load failure will sometimes cause a firmware crash/panic

P1xellat3d, 2023-02-16:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.1

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
When the MMU2S is attempting to recover from a failed load (e.g. when multiple load attempts are unsuccessful, and it returns to the main screen with a "MMU load failed" message), prompting the printer to re-attempt a MMU load by pressing the menu button will cause a brief loading of 10cm of filament from the MMU, followed by a firmware crash.

This issue occured on filament slot 2 for me, however I do recall it happening in other slots too.
  
**To Reproduce**

1. Create a scenario where the MMU load is unsuccessful. You can probably do this by tricking the IR sensor on the extruder side.
2. Repeat step 1 until the printer decides that the MMU load is unsuccessful, and displays a message "MMU load failed" on the printer screen.
3. Press the button on the menu to prompt the printer to re-attempt to load from the MMU.
4. The printer will attempt to load from the MMU again, and the MMU will push filament out through the bowden tube for about 10cm before a panic/firmware crash.
  
**Expected behavior**
The printer should complete the MMU load attempt without a firmware crash.
  
**G-code**
G-CODE attached. Note that the error is dependant on a MMU load fail, so you may not get exact results with this G-Code unless you trigger a MMU load failure.

**Crash dump**
Crash dump also attached.

[G-Code and Crash dump.zip](https://github.com/prusa3d/Prusa-Firmware/files/10760829/G-Code.and.Crash.dump.zip)

gudnimg, 2023-02-16:
Sounds related to https://github.com/prusa3d/Prusa-Firmware/issues/3989

P1xellat3d, 2023-02-16:
> Sounds related to #3989

You are correct, this sounds like the same problem. Feel free to merge with the existing issue. Thank you.

Prusa-Support, 2023-03-25:
Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/3989.
No need to merge issues and no worries, the provided information will be accessible from the other issue: mentioning an issue links the two issues.

Michele Moramarco
Prusa Research

closed 2023-03-25 by Prusa-Support