prusa3d/Prusa-Firmware #4024: Einsy Rambo rev.10.a - XFLASH message requires confirmation after every reset/boot up.

ulfertg, 2023-02-17:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.1

**MMU upgrade** - none
**MMU upgrade firmware version** - none

**SD card or USB/Octoprint**
USB/Octoprint

**Describe the bug**
I have an early version of the MK3 with Einsy Rambo rev.10.a.
After every boot up the Display states: 
> External SPI flash XFLASH is not responding.
> Language switch unavailable.

Unlike previous FW-Versions up to 3.11.x this message needs to be confirmed by pressing the printers dial.

I use the printer with Octoprint, so I turn on the printer, walk to the computer, connect Octoprint and then need to walk back to the printer to press the button to acknowledge that message.
  
**To Reproduce**
Boot up an MK3S with Einsy Rambo rev.10.a and latest firmware
  
**Expected behavior**
I wish I could disable this message completely (since in my case it's no problem, I know that my printer only supports English), the pre 3.12 behaviour (message disappears by itself after a short time) would be also preferable (for me) over the current behaviour.

**Video**
https://user-images.githubusercontent.com/36745440/219602574-e53bf2d7-6039-47d8-9ae4-07db24192240.mp4

tprelog, 2023-03-03:
I'm facing also facing this exact issue on the latest firmware

**Printer type -** MK3S
**Printer firmware version -** 3.12.2

**SD card or USB/Octoprint**
USB/Octoprint

**Describe the bug**
I have an early version of the MK3 with Einsy Rambo rev.10.a.
After every boot up, the Display states:

```
External SPI flash XFLASH is not responding.
Language switch unavailable.
```

Unlike previous FW-Versions, this message must be confirmed _every time_ by pressing the printer's dial.

I also agree with the following statement already made. 

>I wish I could disable this message completely (since in my case it's no problem, I know that my printer only supports English), the pre 3.12 behaviour (message disappears by itself after a short time) would be also preferable (for me) over the current behaviour.

tprelog, 2023-03-03:
@ulfertg - I found a way to revert to the old behavior if you compile your firmware. _Try this at your own risk_

In `Firmware/Marlin_main.cpp` edit the following code as shown below ( Line `1084` in FW 3.12.2 ) - Here, I've also shortened the error message, and it will only display for one second.

```C
static void xflash_err_msg()
{
    //puts_P(_n("XFLASH not responding."));
    //lcd_show_fullscreen_message_and_wait_P(_n("External SPI flash\nXFLASH is not res-\nponding. Language\nswitch unavailable."));
    lcd_clear();
    lcd_puts_P(_n("\nExternal SPI Error:\nXFLASH: no response"));
    lcd_wait_for_click_delay(1);
}
```

For reference, here's a screenshot of the changed message (Note this does not change the color of your display to red)

[image]

ulfertg, 2023-03-05:
@tprelog Thank you for sharing this information!
Right now I'm tempted to simply revert to the latest 3.11.x fw, the main new feature (thermal model) isn't working on my i3 anyways (it has a Revo installed).
And I'm not sure if I'm willing to look into compiling the fw myself (with the prospect to having to do it with every new fw from now on) - I have too much other projects going on already.
But in case I'll change my mind about that, I'm glad that you've already found and pointed to the specific code snippet.

Prusa-Support, 2023-03-31:
Thank you all for the report and the suggestion.

Sometimes the error is caused by a short circuit or physical damage on the board (e.g. https://github.com/prusa3d/Prusa-Firmware/issues/3567). A replacement may be needed.

Sometimes reviewing/disconnecting the wires and flashing the firmware again helps.
https://help.prusa3d.com/article/external-spi-flash-w25x20cl-xflash-not-responding-error_112019

In your specific case, the Einsy revision is old and would need a daughterboard (initially included - no longer offered). 
If you are missing the daughterboard the error will be recurrent.

Other than this, it would be safe to discard the message and it should be possible by default.
Our developers will look into this.

Michele Moramarco
Prusa Research

ulfertg, 2023-05-08:
> @ulfertg - I found a way to revert to the old behavior if you compile your firmware. _Try this at your own risk_

@tprelog Today I finally gave your solution a try and after a bit of tinkering with a virtual Linux on my Mac I managed to compile the FW with the changes you've suggested.

Works like a charm, I only regret not having done this earlier.
Thank you!

3d-gussner, 2023-05-08:
@ulfertg Just to be sure. Did you flash the firmware with Prusa Slicer as this is the only supported way to flash the firmware and also the xflash correctly.
Flashing the firmware with OctoPrint Firmware Flasher will cause issues as it doesn't flash the external xflash correctly!

Please try to flash the firmware with PrusaSlicer and provide the Advanced output log to verify that your xflash has been flashed correctly.

ulfertg, 2023-05-08:
@3d-gussner I did indeed flash the FW with OctoPrint Firmware Flasher.
But as far as I understand the issue described in my first post is because I have an older revision of the Einsy Rambo and is to be expected - I saw it also with previous versions of the FW (3.11.x), but it wasn't persistent, so I didn't care much about it.

Should I still try flashing the official 13.12.2 with PrusaSlicer?

3d-gussner, 2023-05-08:
Yes please always flash with PrusaSlicer. There is a PR for the Octoprint Firmware flasher which would solve the problem in Octoprint. Maybe it would be a good idea to ask the owner of the octoprint flasher to consider to merge this PR.

3d-gussner, 2023-05-08:
See https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325 maybe the OctoPrint community can kindly ask @benlye to consider this PR from @arekm.

3d-gussner, 2023-05-08:
@ulfertg It can be that the EINSY has an issue but to be sure try PrusaSlicer Firmware flashing first.

ulfertg, 2023-05-09:
@3d-gussner I flashed the official 3.12.2 with PrusaSlicer as you've requested, I'll attach the output log below.
After that I saw the same behaviour as described in the first post in this thread.
Flashed back to the altered version described by tprelog and it works again as expected (no persistent error msg about the XFLASH not responding.
As far as I understand the matter the Einsy board in my Printer is simply too old, this behaviour is to be expected and being described here: https://help.prusa3d.com/article/external-spi-flash-w25x20cl-xflash-not-responding-error_112019

Since it seems that in my case it's not really a problem, I very much prefer a non-persistent error message.

`avrdude-slic3r -v -p atmega2560 -c wiring -P /dev/cu.usbmodem4343301 -b 115200 -D -U flash:w:0:/Users/ulfertgundermann/Downloads/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 31 2023 at 17:18:47
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/cu.usbmodem4343301
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: reading input file "/Users/ulfertgundermann/Downloads/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex"
avrdude-slic3r: writing flash (247882 bytes):

Writing | ################################################## | 100% 43.96s

avrdude-slic3r: 247882 bytes of flash written
avrdude-slic3r: verifying flash memory against /Users/ulfertgundermann/Downloads/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex:
avrdude-slic3r: load data flash data from input file /Users/ulfertgundermann/Downloads/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex:
avrdude-slic3r: input file /Users/ulfertgundermann/Downloads/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex contains 247882 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 31.97s

avrdude-slic3r: verifying ...
avrdude-slic3r: 247882 bytes of flash verified

avrdude-slic3r done.  Thank you.

avrdude-slic3r -v -p atmega2560 -c arduino -P /dev/cu.usbmodem4343301 -b 115200 -D -u -U flash:w:1:/Users/ulfertgundermann/Downloads/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 31 2023 at 17:18:47
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/cu.usbmodem4343301
                Using Programmer              : arduino
                Overriding Baud Rate          : 115200
avrdude-slic3r: prusa_init_external_flash(): MK3 printer emitted incorrect cfm code: `echo: 3.12.2-`
avrdude-slic3r: arduino_open(): Failed to initialize MK3 external flash programming mode
avrdude-slic3r: Could not open port: /dev/cu.usbmodem4343301

avrdude-slic3r done.  Thank you.

`

daleward, 2023-07-17:
How to edit and compile this change has been 8 years since doing any programing. Or does some one have the file in the hex ready to go?

ulfertg, 2023-07-19:
I followed these instructions: https://github.com/prusa3d/Prusa-Firmware#pf-build
If you have a Linux running on x86 it's pretty straight forward.