prusa3d/Prusa-Firmware #4035: [BUG]3.12.1 Firmware crash after octopi startup

MichielVT, 2023-02-19:
**Printer type** - MK3S
**Printer firmware version** -  3.12.1

**MMU upgrade** No MMU

**SD card or USB/Octoprint**
  Octopi on Raspberry pi 3 connected via USB

**Describe the bug**
  After upgrading from firmware MK3S_3_10_0 to 3.12.1, I'm experiencing a firmware crash after octopi is fully started. On previous versions, this caused a restart of the printer FW and after that it was working fine. With the new firmware a firmware crash occurs when octopi is fully started. After recovering from the crash, the setup is working fine.
The raspbery pi is connected to the printer PSU, so both start at the same time
  
**To Reproduce**
  start printer and octopi at the same time, a firmware crash occurs after octopi is fully started
  
**Expected behavior**
  No firmware crash
  
**G-code**
  No specific G-Code

**Video**
  Please attach a video. It usually helps to solve the problem.

sarusani, 2023-02-19:
My octopi and printer are on the same power strip and always start at the same time. After OctoPrint is ready and connects, the printer restarts as it always did. (I'm also on 3.12.1)

Does this happen all the time for you?
Did you change anything else?
Have you tried using another USB cable?

MichielVT, 2023-03-06:
Hi, it's indeed happening all the time for me. I did not try another usb cable yet. No other changes were made. The fw crash screen states that data has been logged, but I don't know where I can retrieve that log. Can I somehow opload the log here for further inspection?

3d-gussner, 2023-03-06:
@MichielVT Please update to FW 3.12.2 as there is a fix for host systems like Octoprint.
Please contact Live support and have a look at https://help.prusa3d.com/article/crash-dump_364959

MichielVT, 2023-03-08:
@3d-gussner  thank you for the information, after upgrading to 3.12.2 I no longer had the crashes after startup.

3d-gussner, 2023-03-08:
Please consider to close this issue

Prusa-Support, 2023-03-31:
Please allow me to close this issue as solved.

Michele Moramarco
Prusa Research

closed 2023-03-31 by Prusa-Support