prusa3d/Prusa-Firmware #4038: Thermal Anomaly problems, mainly at print ending

Escrich, 2023-02-19:
**Printer type** -MK3S
**Printer firmware version** - 3.12.1

**MMU upgrade** - No MMU

Printing from SD Card

Thermal anomaly text on printer's screen, just after print part, when is going up and moving hotbed forward to totally finish
  
Some problems related with this thermal anomaly, and some others referring gantry misalignment after last firmware update, it was necessary to do a PID tunning, other wise was no possible to print anything because of the Thermal anomaly error, wires are not broken at all, printer remains as in the original state with only minor changes, made long time ago, the changes are layer fan duct, changed as in the picture, and aluminium block changed by a brass one, all thermistors are originals form Prusa, original Pinda sensor too.

All messages and problems started after last firmware change.

After last PID tunning, I can print, but I get the error at print ending

[image]

3d-gussner, 2023-02-21:
@Escrich Thanks for the report. The Thermal Model is tailored around the "stock" E3D hotend and we expected that modified hotends, heat blocks, fans/shrouds may cause issues.

The brass heatblock has a different thermal behavior and your fan duct may cool this heatblock even faster.

> it was necessary to do a PID tunning, other wise was no possible to print anything

A PID tuning is always recommend but with changes like you have it is a must. 

Can you please provide some thermal model debug logfiles?

Please read [prusa.io/tm-cal ](https://help.prusa3d.com/article/thermal-model-calibration_382488 )
We need log files ideally with timestamps, do you have by chance OctoPrint or a linux connected?

Run the thermal model calibration in debug mode using gcodes
 
```
M503 ; report settings
M155 S1 C3 ;enable advanced temp and fan logging
D70 S1 ;enable thermal model debug logging
M310 A F1 ;run thermal model calibration
M500 ;save new values
```

Please also try to run a serial logger during the prints to capture the TM error: messages

>and some others referring gantry misalignment

What do you mean?

Escrich, 2023-02-21:
Dear all, this is the log generated following your instructions, I hope it will help.

 

One more thing I detected after firmware change, I have, from years ago, defined two build plates, the smooth one, and the textured one, always I have defined the Textured one as the default one, then randomly, since the firmware update, the current one changes on the machine, without to change or touch anything by myself, and you need some attempts, not at first, to finally set the one you want, you can come into the menu and choose the one you like, but not always change, o simply the machine sets the opposite one as the chosen one.

 

Regards:

 

Jose M. Escrich

Arganda del Rey 

Madrid 28500

Spain

Enviado el: martes, 21 de febrero de 2023 8:08
Para: prusa3d/Prusa-Firmware
CC: Jose M. Escrich; Mention

 

2023-02-21 14:03:54,736 - Connecting...
2023-02-21 14:03:55,784 - start
2023-02-21 14:03:55,784 - Printer is now online.
2023-02-21 14:03:57,559 - echo: 3.12.1-5686
2023-02-21 14:03:57,559 - echo: Last Updated: Jan 20 2023 10:31:23 | Author: (none, default config)
2023-02-21 14:03:57,559 - Compiled: Jan 20 2023
2023-02-21 14:03:57,575 - echo: Free Memory: 2595  PlannerBufferBytes: 1792
2023-02-21 14:03:57,575 - echo:Stored settings retrieved
2023-02-21 14:03:57,691 - adc_init
2023-02-21 14:03:57,691 - CrashDetect ENABLED!
2023-02-21 14:03:58,803 - FSensor ENABLED
2023-02-21 14:03:58,994 - Sending 0xFF
2023-02-21 14:03:59,010 - echo:SD card ok
2023-02-21 14:04:25,674 - MMU not responding - DISABLED
2023-02-21 14:04:47,831 - SENT: M105
2023-02-21 14:04:47,831 - RECV: ok T:203.3 /0.0 B:19.6 /0.0 T0:203.3 /0.0 @:0 B@:0 P:23.6 A:26.0
2023-02-21 14:04:50,972 - SENT: M105
2023-02-21 14:04:50,972 - RECV: ok T:201.4 /0.0 B:19.8 /0.0 T0:201.4 /0.0 @:0 B@:0 P:23.3 A:25.9
2023-02-21 14:04:54,099 - SENT: M105
2023-02-21 14:04:54,105 - RECV: ok T:199.7 /0.0 B:19.9 /0.0 T0:199.7 /0.0 @:0 B@:0 P:23.6 A:26.1
2023-02-21 14:04:54,975 - >>> m503
2023-02-21 14:04:54,991 - SENT: M503
2023-02-21 14:04:54,991 - RECV: echo:Steps per unit:
2023-02-21 14:04:54,991 - RECV: echo:  M92 X100.00 Y100.00 Z400.00 E280.00
2023-02-21 14:04:54,991 - RECV: echo:UStep resolution:
2023-02-21 14:04:54,991 - RECV: echo: M350 X16 Y16 Z16 E32
2023-02-21 14:04:55,006 - RECV: echo:Maximum feedrates - normal (mm/s):
2023-02-21 14:04:55,006 - RECV: echo:  M203 X200.00 Y200.00 Z12.00 E120.00
2023-02-21 14:04:55,006 - RECV: echo:Maximum feedrates - stealth (mm/s):
2023-02-21 14:04:55,006 - RECV: echo:  M203 X100.00 Y100.00 Z12.00 E120.00
2023-02-21 14:04:55,022 - RECV: echo:Maximum acceleration - normal (mm/s2):
2023-02-21 14:04:55,022 - RECV: echo:  M201 X1000 Y1000 Z200 E5000
2023-02-21 14:04:55,022 - RECV: echo:Maximum acceleration - stealth (mm/s2):
2023-02-21 14:04:55,029 - RECV: echo:  M201 X960 Y960 Z200 E5000
2023-02-21 14:04:55,029 - RECV: echo:Acceleration: P=print, R=retract, T=travel
2023-02-21 14:04:55,029 - RECV: echo:  M204 P1000.00 R1250.00 T1000.00
2023-02-21 14:04:55,044 - RECV: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
2023-02-21 14:04:55,060 - RECV: echo:  M205 S0.00 T0.00 B0.00 X8.00 Y8.00 Z0.40 E4.50
2023-02-21 14:04:55,060 - RECV: echo:Home offset (mm):
2023-02-21 14:04:55,060 - RECV: echo:  M206 X0.00 Y0.00 Z0.00
2023-02-21 14:04:55,060 - RECV: echo:PID settings:
2023-02-21 14:04:55,060 - RECV: echo:   M301 P21.45 I2.15 D53.58
2023-02-21 14:04:55,060 - RECV: echo:PID heatbed settings:
2023-02-21 14:04:55,060 - RECV: echo:   M304 P126.13 I4.30 D924.76
[... 879 lines cut ...]
2023-02-21 14:07:33,723 - RECV: TML 0 1 0 42afa000 41d4eaab
2023-02-21 14:07:33,986 - RECV: TML 0 1 0 42af5a42 41d37555
2023-02-21 14:07:34,193 - RECV: T:87.7 /0.0 B:20.3 /0.0 T0:87.7 /0.0 @:0 B@:0 P:24.5 A:26.4
2023-02-21 14:07:34,193 - RECV: E0:3840 RPM PRN1:2040 RPM E0@:255 PRN1@:255
2023-02-21 14:07:34,256 - RECV: TML 0 1 0 42af0000 41d69555
2023-02-21 14:07:34,425 - RECV: echo:busy: processing
2023-02-21 14:07:34,527 - RECV: TML 0 1 0 42aeadf3 41d73555
2023-02-21 14:07:34,795 - RECV: TML 0 1 0 42ae6000 41d41555
2023-02-21 14:07:35,059 - RECV: TML 0 1 0 42ae1627 41d62aab
2023-02-21 14:07:35,197 - RECV: T:87.0 /0.0 B:20.1 /0.0 T0:87.0 /0.0 @:0 B@:0 P:24.3 A:26.8
2023-02-21 14:07:35,212 - RECV: E0:3840 RPM PRN1:2040 RPM E0@:255 PRN1@:255
2023-02-21 14:07:35,328 - RECV: TML 0 1 0 42add069 41d7a000
2023-02-21 14:07:35,614 - RECV: TML 0 1 0 42ad65be 41d52000
2023-02-21 14:07:35,877 - RECV: TML 0 1 0 42ad1be6 41d55555
2023-02-21 14:07:36,147 - RECV: TML 0 2 0 42acbd8a 41d5f555
2023-02-21 14:07:36,216 - RECV: T:86.4 /0.0 B:20.1 /0.0 T0:86.4 /0.0 @:0 B@:0 P:24.4 A:26.7
2023-02-21 14:07:36,216 - RECV: E0:3840 RPM PRN1:2040 RPM E0@:255 PRN1@:255
2023-02-21 14:07:36,348 - SENT: M105
2023-02-21 14:07:36,417 - RECV: TML 0 0 0 42ac77cc 41d5f555
2023-02-21 14:07:36,433 - RECV: echo:busy: processing
2023-02-21 14:07:36,680 - RECV: TML 0 1 0 42ac1d8a 41d52000
2023-02-21 14:07:36,966 - RECV: TML 0 2 0 42abd3b1 41d8e000
2023-02-21 14:07:37,220 - RECV: T:85.9 /0.0 B:20.3 /0.0 T0:85.9 /0.0 @:0 B@:0 P:24.4 A:27.1
2023-02-21 14:07:37,220 - RECV: E0:3840 RPM PRN1:2040 RPM E0@:255 PRN1@:255
2023-02-21 14:07:37,236

3d-gussner, 2023-02-21:
@Escrich Hola Jose, thanks but I forgot to mention that you have to wait first to cool down the printer before you execute the gcodes.

1. Wait until printer is cooled down to room temperature
2. execute  the gcode command one after another and not a copy paste of all.
3. `M503`
4. `M155 S1 C3`
5. `D70 S1`
6. `M310 A F1`
7. When finished save the new values with `M500`

Which terminal software did you use? I see it sends `M105` every few seconds.

Please don't use Pronterface as it may corrupt the log files. OctoPrint and Putty don't mess up the debug logging.

Please check other issues about your feature request and if not there report a new one.

Escrich, 2023-02-25:
Dear all, this is a really dangerous matter, sometimes, I start to print something, and after see how the firsts layers go, I can go to the street, without any worry about my Prusa Original, mounted by you, but today, after return from street, I came in panic, I was printing a simple part using ABS, total printing time 17 minutes, the bed arises to 100 ºC, when I returned, after three hours, the error thermal protection was on the screen and the bed was still at 100ºC, Three hours at those temperature with no one around, please let me know how to disconnect the thermal image, or please again, inform me if I need to come back to the previous Prusa firmware again, because it’s not on my aim to place fire to my house, the machine is stopped waiting for your answer.

 

Regards:

 

Jose M. Escrich

Arganda del Rey 

Madrid 28500

Spain

Enviado el: martes, 21 de febrero de 2023 19:56

Prusa-Support, 2023-03-31:
I'm sorry to hear about that, but the heatbed is designed to withstand to100°C for even the longest ABS prints. There is no danger.
I assume the printer was waiting for user intervention and holding the heatbed temperature to prevent the printed object from detaching.

Back to the thermal anomaly now.
The recommended steps would be:
1. new PID tuning at 230-250° (or the most frequently used print temperature);
2. auto-home to put the print head close to the bed, and increase thermal stress during the calibration;
3. new thermal model calibration.

In your case though, we can't guarantee this will be sufficient, because you are not using all Prusa stock parts.
Reverting to full genine Prusa parts and design may solve the problem.
Until then, there are ways to disable the thermal model protection. We don't recommend that but you can.
a) Either downgrade the firmware to version 3.11;
b) or send G-code `M310 S0` to disable it followed by `M500` to save the change.

However, before any change, we would appreciate another thermal model calibration log as per the new instructions described at https://github.com/prusa3d/Prusa-Firmware/issues/4038#issuecomment-1438953147.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-03-31:
A little update.

The most recent [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) release should be more broadly compatible.
Please give it a try.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-19:
Is this issue still relevant? Are calibration logs available?
Please close the issue if the problem doesn't persist, otherwise, we may close it as stale.

Michele Moramarco
Prusa Research

closed 2023-06-28 by Prusa-Support