prusa3d/Prusa-Firmware #4039: [bug]MK3s+ will not complete z calibration - Microswiss M2 hardened steel nozzle installed.

3dhomeward, 2023-02-19:
**Printer type** - MK3s+
**Printer firmware version** 
 2.5.0

**MMU upgrade** - no
**MMU upgrade firmware version** 
not applicable

**SD card or USB/Octoprint** 
SD card
  
**Describe the bug**
  The Z calibration wizard stops when the nozzle drops to the bed. "Calibration failed! Check the axes and run again." I have done the temp model calibration.

**To Reproduce**
  I select z calibration on the LCD. I place office paper on the bed.
  
**Expected behavior**
  I expected the nozzle to drop to the bed and begin its calibration process.

3d-gussner, 2023-02-20:
@3dhomeward Thanks for the bug report but Printer firmware version `2.5.0` in the description doesn't make sense, this may be the Prusa Slicer version.

Please verify if your PINDA is correct see https://help.prusa3d.com/article/nozzle-crashed-into-the-heatbed_112003.  

As you have run the Thermal model calibration I guess you have at least FW 3.12. installed.

Please check LCD Support -> Firmware version and if it is older than FW3.12.1 please checkout https://help.prusa3d.com/downloads or https://github.com/prusa3d/Prusa-Firmware/releases for a new version.

3dhomeward, 2023-02-20:
@3d-gussner Thank you, yes, I had the wrong number there. I changed it.
I have a new SuperPINDA. I bought it in January. I did not have any issues with calibration until I installed the update.

Prusa-Support, 2023-02-21:
@3dhomeward. Thanks for reaching us.

Maybe a complete calibration via Wizard would help, or the self-test included in the Wizard will highlight other anomalies.
There is a chance that the PINDA probe or the nozzle is not correctly installed, or that the Z axis doesn't move correctly.
This kind of problem can most often be solved through a proper troubleshooting session.

Please discuss your problem with our Tech Support before advancing this issue on GitHub, and possibly bring documentation about the problem with you.
We are available 24/7 via Live Chat or email.
https://help.prusa3d.com/article/customer-support_2287
https://help.prusa3d.com/article/photo-and-video-documentation_112155

If the chat doesn't work for you, you can write an email to my attention at info@prusa3d.com, with a reference to the GitHub issue (Prusa-Firmware/issues/4039).

Michele Moramarco
Prusa Research

3dhomeward, 2023-02-21:
For the record, I went through the complete calibration Wizard more than once and each time it stopped when it reached the z calibration part. The new SuperPINDA was installed more than a week before upgrading the firmware. I used z calibration a few times after SuperPINDA was installed without any issues. I did not experience a problem with z calibration until I upgraded the firmware.

ackerthehacker2, 2023-02-22:
Did you check with a piece of metal if your SuperPINDA is working ...and what does the sensor say in the LCD Support -> Sensor Info  (you you see PINDA turn on and off)

3dhomeward, 2023-02-22:
SuperPINDA was working well before I updated the firmware. I completed a z calibration the day before updating, after I moved the printer.

3dhomeward, 2023-02-23:
I went ahead and tested SuperPINDA anyway to be sure it was working according to "P.I.N.D.A./SuperPINDA Sensor testing" and it works as it should. Some other things I did were 1. Reinstall the Ruby nozzle 2. Run the wizard 3 times after installation. The LCD message is the same as it was with the steel nozzle. I will just go through troubleshooting since neither nozzle will calibrate.

ackerthehacker2, 2023-02-23:
.Sounds like your Super Pinda is too far away from the bed and is not triggering or loose.   I used this to set it :   

￼

But since you can’t print you are stuck with the wire tie.    Did you do the full xyz calibration and where it checks the 4 spots …remove the steel sheet replace with paper and does the paper move at all?

3dhomeward, 2023-02-23:
I watched the printer checking the nine spots it checks in first layer calibration during one of the processes I tried and wondered if it could do a first layer calibration despite not completing the z-axis calibration.The first message on my LCD is that a Z calibration is still needed. I set a nozzle temperature. The machine completed a first layer calibration and I printed two cubes.

Prusa-Support, 2023-03-03:
There are various ways to check the PINDA and its readings.
https://help.prusa3d.com/article/p-i-n-d-a-superpinda-sensor-testing_2091
Also, the PINDA could be simply wobbling.

It is a fact that the calibration is sensitive to the tenth of a millimeter but the printer geometry can slightly change for various reasons. From my experience as a technician lowering the PINDA half a ring and repeating the whole calibration can help.

Other than this, the complete troubleshooting may require some time or special attention.
The official Customer Support can guide you through the solution (and sometimes offer warranty replacements).

If you are looking at a bug, then we should focus on the reproducibility of the issue for other printers.
I think this case deserves regular troubleshooting though, so please give it a try reaching out the Customer Support, possibly with evidence at the hand.

Michele Moramarco
Prusa Research

3dhomeward, 2023-03-04:
I have been printing despite receiving opening messages about z-axis calibration not being completed. SuperPINDA is working as expected.

ackerthehacker2, 2023-03-09:
The problem is when you run the wizard ...xyz calibration fails as a calibration issue when it brings the extruder down to the bed.   The Super Pina is activated and you get the error.  No matter what you do with the PINDA it will error when it brings it down to the bed.   There seems to be a funny noise near the top of Z as it brings it up to start, or it only brings it up a few mm.  Z:5.00? is shown on the LCD.

 If you raise the extruders all the way up and then start xyz, it works fine.

3dhomeward, 2023-03-10:
I will try this after I have received a new extruder.

Prusa-Support, 2023-03-27:
@ackerthehacker2 The noise when the Z-axis moves all the way up at the beginning of the calibration is expected. It must happen: the X-ends must hit the Z-top and level up against them. The noise is a sign that the X-ends are hitting the Z-tops and making the Z-motors skip steps, which is totally fine and expected.
The axis position is zeroed every time the printer resets, there is no other way. That's why the printer will always find the home position at the beginning of the prints. When you see Z=5 on the top position, it means the initial position was 5 mm away from that top. No worries, this is not concerning in any way but the expected behavior.

.

@3dhomeward Skipping the calibration should be avoided. Otherwise, the prints could be subject to failure and the printer may be subject to damages.
A proper troubleshooting session should solve the problem. What was the solution suggested by the official Customer Support?
This case deserves individual troubleshooting and GitHub is not really intended for individual support.
You can write an email to my attention (see signature) at info@prusa3d.com, with a reference to this issue (4039) for clarification.

The bug in this issue is not confirmed and the issue may be closed as invalid.

Michele Moramarco
Prusa Research

wisonlau, 2023-04-01:
I have the same problem, using E3D revo six on the hot end.

gudnimg, 2023-04-01:
Just adding, the firmware outputs this message if the Z position does not meet this condition: $$4.85\text{mm} \leq \text{Z-axis machine position}\leq 5.15\text{mm}$$ In other words it must be 5.0mm +- 0.15mm when the nozzle head is close to the bed.

Does the LCD show the Z-position on the status screen?

wisonlau, 2023-04-01:
> @3dhomeward. 感谢您联系我们。
> 
> 也许通过向导进行完整校准会有所帮助，或者向导中包含的自检会突出显示其他异常情况。 有可能是 PINDA 探头或喷嘴没有正确安装，或者 Z 轴没有正确移动。 此类问题通常可以通过适当的故障排除会话来解决。
> 
> 在 GitHub 上推进此问题之前，请与我们的技术支持讨论您的问题，并可能随身携带有关该问题的文档。 我们通过实时聊天或电子邮件全天候 24/7 提供服务。 [https://help.prusa3d.com/article/customer-support_2287 ](https://help.prusa3d.com/article/customer-support_2287) https://help.prusa3d.com/article/photo-and-video-documentation_112155
> 
> 如果聊天不适合您，您可以通过[info@prusa3d.com](mailto:info@prusa3d.com)写一封电子邮件引起我的注意，并参考 GitHub 问题 (Prusa-Firmware/issues/4039)。
> 
> Michele Moramarco Prusa 研究

[image]

I had the same problem, but I found a temporary solution.

[image]

The cause of this problem is that the distance of the machine is not consistent with the actual distance.
I set the machine to x-axis 210, but the real distance of the machine is not 210. I need to turn off the machine and screw the z-axis screw to the top to complete the z-calibration.
I hope that when the machine is leveled, let the user confirm that the z-axis is at the top to avoid the same error.

michael-borck, 2023-04-15:
Based on the picture, try removing the bearing for the z-axis lead screws and see if you still have the problem.

Why?   I suspect you have some binding in the z-axis toward the top preventing early triggering of the sensor less homing in the z direction.   It looks like you have fixed the top of the lead screw with a bearing.  Although the bearing can 'spin' there may be tension in the lead screw if it not perfectly 'straight' as it wants to move x/y direction.  Your guide/smooth rods will keep things straight, the lead screw is there to do the lifting.

wisonlau, 2023-04-18:
> Based on the picture, try removing the bearing for the z-axis lead screws and see if you still have the problem.
>
> Why? I suspect you have some binding in the z-axis toward the top preventing early [...]

Thank you, it's this issue that caused it

Prusa-Support, 2023-05-09:
This issue may be closed as invalid.
So far, we couldn't confirm a bug but that the issue may be caused by assembly-related problems.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-16:
No bug confirmed but assembly-related concerns are still valid.
Please allow me to close this issue as not relevant to the repository and possibly invalid.

For help with this matter, please refer to our Customer Support.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

closed 2023-05-16 by Prusa-Support