prusa3d/Prusa-Firmware #4040: [BUG] Thermal Anomaly During Calibration - Factory build MK3S

iksaif, 2023-02-20:
**Printer type** - MK3S+
**Printer firmware version** - 3.12.0

**Describe the bug**
Thermal calibration fails on a factory built printer. Everything else looks fine

Logs attached  
[20230220_123825__putty.log](https://github.com/prusa3d/Prusa-Firmware/files/10784662/20230220_123825__putty.log)

By the way the instructions here https://help.prusa3d.com/article/thermal-model-calibration_382488 are a bit misleading

Copying `M310 ;report current temp model settingsM155 S1 C3 ;enable advanced temp and fan loggingD70 I1 ;enable temp model debug loggingM310 A F1;run temp model calibration` gives a bunch of "command not found!". But typing the gcodes by hand seems to work (with putty, configured like the doc says)

leptun, 2023-02-20:
You must not send the gcode comments alongside the gcode command. The `command not found!` message is generated by the USB to serial chip interpreting the `;` as the escape command for commands to that chip. Hosts such as Octoprint do not send the comments after the gcode command, so it doesn't generate this issue.

On an unrelated note, the formatting of that page seems to be messed up. Newlines seem to be missing.

iksaif, 2023-02-20:
Yes I think that's the missing newlines that messed it up,
anyway I managed to run the commands as you can see at the end of the log (https://github.com/prusa3d/Prusa-Firmware/files/10784662/20230220_123825__putty.log)

Note that with this firmware the temperature seems all messed up and the PLA seems to be bubbling, I'll try to revert to the previous version

iksaif, 2023-02-20:
The relevant part seems to be

```
T:192.5 /230.0 B:18.9 /0.0 T0:192.5 /230.0 @:127 B@:0 P:0.0 A:23.0
E0:3660 RPM PRN1:0 RPM E0@:255 PRN1@:0
T:193.6 /230.0 B:18.8 /0.0 T0:193.6 /230.0 @:127 B@:0 P:0.0 A:23.2
E0:3660 RPM PRN1:0 RPM E0@:255 PRN1@:0
echo:busy: processing
T:194.3 /230.0 B:18.9 /0.0 T0:194.3 /230.0 @:127 B@:0 P:0.0 A:23.1
E0:3660 RPM PRN1:0 RPM E0@:255 PRN1@:0
T:194.8 /230.0 B:18.2 /0.0 T0:194.8 /230.0 @:127 B@:0 P:0.0 A:23.0
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0
echo:busy: processing
T:196.1 /230.0 B:18.5 /0.0 T0:196.1 /230.0 @:127 B@:0 P:0.0 A:23.1
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0
TM: error |-1.283469|>1.200000
LCD status changed
TM: error |-1.263687|>1.200000
TM: error |-1.403226|>1.200000
TM: error |-1.316440|>1.200000
T:196.1 /230.0 B:18.9 /0.0 T0:196.1 /230.0 @:127 B@:0 P:0.0 A:23.2
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0
TM: error |-1.434968|>1.200000
TM: error |-1.334314|>1.200000
TM: error |-1.403121|>1.200000
echo:busy: processing
TM: error |-1.480621|>1.200000
T:196.1 /230.0 B:18.2 /0.0 T0:196.1 /230.0 @:127 B@:0 P:0.0 A:23.0
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0
TM: initial R 230C
TM: error |-1.779592|>1.200000
TM: error triggered!
ok
// action:cancel
Error:Printer stopped due to errors. Supervision required.
TM: calibr. failed!
echo:Temperature Model settings:
echo:  M310 I0 R20.50
echo:  M310 I1 R18.40
echo:  M310 I2 R16.70
echo:  M310 I3 R15.20
echo:  M310 I4 R14.10
echo:  M310 I5 R13.30
echo:  M310 I6 R12.70
echo:  M310 I7 R12.10
echo:  M310 I8 R11.70
echo:  M310 I9 R11.30
echo:  M310 I10 R11.00
echo:  M310 I11 R10.80
echo:  M310 I12 R10.60
echo:  M310 I13 R10.40
echo:  M310 I14 R10.20
echo:  M310 I15 R10.10
echo:  M310 P38.00 C13.24 S1 B1 E1.74 W1.20 T-7.00
TM: error |-1.813819|>1.200000
tmc2130_home_enter(axes_mask=0x04)
TM: error cleared
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
T:194.4 /0.0 B:18.8 /0.0 T0:194.4 /0.0 @:0 B@:0 P:0.0 A:23.1
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0

```

3d-gussner, 2023-02-21:
> Note that with this firmware the temperature seems all messed up and the PLA seems to be bubbling, I'll try to revert to the previous version

Does the PLA print normal with the previous firmware, no bubbling?

If you have bubbling PLA the temperature must be way higher than it should be. Exactly this kind of issue the Thermal Model should find/report.

Please contact tech support.

iksaif, 2023-02-21:
> Does the PLA print normal with the previous firmware, no bubbling?

Yes it printed fine with 3.11.

The bubbling might or not be related to the fact that it's also the first print in the enclosure, and maybe the filament guide is moist. I reverted to 3.11 and after a while I was able to print fine again.

3d-gussner, 2023-02-21:
@iksaif Thanks for the quick response! 

Can you please follow these steps?
 
Connect to your printer via USB and a serial terminal.

Send `M503`  to report settings

Run the PID tuning from LCD Calibration -> PID calibration

```
M155 S1 C3 
D70 S1
M310 A F1
M500 
```

Also try to log your prints with `M155 S1 C3` in the start gcode to see when TM triggers.

iksaif, 2023-02-22:
[20230222_153007__putty.log](https://github.com/prusa3d/Prusa-Firmware/files/10804866/20230222_153007__putty.log)

Here is the log of running the PID Calibration then the commands you suggested

3d-gussner, 2023-02-22:
@iksaif Sorry not being very clear with the steps.
Before you run the TM calibration the printer must be cooled down to room temperature.
1. Wait until the printer is idle and cooled down to room temperature
2. Send `M503` to report current settings
3. Send `M155 S1 C3` to auto report Temps and fans
4. Send `D70 S1` to enable TM debug logging. This can only log a "short" time due to resources and that it is important to have the printer "cold"
5. Send `M310 A F1` to run the TM calibration
6. Send `M500` to store the information.

iksaif, 2023-02-22:
Thanks, here is the log starting cold
[20230222_164807__putty.log](https://github.com/prusa3d/Prusa-Firmware/files/10805714/20230222_164807__putty.log)

3d-gussner, 2023-02-22:
@iksaif Thanks, what happened during the TM calibration? I see that it didn't finish.

iksaif, 2023-02-22:
I stopped the printer as soon as "thermal anomaly " was triggered and the
printer started bopping very loudly

3d-gussner, 2023-02-23:
@iksaif Thanks again for the quick answer. We are close to a new release which should improve the Thermal model calibration.
Can you try this firmware with the same steps as above and provide the log files and feedback?

**This isn't an official release just a test firmware!**
 
[prusa3d_fw_MK3S_MK3S+_t3_12_2_5711.zip](https://github.com/prusa3d/Prusa-Firmware/files/10810816/prusa3d_fw_MK3S_MK3S%2B_t3_12_2_5711.zip)

iksaif, 2023-02-23:
[20230223_211400__putty.log](https://github.com/prusa3d/Prusa-Firmware/files/10818043/20230223_211400__putty.log)

Here it is

iksaif, 2023-02-23:
Yesterday

```
echo:busy: processing
TML 0 2 7f 4345c35e 41b00000
T:197.8 /230.0 B:17.7 /0.0 T0:197.8 /230.0 @:127 B@:0 P:0.0 A:22.0
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0
TML 0 0 7f 43463943 41afc000
TML 0 1 7f 43467ca2 41af0000
TML 0 2 7f 434695e5 41b10000
TM: error |-1.285994|>1.200000
```

Today

```
echo:busy: processing
TML 0 2 7f 43451af3 41ad8000
T:197.1 /230.0 B:17.7 /0.0 T0:197.1 /230.0 @:127 B@:0 P:0.0 A:21.7
E0:3840 RPM PRN1:0 RPM E0@:255 PRN1@:0
TML 0 1 7f 43450a1b 41ae4000
TML 0 0 7f 434555e5 41ad4000
TML 0 2 7f 4345aa1b 41ad4000
TM: error |-1.229254|>1.200000
```

alexgao930, 2023-02-25:
I am having the same issue as @iksaif. I flashed my printer to the newest testing version, but printer still won't be able to  complete the calibration. 

Recv: echo:busy: processing
Recv: T:49.8 /230.0 B:22.0 /0.0 T0:49.8 /230.0 @:127 B@:0 P:0.0 A:29.0
Recv: E0:0 RPM PRN1:0 RPM E0@:0 PRN1@:0
Recv: TM: error |-1.378109|>1.200000
Recv: TML 0 0 7f 4248d555 41e77555
Recv: TM: error |-1.412857|>1.200000
Recv: TML 0 1 7f 4249aaab 41ec0aab

Any ideal?

iksaif, 2023-03-06:
👋 Any news about this one ? I'm happy to provide any info if needed. I'm not back on the previous firmware and printing without any issue

Prusa-Support, 2023-04-01:
Sorry for the delay.
Would you mind giving it one more try with the most recent [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) release?

For a safer and more accurate thermal model calibration please do as follows.
1. Run a PID tuning at 230-250° (or the most frequently used print temperature).
2. Run an auto-home so that the print-head will be close to the bed.
This will increase thermal stress during the thermal model calibration.
3. Run the thermal model calibration again (calibration menu).
Please make sure that the printer won't be reached by airflows (fans, A.C., opened windows/doors...). 

If the above doesn't help, please provide clear close-up pictures of the heater-block block from below and the two sides. Make sure that the thermistor capsule and hear cartridge are visible from both sides.

Michele Moramarco
Prusa Research

iksaif, 2023-04-03:
Since last time, I eventually had a huge problem with the heater block
which started melting the extruder PETG parts themselves while reporting a
~230C temperature.
I ordered replacement parts for the E-axis (PETG + full heater block)  and
rebuilt everything. Now I don't have bubbling filament anymore.

I still need to try to upgrade to 3.12.2, which I might try during the week
and report back.

Prusa-Support, 2023-04-21:
I'm sorry about that.
Thermal anomalies shouldn't be taken lightly.
Please don't ignore _thermal anomaly_ warnings and report back to the official support channels in case of troubles. Our Customer Support will help the customer to check the hardware and find a suitable solution.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-30:
In the case of genuine Prusa stock parts, first thing first, please reach out to our Support in our official support channels.
https://help.prusa3d.com/article/customer-support_2287

Thank you.

Michele Moramarco
Prusa Research