prusa3d/Prusa-Firmware #4045: [BUG]3.12.1 crashes in a middle of a print

ackerthehacker2, 2023-02-22:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.1

**MMU upgrade** - none
**MMU upgrade firmware version** - n/a

**SD card or USB/Octoprint**
 USB/Octoprint

**Describe the bug**
Printing a job multiple times ...have already printed this 5 times ... in the middle of print, the printer crashes/resets itself
  
**To Reproduce**
Not doing anything special.  Have a serial.log
  
**Expected behavior**
Not crash
  
[roll_PrusaRoller_0.2mm_PETG_MK3S_6h27m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/10799771/roll_PrusaRoller_0.2mm_PETG_MK3S_6h27m.gcode.zip)
[serial.log.zip](https://github.com/prusa3d/Prusa-Firmware/files/10799772/serial.log.zip)

**G-code**
attached

**Video**
  Please attach a video. It usually helps to solve the problem.

ackerthehacker2, 2023-02-22:
Its not octoprint crashing ... Octoprint has been up for almost 2 days

v1p3r0u5, 2023-02-22:
@ackerthehacker2 Just out of curiosity, as it sounds very similar to a problem I am having. Do you print inside an enclosure or do you print somewhere with higher ambient temperatures than 30 degrees celsius?

ackerthehacker2, 2023-02-22:
I have a space heater in my office and its 78F which is 26C

v1p3r0u5, 2023-02-22:
So then it is most likely not the same issue as I am having.

gudnimg, 2023-02-27:
@ackerthehacker2 it looks like a power supply issue. UVLO/INT4 is triggered when the printer is turned off.
[image]

[image]

shawly, 2023-03-14:
Had this exact UVLO issue yesterday in the middle of a print, this happened after I upgraded to 3.12.2 and I also have a lot of stringing since the upgrade after the thermal calibration...

Prusa-Support, 2023-04-04:
Thanks for reaching out to us.

Did a _FW crash_ error occur?
If so, a dump should be automatically generated and available to be downloaded.
https://help.prusa3d.com/article/crash-dump_364959

It may be useful for our developers to understand the problem.

Michele Moramarco
Prusa Research

Rob70ar, 2023-04-24:
I’m having the same problem running 3.12.2 and 3.12.1. So far the workaround seems to use 3.11 😡😡😡

DidaJI, 2023-05-07:
I also have the same problem, I had to go back to version 3.11

ackerthehacker2, 2023-05-07:
Unfortunately, you can’t do that with a Revo nozzle.

Prusa-Support, 2023-05-19:
In case of _FW crash_ error please provide the dump file for our developers to move the investigation forward.
https://help.prusa3d.com/article/crash-dump_364959

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-23:
We couldn't confirm the presence of a bug.
The issue can be closed as stale now but feel free to comment below in case you have further feedback to share.

Michele Moramarco
Prusa Research

closed 2023-08-23 by Prusa-Support