prusa3d/Prusa-Firmware #4049: [BUG] Firmware 3.12.1 Thermal anomaly triggered during nozzle change

lubarb, 2023-02-23:
**Printer type** - MK3S
**Printer firmware version** - 3.12.1

**SD card or USB/Octoprint**
  Operation from the screen

**Describe the bug**
  Removing the nozzle triggers the thermal alarm and turns off the nozzle heating.
  
**To Reproduce**
  Heat the nozzle to 280. Attach a wrench to the heater block and try to loosen the nozzle. The same thing happens when trying to install the new nozzle.
  
**Expected behavior**
  There should be a way to follow the nozzle change instructions without the printer triggering an alarm and changing the temperature.

mix579, 2023-02-23:
Did you disable the thermal protection as per instructions in the 3.12 release notes:

Nozzle change menu

We added the necessary Nozzle change menu option under Settings -> HW Setup which disables the temp model temporarily. Please read [Changing or replacing the nozzle (MK2.5S/MK3S/MK3S+)](https://prusa.io/nozzle-mk3s) before you start. Please also keep in mind that changing the nozzle will need the first layer adjusted, as you will never be able to tighten the nozzle at exactly the same height as the previous one. We made this change thanks to the feedback from our community testers (see https://github.com/prusa3d/Prusa-Firmware/issues/3719).

lubarb, 2023-02-24:
Using prusaslicer 5, I don't believe I was prompted to update to the 3.12
series firmware, so I was running an 3.11 version of the firmware.

When I tried out the prusaslicer 6 beta, I was prompted to update to
3.12.1. When I updated to 3.12.1, I read the update for 3.12.1, but it
didn't occur to me to look further back.

So no, I hadn't read the 3.12.0 release notes. And I only consulted the
nozzle change procedures in order to confirm the 280 degree temperature. My
mistake.

mix579, 2023-02-24:
It was easy to miss. And in fact the new nozzle change procedure is everyone but great. Once it hit 280 the screen is filled with some silly message instead of showing the actual temperature. So if you use a wrench on the nozzle and heater block, and the temperature drops you don't really know when it's back to 280 to tighten the nozzle. You just have to wait a bit and hope it's back. I have opened a Github ticket and suggested to add the actual nozzle temperature to that screen.

mix579, 2023-02-24:
I can't speak for Prusa's official position on that but I have swapped nozzles several times now and not repeated the calibration, with no issues. It would be super inconvenient if required. I'm all for safety but this whole thermal protection model feels very half baked. Even worse on my Minis where I can't load filament after starting the printer without getting an error. I have to first preheat the nozzle manually to 170 or so. And then it will load fine. From then on no problems for days, loading and unloading are fine even from cold, until it turn it off. And on the Mini there's no way to disable it unlike the Mk3

knightsljx, 2023-02-26:
Use the "Nozzle Change" mode in the settings. If you manually heat the nozzle, it will trigger the warning. It's not a bug

closed 2023-02-26 by lubarb