prusa3d/Prusa-Firmware #4052: [BUG]Missing  "Resume Print" when Paused

ackerthehacker2, 2023-02-24:
MK3S+
FW 3.12.1

No MMU
Octoprint

**Describe the bug**
From the LCD screen, you can "Pause Print" but there is no "Resume Print"
  
**To Reproduce**
1. Start print
2. On LCD, Pause Print
3.   Try to Resume ..there is no Resume option
  
**Expected behavior**
Have a resume Option like previously

Prusa-Support, 2023-04-13:
Thanks for reaching us.

The standalone printer with FW 3.12.2 (most recent) behaves normally.
Please upgrade the firmware and try again.

Is it safe to say this is Octoprint related?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-03:
Is this issue still relevant?

As explained in https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-855201971, Octoprint could override some tasks which are normally attributed to the firmware.
Are you printing via Octoprint or via SD card?
Are you able to reproduce the problem in both ways?

Have you tested and compared older firmware versions?
I don't want the recent firmware adjustments that fixed issue https://github.com/prusa3d/Prusa-Firmware/issues/3323, to have broken something else related to it.
For now, though, this bug is not confirmed.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-24:
The bug is not confirmed and the issue has been mostly silent so far.
The issue will be closed as stale.
Please provide more feedback if the problem is still occurring, as we may consider re-opening the issue.

Michele Moramarco
Prusa Research

closed 2023-05-24 by Prusa-Support

spezifanta, 2024-01-14:
Just ran in this situation. Paused the print via LCD, but was missing the "Resume Printing" entry. 
As I started the print using OctoPrint, so it was possible to **resume the print from OctoPrint**.

[image]

Prusa-Support, 2024-02-17:
It seems that the feature was working fine so far, in fact it was broken and re-fixed on the next firmware which in is early testing state - yet to be posted (https://github.com/prusa3d/Prusa-Firmware/issues/4554). 😄 

I don't want this to be a matter of Octoprint configuration. 

> To make sure everything works correctly, you need to update the Octoprint profile with the following G-codes (in the GCODE Scripts section):
> 
> After print job is cancelled: M604
> After print job is paused: M601
> Before print job is resumed: M602

https://blog.prusa3d.com/prusa-autumn-news-summary-2023-new-firmware-prusaslicer-mk3-5-info-and-more_86935/

Michele Moramarco
Prusa Research

bwx, 2024-09-07:
> MK3S+ FW 3.12.1
> 
> No MMU Octoprint
> 
> **Describe the bug** From the LCD screen, you can "Pause Print" but there is no "Resume Print"
> 
> **To Reproduce**
> 
> 1. Start print
> 2. On LCD, Pause Print
> 3. Try to Resume ..there is no Resume option
> 
> **Expected behavior** Have a resume Option like previously

The same problem and FW 3.14.0-8066 - problems was't solved. I lost 2 hours poorly.

schluete, 2024-12-19:
+1, I paused the print from PrusaLink 0.7.2 (FW 3.14.0-8066) via the "Pause Print" button, but there's no way to resume the print, neither via PrusaLink (the button still says "Pause Print" and doesn't do anything when pressed again) nor the LCD (the only option there is to "stop print"). This is really unfortunate :/

svenstaro, 2024-12-28:
Can this be reopened? The issue is not fixed for me and other users.

3d-gussner, 2024-12-30:
Please update to FW 3.14.1 and PrusaLink 0.8.1 (calling it PL from here)

We introduced in FW 3.14.0 some new features which only work with PL 0.8.1.
Other host - printer interactions also have been improved and one of these is the Pause/Resume function.

Please keep the corresponding versions up to date:

- Printer FW 3.14.0 <---> PrusaLink 0.8.1
- Printer FW 3.14.1 <---> PrusaLink 0.8.1 or 0.8.2 (to be released soon)
- Printer with MMU 3.14.1 <---> PrusaLink 0.8.2 (to be released soon)

3d-gussner, 2024-12-30:
> Can this be reopened? The issue is not fixed for me and other users.

Are you using PrusaLink / PrusaConnect or Octoprint?

svenstaro, 2025-01-03:
It was indeed an error on my end. I'm using Octoprint and had to install [Activate Prusa HostTimer](https://plugins.octoprint.org/plugins/ActivatePrusaHostTimer/). This wasn't so clear to me reading just the firmware update logs. Sorry for the noise.

3d-gussner, 2025-01-03:
@svenstaro Thanks for the feedback and glad to hear it is now working for you.

> Sorry for the noise.

Don't be sorry, Release notes are often long and "easy" to "oversee" some notes. :wink:

svenstaro, 2025-01-03:
Actually, I just had a case where the printer was paused due to a thermal anomaly and it didn't allow me to offer to resume despite the named plugin being active. Is this perhaps a bug? The plugin is definitely active since I get the menu items that are produced by it (such as printer ready).

3d-gussner, 2025-01-03:
> Actually, I just had a case where the printer was paused due to a thermal anomaly and it didn't allow me to offer to resume despite the named plugin being active. Is this perhaps a bug? The plugin is definitely active since I get the menu items that are produced by it (such as printer ready).

Please enable in OctoPrint the serial logging (it can get big so clean it up from time to time or use it only when you want to trouble shoot something).
If the Thermal Model pauses then it probably was an error and you should investigate why that happened.
A TM Error needs to be cleared via the LCD menu, as it detected a **major** issue.

6ffm70, 2025-01-16:
To wrap this up with and to confirm the solution that worked for me:
i3 Mk3s with FW v3.14.1
PrusaSlicer 2.8.0
Octoprint 1.10.3 

When I put a pause marker into the gcode by slicer, the printer would not heat up after resume, with no option on the printer menu to heat up. Heating up before resume by Octoprint and resuming from the printer menu would result in the printer finishing without extruding filament after the pause.

Pausing the printer manually from the menu would give no option to resume in the printer menu.

**This fixed it:**
> To make sure everything works correctly, you need to update the Octoprint profile with the following G-codes (in the GCODE Scripts section):
> 
> After print job is cancelled: M604 
> After print job is paused: M601 
> Before print job is resumed: M602

I out-commented everything else that was written in the text boxes before. Pause & resume now work like they should; nozzle heats up and continues the print.

svenstaro, 2025-01-18:
> To wrap this up with and to confirm the solution that worked for me: i3 Mk3s with FW v3.14.1 PrusaSlicer 2.8.0 Octoprint 1.10.3
>
> When I put a pause marker into the gcode by slicer, the printer would [...]

Thanks for this! It should be noted that depending on the board, for cancelled it's either M604 or M603. See [here for Buddy](https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112) and [here for Einsy RAMBo](https://help.prusa3d.com/article/prusa-firmware-specific-g-code-commands_112173).