prusa3d/Prusa-Firmware #4056: Y axis crashes

jemarro12, 2023-02-28:
Printer shows random y axes crashes during print.
It can show up to 15 fails/hour.
With crash detection turned off any problem.

jemarro12, 2023-03-13:
Discarded any kinematic/lubrication issue, my problem is still there.
If crash detect disabled it prints flawlessly, during printing at the moment I turn on crash detection it starts triggering several crashes and autohome as it should do...
Any solve??
Any info u need???

ParCan, 2023-04-05:
Me as well.
Always at the same point in the bed travel.
Y axis detects a crash. Homes the nozzle and restarts where it left off
Firmware 3.12.2
Y stripped, Cleaned and lubricated. It's totally free with no tight spots or play
Self tests OK.
Belt tests OK.
[image]
No issues before the FW Upgrade.

Prusa-Support, 2023-04-08:
Thanks for reaching us.

Please allow me to close this issue as duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/2653.
You can monitor the other issue and take part in the discussion.

Michele Moramarco
Prusa Research

closed 2023-04-08 by Prusa-Support