prusa3d/Prusa-Firmware #4058: [BUG] Thermal Anomaly Warnings Caused by Part Geometry

glenray, 2023-03-01:
Printer type: MK3S+ i3
Printer firmware: 3.12.1
MMU: None
SD card and Octoprint

I get repeated thermal anomaly warnings when printing parts with similar characteristics. 
1. The AA battery box: https://files.printables.com/media/prints/249289/stls/2242031_aee8e12e-afe1-4d3c-8ef4-a8c96a99aa9f/aa-batteries.stl
2. This small parts drawer: https://files.printables.com/media/prints/265209/stls/2363476_617806f0-a693-489b-a15d-deb40e865c81/drawer_1x1x1_six-div_ramps.stl

Both parts are boxes with small internal compartments. My theory is that the part fan blows into the small compartments. That air has nowhere to go except straight up onto the hot end, creating additional hot end cooling not accounted for by the thermal model.

The battery box starts to issue warnings at about 30% completion. The part drawer warnings start at about 50% completion. The warnings grow more frequent as the height of the part grows until my wife threatens harm to both me and my printer!

Both models print fine after reverting back to firmware 3.11. Running firmware 3.12, all other parts print without warnings. 

It would be nice if 1) the warning would sound only once or 2) the sound could be turned off after the first warning, or 3) the thermal model could be turned off on a per-print basis when the part looks like it might mix up the thermal model.

Thanks for you work on this!

Regner53, 2023-03-27:
Yes!  I do not understand **WHY** you would go around the silent feature for this!  In my case the alarms are more frequent at the beginning of the print.  I will be reverting in the morning....

vzombie, 2023-04-02:
I think I've noticed the same thing as well. Never had any thermal anomaly issues before 3.12.x firmware. I have a stock MK3S+ which is relatively new (built in Jan 2023).

The first time it happened, I contacted Prusa support, who got me to check some things:

- visually inspect thermistor cable (looked good)
- do a preheat and wiggle thick cable bundle at print head to check for any fluctuations (there were none) 
- confirmed a thermal model calibration was done when updating to 3.12.1 (it was done)
- try jiggling different parts of cable bundle, near Einsy board, near print head; even bend the cable (no temp drops)
- open up Einsy case, unplug/replug thermistor cable (did that)
- upgrade to version 3.12.2 (did that)

I also did a maintenance and made sure to clean around the fans.

I printed several things since without issue, but today, I got the same error. It corresponded to temperature fluctuations which I believe were due to part geometry:

Here are some screenshots which show the temp chart and corresponding geometry: [https://imgur.com/a/GSArPCQ]:

1. Initial temp chart showing the dip I believe caused the issue.
2. Geometry corresponding to chart in #1.
3. Temp chart showing normalization as walls (and hotend) raise, changing the geometry.
4. Geometry corresponding to chart in #3.
5. Temp normalized within +/- 1°C (walls even higher)
6. Geometry corresponding to chart in #5.

Note that the THERMAL ANOMALY error remains on the screen for me for the entirety of the print once it is triggered, even after the anomaly disappears. Printing continues normally.

Prusa-Support, 2023-04-08:
Thanks for reaching us.

Would you mind giving it one more try with the most recent [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) release?

For a safer and more accurate thermal model calibration please do as follows.
1. Run a PID tuning at 230-250° (or the most frequently used print temperature).
2. Run an auto-home so that the print-head will be close to the bed.
This will increase thermal stress during the thermal model calibration.
3. Run the thermal model calibration again (calibration menu).
Please make sure that the printer won't be reached by airflows (fans, A.C., opened windows/doors...). 

If the above doesn't help, please provide clear close-up pictures of the heater-block block from below and the two sides. Make sure that the thermistor capsule and hear cartridge are visible from both sides.

Michele Moramarco
Prusa Research

glenray, 2023-04-11:
Michele, thanks for looking at this with me. Your procedure seems to have worked for me. I was able to print the battery box without any warnings.

The temperature fluctuated no more than +/-3 degrees. Before, it fluctuated by as much as 8 to 10 degrees either way.

I am relatively new to 3d printing and did not know about PID tuning. It is listed as "advanced" and "optional" in the manual, so I skipped right over it. My bad. For the benefit of future novices, PID tuning is designed to get hot end temperature fluctuations under control.

Many thanks for your help.
Glen Pritchard