prusa3d/Prusa-Firmware #4061: Unexpected filament change ( 3.12.2)

domverity, 2023-03-03:
MK3S, just updated to 3.12.2, running the attached GCODE, paused and behaved as if the GCODE instructed a filament change in the following approximate position (shown in Prusaslicer screenshot) during the first layer. I've never seen this behaviour before... I unloaded and reloaded the same filament and it continued as expected.

[image]
 
[Merged_0.6n_0.3mm_PLA_MK3S_3h57m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/10883138/Merged_0.6n_0.3mm_PLA_MK3S_3h57m.gcode.zip)

mattfox27, 2023-03-06:
Yep I'm getting the same bug as well, when using mmu about half of the filament changes pause and beep like m600 has been issued

3d-gussner, 2023-03-06:
@domverity @mattfox27 Thanks for reporting this issue.
@mattfox27 As you mention the MMU please describe it a bit more.

mattfox27, 2023-03-06:
When doing multi material prints with MMU2s running 1.06fw 3.12.2 prusa firmware, sliced using latest prusaslicer.  About every hour or so into a 8hr multi material print it will stop and issue a manual firmware switch where it beeps and the screen will say filament changed correctly?  Yes/no.  Upon pressing yes mmu switches to correct filament and print resumes

3d-gussner, 2023-03-06:
@domverity Do you have by change the serial logging enabled in Octoprint? If so please share your log file to hopefully see more.
If not, please enable in OctoPrint the serial logging see https://help.prusa3d.com/article/crash-dump_364959#octoprint

Also please fill in the issue template when opening an issue, this helps a lot.

Printer type - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
Printer firmware version - [e.g. 3.8.1, 3.8.1-RC1, ...]

MMU upgrade - [e.g. MMU2S, MMU2, MMU1]
MMU upgrade firmware version - [e.g. 1.0.6, 1.0.6-RC2, ...]

SD card or USB/Octoprint
Please let us know if you print via SD card or USB/Octoprint

Describe the bug
A clear and concise description of what the bug is.

To Reproduce
Please describe steps to reproduce the behavior.

Expected behavior
A clear and concise description of what you expected to happen.

G-code
Please attach a G-code. This will make it easier for us to replicate the error.

Video
Please attach a video. It usually helps to solve the problem.

mattfox27, 2023-03-08:
I just had this happen again, im not using octoprint is there any logging being done with just SD Card?  I suppose I can hookup octoprint to capture logs if needed

3d-gussner, 2023-03-08:
@mattfox27 No we don't have the "power" and resources to log to the SD card. Please try to log with a serial terminal like Octoprint or putty.

TwadaCZ, 2023-03-09:
Hi @3d-gussner ,
I report to Prusa helpdesk this issue from 3.12.1... I see this issue with 3.12.2 too... on my MK3s+ without MMU2...

Only if I print PETG, printer beep and want to uload and load filament... Randomly... Sometimes twice during 2 hours print... sometimes only during first 15 minutes... sometimes during last 15 minutes...

If I print PLA, there I did not see this issue with reload filament... during 20 hrs. prints...

I did not see it at my MK3s+ with MMU2, because there I did not print PETG...

When I downgrade to 3.11.0, everything working properly.

Thx, TwadaCZ

domverity, 2023-03-09:
@3d-gussner Apologies for not following the template, I will do so in future.

I re-ran the gcode where I saw the issue with serial logging enabled in Octoprint but unfortunately the issue did not happen. I will continue printing with serial logging on and post a log if I catch it happening.

mattfox27, 2023-03-10:
I was running octoprint with serial logging enabled and I caught it at about 21:37 is when it happened

`2023-03-09 21:35:26,147 - Recv: T:205.1 /205.0 B:60.0 /60.0 T0:205.1 /205.0 @:63 B@:35 P:0.0 A:42.0
2023-03-09 21:35:28,043 - Recv: ok
2023-03-09 21:35:28,580 - Send: N333335 T1*13
2023-03-09 21:35:28,585 - Recv: T:205.1 /205.0 B:60.0 /60.0 T0:205.1 /205.0 @:62 B@:32 P:0.0 A:42.0
2023-03-09 21:35:28,683 - Recv: mmu_get_response - begin move: T-code
2023-03-09 21:35:28,711 - Recv: MMU <= 'T1'
2023-03-09 21:35:28,734 - Recv: Unloading finished 2
2023-03-09 21:35:30,136 - Recv: T:205.0 /205.0 B:60.0 /60.0 T0:205.0 /205.0 @:64 B@:31 P:0.0 A:42.0
2023-03-09 21:35:30,763 - Recv: mmu_get_response - begin move: load
2023-03-09 21:35:32,144 - Recv: T:205.0 /205.0 B:60.0 /60.0 T0:205.0 /205.0 @:65 B@:31 P:0.0 A:41.9
2023-03-09 21:35:34,151 - Recv: T:205.0 /205.0 B:60.0 /60.0 T0:205.0 /205.0 @:64 B@:24 P:0.0 A:41.9
2023-03-09 21:35:36,162 - Recv: T:205.1 /205.0 B:60.0 /60.0 T0:205.1 /205.0 @:62 B@:22 P:0.0 A:42.0
2023-03-09 21:35:37,600 - Recv: echo:busy: processing
2023-03-09 21:35:38,164 - Recv: T:205.1 /205.0 B:60.1 /60.0 T0:205.1 /205.0 @:63 B@:12 P:0.0 A:42.1
2023-03-09 21:35:40,168 - Recv: T:205.2 /205.0 B:60.1 /60.0 T0:205.2 /205.0 @:62 B@:23 P:0.0 A:41.9
2023-03-09 21:35:42,175 - Recv: T:205.0 /205.0 B:60.1 /60.0 T0:205.0 /205.0 @:65 B@:13 P:0.0 A:42.0
2023-03-09 21:35:44,178 - Recv: T:205.0 /205.0 B:60.1 /60.0 T0:205.0 /205.0 @:65 B@:19 P:0.0 A:41.8
2023-03-09 21:35:46,186 - Recv: T:205.2 /205.0 B:60.1 /60.0 T0:205.2 /205.0 @:61 B@:19 P:0.0 A:41.8
2023-03-09 21:35:46,603 - Recv: echo:busy: processing
2023-03-09 21:35:48,193 - Recv: T:205.2 /205.0 B:60.1 /60.0 T0:205.2 /205.0 @:62 B@:22 P:0.0 A:41.8
2023-03-09 21:35:49,204 - Recv: MMU <= 'A'
2023-03-09 21:35:49,798 - Recv: MMU => 'ok'
2023-03-09 21:35:49,851 - Recv: mmu_get_response() returning: 1
2023-03-09 21:35:50,197 - Recv: T:205.0 /205.0 B:60.0 /60.0 T0:205.0 /205.0 @:65 B@:32 P:0.0 A:41.9
2023-03-09 21:35:52,202 - Recv: T:205.1 /205.0 B:60.0 /60.0 T0:205.1 /205.0 @:62 B@:30 P:0.0 A:41.8
2023-03-09 21:35:54,206 - Recv: T:205.0 /205.0 B:60.0 /60.0 T0:205.0 /205.0 @:65 B@:31 P:0.0 A:41.9
2023-03-09 21:35:55,602 - Recv: echo:busy: processing
2023-03-09 21:35:55,767 - Recv: MMU can_load:
2023-03-09 21:35:56,214 - Recv: OOOOOOOOOOOOOOOOOOOOOOT:205.2 /205.0 B:60.1 /60.0 T0:205.2 /205.0 @:62 B@:19 P:0.0 A:41.9
2023-03-09 21:35:56,351 - Recv: OOOOOOOO succeeded.
2023-03-09 21:35:56,367 - Recv: ok
2023-03-09 21:35:56,398 - Send: N333336 G1 F10800*66
2023-03-09 21:35:56,430 - Recv: ok
2023-03-09 21:35:56,462 - Send: N333337 G1  X184.781 Y132.975*53
2023-03-09 21:35:56,481 - Recv: ok
2023-03-09 21:35:56,522 - Send: N333338 M900 K0.05*97
2023-03-09 21:35:56,540 - Recv: LA10C: Linear Advance mode: 1.5
2023-03-09 21:35:56,548 - Recv: echo:Advance K=0.05
2023-03-09 21:35:56,557 - Recv: ok
2023-03-09 21:35:56,573 - Send: N333339 M900 K30*120
2023-03-09 21:35:56,589 - Recv: K out of allowed range!
2023-03-09 21:35:56,599 - Recv: echo:Advance K=0.05
2023-03-09 21:35:56,607 - Recv: ok
2023-03-09 21:35:56,631 - Send: N333340 G4 S0*90
2023-03-09 21:35:56,655 - Recv: ok
2023-03-09 21:35:56,708 - Send: N333341 G1 E12.0000 F1140*55
2023-03-09 21:35:56,731 - Recv: ok
2023-03-09 21:35:56,756 - Send: N333342 G1  X235.336   E42.0000 F1011*64
2023-03-09 21:35:56,786 - Recv: ok
2023-03-09 21:35:56,803 - Send: N333343 G1  X184.781   E6.0000 F708*78
2023-03-09 21:35:56,826 - Recv: ok
2023-03-09 21:35:56,843 - Send: N333344 G1  Y132.475*105
2023-03-09 21:35:56,888 - Recv: ok
2023-03-09 21:35:56,906 - Send: N333345 G1  X176.961  E0.2972 F1584*90
2023-03-09 21:35:56,926 - Recv: ok
2023-03-09 21:35:56,943 - Send: N333346 G1  X176.211 Y132.975   E0.0343*77
2023-03-09 21:35:56,962 - Recv: ok
2023-03-09 21:35:56,977 - Send: N333347 G1  X234.961   E2.2329 F1800*122
2023-03-09 21:35:57,005 - Recv: ok
2023-03-09 21:35:57,020 - Send: N333348 G1  X235.711 Y133.475   E0.0343*78
2023-03-09 21:35:57,040 - Recv: ok
2023-03-09 21:35:57,055 - Send: N333349 G1  X176.961   E2.2329 F2198*122
2023-03-09 21:35:57,080 - Recv: ok
2023-03-09 21:35:57,096 - Send: N333350 G1  Y133.975  E0.0190*51
2023-03-09 21:35:57,118 - Recv: ok
2023-03-09 21:35:57,133 - Send: N333351 G1  X234.961   E2.2044 F4200*122
2023-03-09 21:35:57,335 - Recv: ok
2023-03-09 21:35:57,352 - Send: N333352 G1  Y134.475  E0.0190*59
[... 100 lines cut ...]
2023-03-09 21:37:42,656 - Recv: T:204.2 /205.0 B:60.0 /60.0 T0:204.2 /205.0 @:66 B@:34 P:0.0 A:41.5
2023-03-09 21:37:44,659 - Recv: T:203.4 /205.0 B:60.0 /60.0 T0:203.4 /205.0 @:76 B@:28 P:0.0 A:41.3
2023-03-09 21:37:46,666 - Recv: T:203.7 /205.0 B:59.9 /60.0 T0:203.7 /205.0 @:71 B@:38 P:0.0 A:41.4
2023-03-09 21:37:47,534 - Recv: echo:busy: paused for user
2023-03-09 21:37:48,669 - Recv: T:203.8 /205.0 B:59.9 /60.0 T0:203.8 /205.0 @:69 B@:33 P:0.0 A:41.4
2023-03-09 21:37:50,676 - Recv: T:204.5 /205.0 B:59.9 /60.0 T0:204.5 /205.0 @:59 B@:31 P:0.0 A:41.4
2023-03-09 21:37:52,679 - Recv: T:205.1 /205.0 B:60.0 /60.0 T0:205.1 /205.0 @:50 B@:20 P:0.0 A:41.4
2023-03-09 21:37:54,686 - Recv: T:206.0 /205.0 B:60.0 /60.0 T0:206.0 /205.0 @:39 B@:27 P:0.0 A:41.4
2023-03-09 21:37:56,537 - Recv: echo:busy: paused for user
2023-03-09 21:37:56,689 - Recv: T:206.1 /205.0 B:59.9 /60.0 T0:206.1 /205.0 @:40 B@:30 P:0.0 A:41.3
2023-03-09 21:37:58,704 - Recv: T:206.2 /205.0 B:59.9 /60.0 T0:206.2 /205.0 @:41 B@:32 P:0.0 A:41.4
[... cut ...]
2023-03-09 21:38:12,742 - Recv: T:205.0 /205.0 B:60.0 /60.0 T0:205.0 /205.0 @:55 B@:23 P:0.0 A:41.4
2023-03-09 21:38:14,541 - Recv: echo:busy: paused for user
2023-03-09 21:38:14,747 - Recv: T:205.3 /205.0 B:60.0 /60.0 T0:205.3 /205.0 @:51 B@:20 P:0.0 A:41.3
2023-03-09 21:38:16,740 - Recv: T:205.5 /205.0 B:60.0 /60.0 T0:205.5 /205.0 @:48 B@:32 P:0.0 A:41.2
2023-03-09 21:38:18,770 - Recv: T:205.8 /205.0 B:60.0 /60.0 T0:205.8 /205.0 @:44 B@:34 P:0.0 A:41.3
2023-03-09 21:38:20,758 - Recv: T:205.6 /205.0 B:59.9 /60.0 T0:205.6 /205.0 @:48 B@:49 P:0.0 A:41.2
`

mattfox27, 2023-03-10:
> 2023-03-09 21:36:08,646 - Recv: echo:Enqueing to the front: "M600"
> 2023-03-09 21:36:08,656 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 333348
> 2023-03-09 21:36:08,668 - Recv: Resend: 333349

I posted a log but this might be the error causing it???

schiele, 2023-03-14:
I observe the same problem as described here: During a PETG print of a gcode file from SD card with 2 intentional M600 color change commands included the printer starts behaving as if seen a M600 command in various different places, even long before the first of those commands even appears in the file. If you inspect the gcode file you will find that the first M600 command appears on layer Z=1.8. Still, the printer starts initiating manual filament changes towards the end of the first layer Z=0.2 in many places, typically after one perimeter line was completed before starting the next one despite the gcode file not including any command I would associate with such behavior.

The printer I reproduced this: MK3S+ with MMU2S, though I disconnected the MMU2S from the power source and fed the filament directly into the extruder, such that from a firmware point of view this is a pure MK3S+. Latest firmware (3.12.2) is installed. Printer is running in Stealth Mode.

Printing done from SD card.

gcode file: [protractor_upper_part_depressed_0.2mm_PETG_MK3S_2h14m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/10973773/protractor_upper_part_depressed_0.2mm_PETG_MK3S_2h14m.gcode.txt)

Steps to reproduce: Print the attached gcode file from SD card.

Expected behavior: Printer should only initiate manual filament changes when it hits a M600 command.
Observed behavior: Printer initiates lots of manual filament changes for no obvious reason.

gudnimg, 2023-03-14:
@mattfox27 sorry for the late response, If you see this in the printer's serial log `Enqueing to the front: "M600"` then that  confirms the issue is filament runout. The fsensor reports to the firmware it no longer detects filament so the printer will try to recover the print by injecting M600 into the front of the queue.

@schiele without a log I can only assume you're seeing filament runout too. It's the only way the firmware will inject M600 command. Please check the logs from the printer or check if the filament sensor or FINDA is no longer detecting filament.

Note that when using an MMU, the FINDA sensor can also cause filament runout events. If the FINDA LED is illuminating red that means it doesnt detect filament.

I have seen issues with FINDA before when the filament is bent in a certain way, should be easily fixed by slightly tuning the FINDA position.

PyBonnetainNesterenko, 2023-04-06:
I enabled the serial.log on OctoPrint, ran a print, and here are the lines surronding the M600.
```
2023-04-05 23:15:34,324 - Recv: T:215.2 /215.0 B:61.4 /60.0 T0:215.2 /215.0 @:31 B@:0 P:0.0 A:33.6
2023-04-05 23:15:35,298 - Recv: echo:busy: processing
2023-04-05 23:15:36,327 - Recv: T:215.1 /215.0 B:61.6 /60.0 T0:215.1 /215.0 @:34 B@:0 P:0.0 A:33.3
2023-04-05 23:15:37,302 - Recv: echo:busy: processing
2023-04-05 23:15:38,334 - Recv: T:215.2 /215.0 B:61.7 /60.0 T0:215.2 /215.0 @:32 B@:0 P:0.0 A:33.3
2023-04-05 23:15:39,300 - Recv: echo:busy: processing
2023-04-05 23:15:40,337 - Recv: T:215.2 /215.0 B:61.7 /60.0 T0:215.2 /215.0 @:31 B@:0 P:0.0 A:33.4
2023-04-05 23:15:41,300 - Recv: echo:busy: processing
2023-04-05 23:15:41,680 - Recv: LCD status changed
2023-04-05 23:15:41,682 - Recv: ok
2023-04-05 23:15:41,696 - Send: N21 G1 Z0.2 F720*30
2023-04-05 23:15:41,738 - Recv: ok
2023-04-05 23:15:41,751 - Send: N22 G1 Y-3 F1000*24
2023-04-05 23:15:41,758 - Recv: ok
2023-04-05 23:15:41,768 - Send: N23 G92 E0*118
2023-04-05 23:15:41,775 - Recv: ok
2023-04-05 23:15:41,788 - Send: N24 G1 X60 E9 F1000*91
2023-04-05 23:15:41,799 - Recv: ok
2023-04-05 23:15:41,809 - Recv: fsensor_checkpoint_print
2023-04-05 23:15:41,813 - Send: N25 G1 X100 E12.5 F1000*76
2023-04-05 23:15:41,817 - Recv: Resend: 22
2023-04-05 23:15:41,829 - Recv: ok
2023-04-05 23:15:41,831 - Recv: LCD status changed
2023-04-05 23:15:41,833 - Recv: fsensor_update - M600
2023-04-05 23:15:41,837 - Send: N22 G1 Y-3 F1000*24
2023-04-05 23:15:41,835 - Recv: echo:Enqueing to the front: "M600"
2023-04-05 23:15:41,841 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 21
2023-04-05 23:15:41,845 - Recv: Resend: 22
2023-04-05 23:15:41,853 - Recv: ok
2023-04-05 23:15:41,857 - Send: N23 G92 E0*118
2023-04-05 23:15:42,343 - Recv: T:215.2 /215.0 B:61.9 /60.0 T0:215.2 /215.0 @:30 B@:0 P:0.0 A:33.4
2023-04-05 23:15:43,819 - Recv: echo:busy: processing
2023-04-05 23:15:44,347 - Recv: T:214.9 /215.0 B:61.9 /60.0 T0:214.9 /215.0 @:34 B@:0 P:0.0 A:33.2
2023-04-05 23:15:45,821 - Recv: echo:busy: processing
2023-04-05 23:15:46,354 - Recv: T:215.1 /215.0 B:61.8 /60.0 T0:215.1 /215.0 @:32 B@:0 P:0.0 A:33.3
2023-04-05 23:15:47,821 - Recv: echo:busy: processing
```
So there is indeed (as far as I understand it) a filament runout wrongly detected at 2023-04-05 23:15:41,833.

Prusa-Support, 2023-04-11:
Again we can't exclude a temporary filament outage detection, due to various reasons - some of which are mentioned in https://github.com/prusa3d/Prusa-Firmware/issues/3872#issuecomment-1413930200.

In the case of MMU, there are two filament sensors that need to be calibrated to be sensitive to filament outage but forgiving for little filament inconsistencies (deformed/undersized filament).

These kinds of problems may not be firmware related and could persist on older and newer firmware versions.
We will keep investigating and collecting feedback even after the new firmware release.
Until then, please make sure that your filament is regular, and that the filament sensor triggering mechanism is clean and calibrated adequately. Proper troubleshooting may solve the problem.

Michele Moramarco
Prusa Research

TwadaCZ, 2023-05-16:
Yes, I reported it to chat support immediately after the release 3.12.0... and here (was closed) with 3.12.2 release...

I have same problem on my MK3s (without MMU!!!)... only with PETG print...

I can reproduce this issue alway when print PETG with any 3.12 firmware... every print.

If I downgrade to 3.11.x firmware and print on same printer, same material, same GCODE file... this work perfect and problem does not appear.

Prusa-Support, 2023-06-15:
Would you guys mind if this issue is closed as duplicate of #3872?

This one wasn't closed because it got escalated quite quickly, in parallel with #3872. However neither of the issues show significant progress (just yet).
I suggest closing this one because opened at a later time, and I encourage all users here to join forces in the other issue.

Michele Moramarco
Prusa Research

Michele Moramarco
Prusa Research

domverity, 2023-06-15:
@Prusa-Support fine by me...

Prusa-Support, 2023-06-28:
Please continue reporting on issue https://github.com/prusa3d/Prusa-Firmware/issues/3872.
This one will be closed as a duplicate now. Thanks.

Michele Moramarco
Prusa Research

closed 2023-06-28 by Prusa-Support