prusa3d/Prusa-Firmware #4066: [QUESTION]Can I unload filament after detect under temperature?

ShojiMiyanishi, 2023-03-06:
After print. Air conditioner will stop and temperature were down.
Printer detect under temperature then I can not do any thing.
I hope unload filament and store in airtight case.
I cant do that 3.9.x.

3d-gussner, 2023-03-07:
@ShojiMiyanishi Thanks, and sorry to hear that you are blocked by MINTEMP. At this moment we don't have a solution to heat the printer in this condition.
Sounds silly, but please try to heat the hotend and bed thermistor with a hair dryer to overcome the MINTEMP.

I know it is not a solution but maybe a work around.

Which temperature does your printer shows? Which printer do you have?

Prusa-Support, 2023-04-28:
Hi, we haven't heard back from you, and the conditions that apply to the issue are not very clear.

This issue will be closed as stale and invalid for now.
Regardless of this, feel free to provide further clarifications and relevant feedback.

Michele Moramarco
Prusa Research

closed 2023-04-28 by Prusa-Support