prusa3d/Prusa-Firmware #4075: [BUG]<Firmware 3.12.2 still with Wizard Calibration problem at Hotend>

josepetridelima, 2023-03-11:
**Printer type** MK3S
**Printer firmware version** -  3.12.2

**Describe the bug**
  When we try to run Wizard Calibration it stops at hotend warm test, but the sensor and heater it is works fine. I can print, but fail at wizard. 
  
**To Reproduce**
  Try many times, PID calibration and Wizard calibration, and fail. But Bed level calibration works fine, it can also print without problems. Rom temperature 25ºC.
  
**Expected behavior**
  More users find this problem in the earlier firmware versions, some people say about time limits about hottend, and at some firmware it works others not.

AndWik, 2023-03-11:
I have 3pc Prusa 3S. All equipped with Revo. Upgraded one of the to 3.12.2.
The temp calibration stops after 1 sec!!! Tried if from Selt test and stand-alone when the nozzle, bed and room temperature were all at 19 degrees celcius.
Wouldn't it be better to "learn" ramp-up behavior. If the ramp-up has a normal shape but is too fast it should learn from that and react if the behavior changes.

My 5c

Mr Anders
Sweden

3d-gussner, 2023-03-12:
@AndWik The REVO isn't supported during Wizard at this moment. Please read and try https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1451733927 and comment there your results
@josepetridelima Do you have a stock printer or any modifications like the REVO?

lijxisomer, 2023-03-13:
Prusa really needs to disable that thermal model until it's properly implemented. why can't it work just like PID tuning. just run a calibration and fit the model to the calibration. it's confusing the hell out of everyone. it has no business being implemented in production firmware

knightsljx, 2023-03-13:
Prusa really needs to disable that thermal model until it's properly implemented. why can't it work just like PID tuning. just run a calibration and fit the model to the calibration. it's confusing the hell out of everyone. it has no business being implemented in production firmware

josepetridelima, 2023-03-13:
> @AndWik The REVO isn't supported during Wizard at this moment. Please read and try [#3636 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1451733927) and comment there your results @josepetridelima Do you have a stock printer or any modifications like the REVO?

Hi, I have the Trianglelab v6 hotend. It's works fine for two years (many times I did Wizard Calibration successfully), but now starts to fail during the wizard calibration process. I still print fine with no problems, But wizard mode and PID didn't pass to hotend test. Also, it starts to beep annoyingly when starts to warm up before starts to print. Since now I would like to thank you for your attention to this problem.

3d-gussner, 2023-03-14:
@josepetridelima Thanks, now I see why you have issues.
The Thermal Model is tailored around the E3D v6 which is used by Prusa.
So changing to another hotend or setup may cause issues.
Hope you understand that we focus on the hardware, electronics and parts the original Prusa printers use and can't test every possible combination users may implement/modify.

Please run following steps.
Keep in mind that the LCD Calibration -> Thermal model cal. is for E3Dv6. Also the wizard will use the values for the E3D v6.

- connect via USB with the printer with a serial terminal software (don't use Pronterface please) Putty is a good option, see [here](https://help.prusa3d.com/article/crash-dump_364959#putty) how to setup putty correctly.
  - Be aware connecting via USB will reset your printer, so do that only while it is idle.
- `M503` to report saved values
- `M310 S0` to temporary disable thermal model for PID tuning
- LCD -> Calibration -> PID calibration
- `M310 S1`
- Wait until the printer is back cold
- `M310 A230 F0` to run the thermal model calibration without using existing values (default are for normal E3D v6)
- `M500` to store the values in eeprom
- `M503` to report new values

This should fix the thermal model warning beeps.

Would be great if you could share the `M310` values for other Trianglelab v6 hotend users.
If you want you can test the thermal model by blowing some cold air at the heat block during heat up.

For more information please read https://blog.prusa3d.com/mk3s-3-12-beta-firmware-new-thermal-model-protection-and-blob-detection_71230/ 

Hope that helps.

jagerman, 2023-03-15:
> can't test every possible combination users may implement/modify.

Do you test *any* popular changes that Prusa users make?  A Revo is a huge quality of life improvement to a Prusa i3 and is a wildly popular replacement hotend.  We users certainly don't expect you to test "every possible combination" but does Prusa consider it unreasonable to test *any* of the most popular modifications when making such a significant firmware change?

3d-gussner, 2023-03-15:
@jagerman Hi,
From the beginning of the development of the thermal model we tried to make it as versatile and customizable as possible for our modding community. Every TM related setting can be modified via `M310` gcode.

We also point out in the [Release Notes](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) that REVO needs more attention.

>Update on E3D Revo (MK3/MK3S/MK3S+)
We would like to thank all of the community members who provided their TM calibration logs with the E3D REVO! This really helped to improve the TM development for PTC/REVO hotends.
E3D and Prusa Research have also started a task force to work on the Thermal Model for PTC/Revo hotends.

The community started a github [REVO issue ](https://github.com/prusa3d/Prusa-Firmware/issues/3636) where we now try to collect data from the community to be able to implement some "default" REVO values as we have for the E3D v6.
So if you have a REVO and want to see it working please contribute by following the [steps](https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1451733927)  and provide your logs.

We need at least 50 independent REVO data log files to be able to continue with it.

I know finding things in github isn't easy but please try to search for existing issues/reports before commenting on a topic that is already discussed.

Looking forward for your log files and :crossed_fingers: we get sooner than later a reasonable amount of logs to continue.

Traumlewis, 2023-04-18:
I am using a Rampen CHC hotend and have the same problem. I've turned thermal mode off completely for now, with non-standard hotend there are constant problems.

1.`M310 S0` to disable thermal model for PID tuning
2. LCD -> Calibration -> PID calibration
3. Wait until the printer is back cold
4.`M310 S0` To make sure thermal Mode is really turned off
5.`M500` to store the values in eeprom

The same with Prusa Mini where you can't even calibrate PID yourself. Because Prusa thinks it's not necessary, yes if I leave everything as it is. And no, you cannot pid calibrate the Prusa Mini, the settings will not be applied no matter what you do. you could make your own firmware, but that's a waste of time if certain functions are already in there and disabled. All of this has less and less to do with opensource. 😒

Prusa-Support, 2023-04-28:
Thanks for your feedback.

We got several similar issues reported.
We are putting together all Revo-related cases in one unique issue because it is the most popular alternative hotend.

It may be helpful to have separate issues for different hotend/hardware.
When looking for existing issues, possibly try to include the brand/type of your hotend/hardware among the search keywords. You may find an existing issue where users are collecting information (and logs) about the same hotend/hardware.

About non-stock parts, you can find generic suggestions at https://github.com/prusa3d/Prusa-Firmware/issues/4015#issuecomment-1481686416.
Long story short, you may need to edit a few parameters, or **provide a log** - for the time being, you can disable the thermal anomaly or revert to Prusa stock parts.

.

Features for Prusa MINI are off-topic.
If you are interested in PID tuning please check the relevant issue in the specific repository https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2401.

Michele Moramarco
Prusa Research

closed 2023-11-21 by github-actions[bot]