prusa3d/Prusa-Firmware #4078: [QUESTION]<Why does Z axis calibration invoked by MBL require user input >

vkojchevski, 2023-03-12:
When MBL (G80) is invoked and it detects an uneven X axis, the printer is paused and waits for user input (knob press) before it moves to calibrate the Z axis. While I am pleased to see that it detects and tries to correct the X axis skew by tramming it (calibrating Z axis), I am not sure why it has to get an user input to continue.
This results in a paused printer with both heated elements on and if the user doesn't notice that in time, I assume it will (hopefully) fall under thermal protection timers and turn off the printer, resulting in a, technically, failed print. Besides resetting the printer, you have no choice but to accept the calibration, so I think that the user input requirement is redundant and adds unnecessary complexity.

My question/enhancement would be, is it necessary to have user input when tramming the X axis?
Unless I am overseeing some important reason against, I suggest that at the very least the pause have a beep warning similar to filament change M600 has, or even better, just remove the requirement of user input as the printer is already in "active" state and the user should have done all the preliminary steps (clean bed, ensure no obstacles are in the way of the kinematics... etc).

leptun, 2023-03-12:
The "Some problem encountered, Z-leveling enforced ..." screen is automatically dismissed after 30 seconds even if the user doesn't interact with it. I think it's there just so the user knows what is happening with the printer and nothing else. This applies only to the MK3/S/+. 30 seconds of heaters being enabled isn't really a problem imo.
For MK2.5/S, the message isn't dismissed automatically since the user must also be there to align the Z axis at the top (it can't be done automatically).

vkojchevski, 2023-03-12:
I admit, I didn't wait for too long, so I didn't notice the 30 seconds timer. After that timer it proceeds with the print?

leptun, 2023-03-12:
Yes. On MK3S, it just waits 30 seconds (unless clicked earlier), does the calibration automatically and restarts the MBL process and then the print continues.

vkojchevski, 2023-03-12:
Then I was mistaken, my apologies and thank you for your time to clear things up.

closed 2023-03-12 by vkojchevski