prusa3d/Prusa-Firmware #4079: Firmware Crash

Gabri988, 2023-03-13:
### Description of the bug

I was printing a pla keychain made up of two colors (white and red) everything went well until during a filament change the printer stopped due to an error loading mmu filament. When I realized it, I clicked the knob on the lcd to reset the temperature and then try again to load the filament with mmu. Unfortunately just when mmu was loading to "step" the printer reported the firmware crash error.

Now I'm printing again to try to better understand why I wasn't in front of the printer before when this happened.
If in case I have the same problem again, I will attach log files and photos.

p.s. I use the original printer (unmodified) with prusalink

EDIT **Same file printed with PrusaSlicer 2.5.0 gave no problem.**

PrusaSlicer version 2.6.0 Alpha 5
Printer Original Prusa i3 MK3S+ fw v3.12.2 and MMU2S fw v1.0.6 
Raspberry PI Zero W2 with PrusaLink v.0.7.0rc2

### Project file & How to reproduce

[TYPE1.zip](https://github.com/prusa3d/PrusaSlicer/files/10949000/TYPE1.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0 Alpha 5

### Operating system

Windows 10 Pro 64bit

### Printer model

Original Prusa i3 MK3S+

Prusa-Support, 2023-04-28:
Thanks for reaching us.

We may be looking at board/memory/firmware corruption.

Have you noticed any missing options or lost value/settings in your printer menus?

If the problem is recurrent, there is a chance that a factory reset and a new firmware installation would solve the problem once and for all. https://help.prusa3d.com/article/full-system-refresh-original-prusa-i3_133258
If the firmware crash keeps happening though, our developers would appreciate getting a few dump files for analysis.

Dumps are automatically generated after a "firmware crash".
In the case a dump file is available, the user is warned after the printer startup. What to do?
1. Download the dump file (keep it on your computer, it may be useful later). The dump file (and warning message) will disappear after downloading. You can download it as per this article. https://help.prusa3d.com/article/crash-dump_364959
2. Possibly save a text file together with it and add relevant notes on it: FW version, date/time, how/when the FW crash occurred (last operation), personal notes...
3. If you can reproducible the problem, or it accidentally occurs multiple times, you will have multiple dump files and details for our developers to analyze.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-19:
Is this issue still relevant?
Please close the issue if the problem doesn't persist, otherwise, we may close it as stale.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-08:
I hope that one of the methods above was of help. We can't exclude temporary glitches either.
This issue will be closed as stale now, further feedback would be appreciated though.

Michele Moramarco
Prusa Research

closed 2023-06-08 by Prusa-Support