prusa3d/Prusa-Firmware #4083: [ENHANCEMENT] Move printer head to top on filament runout.

MartijnGevaert, 2023-03-14:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [e.g. MK3S]

**Is your enhancement related to a problem? Please describe.**
When the printer runs out of filament, the head moves to the side, the cable coming from the back of the extruder can then hit large pieces being printed, potentially knocking them out of place.
You can see this in the image below.
[image]

  
**Describe the solution you'd like**
When the filament runs out, move the head to the top before moving it to the side, this will reduce the likelihood of this happening.

3d-gussner, 2023-03-15:
@MartijnGevaert Thanks for the request. we gonna look into that. BTW great picture to show your issue and concern about knocking off a print.

Prusa-Support, 2023-04-28:
Thank you for your suggestion.

Probably parking the print-head to the left side will help too.
However, a well-tensed extruder cable should reduce risks of collision.

Michele Moramarco
Prusa Research

closed 2023-11-14 by github-actions[bot]