prusa3d/Prusa-Firmware #4084: [BUG] FW 3.12.2 Steppers are not disabled after belt test

vadim-ryazanov, 2023-03-15:
MK3S+
FW 3.12.2

New firmware fails to disable steppers after a belt test. Older firmware used to disable steppers in a few seconds after belt test completed.
  
**To Reproduce**
  Run belt test. X and Y carriages are locked after belt test is completed. Adds unnecessary step of going into menu to disable steppers.
  
**Expected behavior**
  Steppers disabled after belt test complete

Prusa-Support, 2023-04-29:
Thanks for your comment.

I believe the stepper motors are only idle at the startup (before they get engaged for an operation).
Once engaged, they remain enabled and hold in position for the whole time.
The motors will be disabled again only after rebooting or via the specific option in _settings_.
Not sure if there are exceptions, I can't think of any at the moment.

This is not a bug but the expected behavior.
If you need to slide the axis by hand, the quickest way is rebooting via _x-reset_ button.
I can't see common use cases where disabling the steppers (without resetting) after the belt test is necessary, and it shouldn't be the expected behavior.

Can this issue be closed?

Michele Moramarco
Prusa Research

vadim-ryazanov, 2023-05-01:
Hi Michele,
I see. In my case this was during maintenance (to which I think bet test belongs), and I tend to move carriages around by hand for inspection etc. So, locked axes are a bit of inconvenience :)
Please feel free to close the issue then.
Regards,
Vadim.

closed 2023-11-14 by github-actions[bot]