prusa3d/Prusa-Firmware #4087: [ENHANCEMENT]Correctly respect M109/M190 arguments

WereCatf, 2023-03-15:
Currently on the MK2* and MK3* firmware, the gcode _M109_ behaves as if you were always using the _R_ - argument to set the temperature, ie. it waits for the hotend regardless, whether it is higher or lower than the requested temperature. If using the _S_ - argument, it's supposed to only wait for the hotend, if the temperature is lower than the requested temperature, but proceed immediately, if it's equal higher.

In other words, e.g. _M109 S170_ behaves the same as if you were using _M109 R170_.

The Prusa Mini* correctly behaves differently for the two different forms of the argument.

3d-gussner, 2023-03-15:
@WereCatf Thanks for the request but it is already in the queue FW 3.13.0-RC2 see https://github.com/prusa3d/Prusa-Firmware/pull/2060

Please consider to close this issue.

WereCatf, 2023-03-15:
> @WereCatf Thanks for the request but it is already in the queue FW 3.13.0-RC2 see #2060
> 
> Please consider to close this issue.

Ah, my bad!

closed 2023-03-15 by WereCatf