prusa3d/Prusa-Firmware #4088: [ENHANCEMENT] Turn off the stepper motors after loading the filament

Pitkox, 2023-03-15:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3S

Is your feature request related to a problem? Please describe.
After loading new filament I need to shut down stepper motors from menu to manually pull up filament from extruder a little bit, to prevent licking when it's hot.

Describe the solution you'd like
Auto shut down motors after loading filament, or pull it a little bit up automatically after correct loading to prevent from licking out of the extruder.

3d-gussner, 2023-03-15:
@Pitkox Thanks for your idea, but this would cause other issues as under extrusion and probably other issues.
This is why I don't see that we gonna add this request to the firmware.
 
Did you try to lower the temp or even use Settings -> Move axis -> Extruder to remove pressure and prevent oozing.
Please also check if the filament temperature isn't to high for that specific material.

Please consider to close this request/issue to reduce the amount of open tickets.

Pitkox, 2023-03-15:
No it's not, what under extrusion? Before every print extruder make preline, am I right? So there is no way to talk about under extrusion, oh maybe only if u talk about that preline... After I pull filemanet a little bit up manualy indeed the preline is about half way shorter, but it can be hard coded longer too. Filament temperature is normal, get from slicer presets and it's good for printing quality.
Using move axis -> extruder is more complicated as settings-> motors off...
What about only turning motors off, this causing problems too?

3d-gussner, 2023-03-16:
@Pitkox The load filament function is also called when you have a filament run out or filament change and there you want to have the filament loaded correctly and "ready" to extrude.

3d-gussner, 2023-03-16:
@Pitkox I can understand that you want to have this feature/enhancement, but as we always struggle with the resources and have to squeeze more and more features in the firmware I don't see this happening.

Please search in the source code for `MSG_LOAD_FILAMENT` and the `MENU_ITEM_SUBMENU_P(_T(MSG_LOAD_FILAMENT)` which will point you to the function. But as i said we use the "same" code to save resources for the menu and other.

Also please keep in mind to use the latest release stable github tag as base to compile your firmware.

Pitkox, 2023-03-16:
Yeah i know, I need to do it every time you release new FW, so sad :(
But i can see that my request is not very popular and let's say "hot"...
Please consider it for future releases, Just turn off stepper motors after load filament when printer is idle, That will be great.
With all respect to your job, thanks for help and for more hope will wait ;)

closed 2023-03-16 by Pitkox