prusa3d/Prusa-Firmware #4089: [BUG] SD card files are not correctly sorted based on modification time

julie777, 2023-03-16:
**Printer type** - [e.g. MK3S+
**Printer firmware version** -  3.12.0.5672

**SD card 

**Describe the bug**
When selecting print from SD card with the sort order set to time I expect the most recently modified file to be at the top of the list. This is the file that I am most likely to want to print.

However, it seems that the list is sorted be creation time instead of modification time. Creation time is the time when I create the file on the SD card. Modification time is the time when I actually changed the file. You can see that in the following table all the files were copied to the SD card in a batch. The creation times are almost identical. However, the modification times were preserved to show when the files were actually modified.

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=en-US style='font-family:Calibri;font-size:12.0pt'>

<div style='direction:ltr'>

Order shown | File | Access time | Modified time | Creation time
-- | -- | -- | -- | --
3 | minibin_0.3mm_PLA_MK3S_26m.gcode | Mar 15 00:00:00   2023 | Mar 15 17:40:30   2023 | Mar 15 17:41:15   2023
4 | New_PTFE_aligner_arm_0.2mm_PLA_MK3S_2h2m.gcode | Mar 15 00:00:00   2023 | Mar 11 14:25:52   2023 | Mar 15 17:41:15   2023
2 | New_PTFE_aligner_arm_0.2mm_PLA_MK3S_53m.gcode | Mar 15 00:00:00   2023 | Mar 11 12:50:16   2023 | Mar 15 17:41:17   2023
1 | Slider_Guide_0.2mm_PLA_MK3S_2h2m.gcode | Mar 15 00:00:00   2023 | Mar 11 14:27:10   2023 | Mar 15 17:41:19   2023

</div>

</body>

</html>
  
**To Reproduce**
  create a few files and then use touch to randomly change the modification times
  copy all the files as a group to the SD card.
  
**Expected behavior**
  Files on the SD card should be show with the most recently modified file at the top of the list
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

DRracer, 2023-03-16:
@julie777 I'm afraid this is a feature, not a bug. The firmware checks both creation and modification time and picks the newer one of the two for comparison. This was we cover most of the user-scenarios. We are aware, that there are edge cases, just like yours, an ideal solution seems not to exist as different operating systems works differently with the FAT32 time stamps.
Moreover, there is one inherent limitation to FAT filesystem in general - timestamp resolution is 3 seconds. If you manage to write multiple files at once within those 3s, they become time-sorted arbitrarily.

julie777, 2023-03-19:
Thanks for the info. I use a script to sync my sdcard to a directory in before I run the card into the room with the printer. I'll make some changes to see if I can set the creation times to match the original files.

closed 2023-03-19 by julie777