prusa3d/Prusa-Firmware #4092: [BUG] M104 S<Temp> and M109 S<Temp> not working from SD-Card

Dephrilibrium, 2023-03-18:
**Printer type** - MK3S
**Printer firmware version** -  3.13.0-BETA1 (customized with additional preheat-profiles)

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
Normally SD-Card. For now testwise also with octoprint (see bug description)

**Describe the bug**
I want to print a Temp-Tower for that I use this Test-Project with a small cylinder.
[Form-Cylinder Test SD and Octo.zip](https://github.com/prusa3d/Prusa-Firmware/files/11008705/Form-Cylinder.Test.SD.and.Octo.zip)

For the temperature-change I use a layer-change script (already added as custom layer-change-GCode-Script in the project):
```
;BEFORE_LAYER_CHANGE
G92 E0.0 

; ---------- Explanation of placeholders ----------
; layer_num       as #
; layer_z         z-heigh in mm
; layer_height    layer-height in mm (e.g. 0.15@0.4Nozzle)

; ---------- Explanation of the line-order ----------
; Temperatures on same order as on the tower
; Top-Temperature is first line
; Bottom-Temperature is last but one line
; First layer on plate (socket) is the last line!

; ---------- Temp-Changer-Lines ----------
; Top-Line is last tower-temp
; Last but no least line is first tower-temp
; Bottom-Line is first-layer
; Z-Value:   [layer_z]
; Layer-Num: [layer_num]
{if    layer_num > 9} M104 S230 M109 S230 ; Top Tower-Temp
{elsif layer_num > 8} M104 S225 M109 S225
{elsif layer_num > 7} M104 S220 M109 S220
{elsif layer_num > 6} M104 S215 M109 S215
{elsif layer_num > 5} M104 S210 M109 S210
{elsif layer_num > 4} M104 S205 M109 S205
{elsif layer_num > 3} M104 S200 M109 S200
{elsif layer_num > 2} M104 S195 M109 S195
{elsif layer_num > 1} M104 S190 M109 S190 ; Bottom Tower-Temp
{else               } M104 S190 M109 S190 ; First Layer & Socket Temp
{endif}
```
Then I exported the GCode to the SD-Card, made a copy of the GCode and uploaded it to OctoPrint. Then:
1.) I printed the SD-Card cylinder first - No Temperature-Change were made during the print.
2.) Directly afterwards I started the print via OctoPrint - Like a miracle the temperature changes its value when the layer changes.

Here you have the temp-graph from octoprint for both (the spike between both prints appeared as I used "Preheat -> PLA" and "Preheat -> Cooldown" due to the printer tried to force me to wait 7 minutes, but why? 🤔). I could bypass this by disconnecting an reconnecting OctoPrint and so I was able to load the job without waiting):
[image]

**To Reproduce**
1.) Download sample-project (error-description)
2.) Slice
3.) Export GCode to SD-Card
4.) Upload same GCode to OctoPrint
5.) Run SD-Card-GCode and OctoPrint one after one

  
**Expected behavior**
SD-Card GCodes should also change the temperature when "hitting" M104 or M109
  
**G-code**
Here you have only the generated GCode file as zip:
[Form-Cylinder Test SD and Octo_0.15mm_PLA_MK3S_4m.zip](https://github.com/prusa3d/Prusa-Firmware/files/11008711/Form-Cylinder.Test.SD.and.Octo_0.15mm_PLA_MK3S_4m.zip)

**Video**
see screenshot in error-description

3d-gussner, 2023-03-18:
@Dephrilibrium It is not a bug in the Firmware but in the gcode you created.
Please check the terminal output during SD print and you will see that the generated lines you have are `unknown` . The printer accepts only ONE gcode command per line! Octoprint is smarter and probably "splits" multiple gcodes in one line into multiple, so the printer "understands" these.
  
`echo:Unknown command: " M104 S220 M109 S220"(2)`

See https://help.prusa3d.com/article/macros_1775

[Form-Cylinder Test SD and Octo_0.15mm_PLA_MK3S_4m_one_gcode_per_line.zip](https://github.com/prusa3d/Prusa-Firmware/files/11009392/Form-Cylinder.Test.SD.and.Octo_0.15mm_PLA_MK3S_4m_one_gcode_per_line.zip)

Dephrilibrium, 2023-03-18:
@3d-gussner
You're right, but partly. I also tested around a little bit and and saw that the problem is the leading `space` in front of M104 which is put in front of the M104 command when GCode Script is converted into executable GCode. When I open the outcoming GCode-file and search for the layer-change code-line line I find:
```
; Bottom-Line is first-layer
; Z-Value:   0.15
; Layer-Num: 0
 M104 S190 M109 S190 ; First Layer & Socket Temp

M73 P2 R4
M73 Q2 S4
G1 E-.8 F2100
```

So when I remove the `space` between `} M104` so that:
`{if    layer_num > 9} M104 S230 M109 S230`
changes into:
`{if    layer_num > 9}M104 S230 M109 S230`
M104 S<Temp> works also on the SD-Card. M109 is still ignored, so that my target-temperature is set but the printer goes on with the next layer and won't wait for reaching the set temperature.

So with the final custom GCode:
```
{if    layer_z > 81.0}M104 S230 M109 S230 ; Top Tower-Temp
{elsif layer_z > 71.0}M104 S225 M109 S225
{elsif layer_z > 61.0}M104 S220 M109 S220
{elsif layer_z > 51.0}M104 S215 M109 S215
{elsif layer_z > 41.0}M104 S210 M109 S210
{elsif layer_z > 31.0}M104 S205 M109 S205
{elsif layer_z > 21.0}M104 S200 M109 S200
{elsif layer_z > 11.0}M104 S195 M109 S195
{elsif layer_z >  1.0}M104 S190 M109 S190 ; Bottom Tower-Temp
{else                }M104 S190 M109 S190 ; First Layer & Socket Temp
{endif}
```

I end with a working solution.

Btw. octoprint does then the same. Because in my case, my MK3S changed the temperatures but moved along with the next layer without waiting for the target-temperature.

For me it seems, that the Prusa-Firmware is not removing leading an tailing spaces/tabs when it parses the code.

But anyway. I found a working solution for me, so I think this issue can be closed then or do you want to add the point with the leading/tailing spaces/tabs?

closed 2023-03-18 by Dephrilibrium