prusa3d/Prusa-Firmware #4095: Constant beep when  Mk3S/+ idle under 15°C with 3.12.2 firmware

danchar4, 2023-03-19:
**Printer type** 
Mk3S/Mk3S+ - original hot-end with silicone sock

**Printer firmware version**
- 3.12.2

**SD card or USB/Octoprint**
- SDCard

**Describe the bug**
The printer is in an unheated garage and temperatures drop below 10°C overnight. If the printer is idle when the temperature drops, then a constant beep is heard with the 3.12.2 firmware. The only way to resolve it is to use a heat gun to warm up the hotend to 15°C and reset the printer.
  
**To Reproduce**
- Upgrade firmware from 3.11.x to 3.12.2
- Run thermal model
- Place printer in environment that is below ~15°C
- Allow printer to idle and wait for the beep

-  does NOT repro on Mk2.5/2.5S with 3.12.2
-  does NOT repro with 3.11.x firmware
  
**Expected behavior**
 - Mintemp error shown on screen _without_ audible alarm

**Workarounds**
- disable thermal model OR
- downgrade to 3.11.x firmware OR
- warm up hotend to 15°C+

**Related issues**
Likely the same root cause as:  #3987 but my issue is with factory hotend and does not need any g-code to reproduce so is the more general case hence filing a separate issue.

danchar4, 2023-03-19:
Oops I missed this issue which is probably the same but on v3.12.0 firmware.  #3898

Prusa-Support, 2023-04-29:
You are right, this is a duplicate. Thanks for reporting anyway.
Our developers are on it.

This issue will be closed as a duplicate.

Michele Moramarco
Prusa Research

closed 2023-04-29 by Prusa-Support