prusa3d/Prusa-Firmware #4099: I need to (firmware) change 9 points of Bed Level to 4

hbougard, 2023-03-21:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
Hello everybody!
total newbie here, I'm trying to revive and learn from an old Prusa MK2 

I got a little frustration because the heatbead is damage, in the middle right (around point #6) its crashed)
I already tried to create Gcode and/or Initial Gcode so it's skips the auto level... and tried to Skipp the dagame point or center points but it keeps Overriding to 9 points...
  
So, i was wondering.
Is there a way to change From Firmware so the machine is order to bed level calibrate 4 points on corners instad of the 9 points in the 3x3 bed matrix??

I hope someone can save my life! its so hard to find heatbeds in my country :(

3d-gussner, 2023-03-22:
@hbougard Sorry to hear that your MK42 bed is damaged and you have issues to run the MBL. 
As MK2 isn't supported anymore please have a look around https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/mesh_bed_calibration.cpp and you gonna have to compile the firmware your self.

Issue is that the MK42 bed has these calibration points at defined locations and you can't "move" them
[image]

You can also "Ignore" the MBL gcode `G80` in PrusaSlicer profile Start Gcode section
[image]

Prusa-Support, 2023-04-29:
Yet an alternative could be attaching a spring steel sheet with strong paper clips along the front and rear edge - **not recommended** though.
At that point, the whole surface will be metallic, and you can possibly customize the [G80 parameters](https://help.prusa3d.com/article/prusa-specific-g-codes_112173) - again **not recommended**.

Can this issue be closed please?

Michele Moramarco
Prusa Research

closed 2023-07-28 by DRracer