prusa3d/Prusa-Firmware #4103: [BUG] Checksum Error during Octopi Print

buliwyf42, 2023-03-22:
Printer type - [MK3S]
Printer firmware version - [3.12.2]

SD card or USB/Octoprint: Octoprint

A Checksum error is reported and it seems the printer requests a line again. 
After several messages like that, the print is canceled from octoprint. 
I printed the same gcode file several times before. 
  

Just print. 
  
Print without error
  
[Hase_3_3_8mm_0.3mm_PLA_MK3S_4h57m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/11039085/Hase_3_3_8mm_0.3mm_PLA_MK3S_4h57m.gcode.zip)

Here is the log from the Terminal:

`
Recv: ok
Send: N8 M73 P100*94
Recv: NORMAL MODE: Percent done: 100; print time remaining in mins: -1; Change in mins: -1
Recv: SILENT MODE: Percent done: -1; print time remaining in mins: -1; Change in mins: -1
Recv: ok
Send: N9 M117 192.168.211.31*22
Recv: LCD status changed
Recv: ok
Send: N10 G4*28
Recv: ok
Send: N11 M221 S100*80
Recv: ok
Send: N12 M900 K0*66
Recv: echo:Advance K=0.00
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N13 {if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538*110
Recv: Error:checksum mismatch, Last Line: 12
Recv: Resend: 13
Printer keeps requesting line 13 again and again, communication stuck
Changing monitoring state from "Cancelling" to "Error"
Send: M112
Send: N14 M112*20
Send: N15 M104 T0 S0*21
Send: N16 M140 S0*82
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
Closing down send loop
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xec6732b0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: start
`

sarusani, 2023-03-23:
This normally means that there's a problem with the connection between your raspi and the printer. If you're using a USB cable, replace it and make sure it's fully plugged in. If you're using the RPI port, check that it's plugged in all the way and the soldering on all the pins is proper.

Prusa-Support, 2023-05-03:
Thanks for reaching out to us.

> This normally means that there's a problem with the connection between your raspi and the printer. If you're using a USB cable, replace it and make sure it's fully plugged in. If you're using the RPI port, check that it's plugged in all the way and the soldering on all the pins is proper.

These are actually good pieces of advice.
Was it helpful?

Besides this, is the issue actually related to the most recent firmware version?
Please try downgrading the firmware to see if the problem persists.
If it persists, please also give it a try with a factory reset of the printer's Einsy board, and try re-flashing Octoprint on the external board.

We can't even exclude a problem specific to the board (Raspberry?), or Octoprint software.
The vendor may be able to assist you with the board, and [Octoprint](https://octoprint.org/)'s community may have further recommended troubleshooting steps to share.
Please keep us posted.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-26:
Is this issue still relevant?

This issue may be closed as stale in a few weeks
We would like to hear back from you.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-15:
This issue will be closed as stale.

Please return to us if you can confirm a firmware bug after reviewing all the above.
We would also appreciate hearing back from you even in the case you identified and solved the problem.

Michele Moramarco
Prusa Research

closed 2023-06-15 by Prusa-Support