prusa3d/Prusa-Firmware #4109: Firmware crash AGAIN

Cmos2112, 2023-03-25:


bdoogan, 2023-03-25:
I am having the exact same issue!! Mk3s with mmu2s, was working flawless ( for the most part) and kept getting the annoying ,message so I figured I would update, now i am getting  TM error message ( fyi- i am running the revo6). I started going through the calibration process and then it errors out on the TM setup. SO frustrating!! This should have been working out prior to being released.

Cmos2112, 2023-03-25:
The TM (thermal model)is known to be incompatible with Revo6. But there is
a fix for that. A simple 2 lines of g-code. Write it in notepad or send it
through Pronterface.

bdoogan, 2023-03-25:
Hi, thanks for the info. Can you let me know where to find that info?

Cmos2112, 2023-03-25:
I can check on Monday at work where the Prusa is. But I'm pretty sure I
found it by searching the exact error that it spit out. And I found the
right code pretty fast.

I may not be as completely useless as I felt in the op but if you never
made a text g-code file it is so simple even I can do it.
In notepad or similar low tech text editor type your g-code, for example

M92 E724 ; Set extruder E-steps
M500 ; Save changes to eeprom

Then click "save as" and select the file type as "all files" or "any file".
Whichever option it gives you. And name this text file something you will
recognize.
For example "Set_esteps.g-code"
If you did it right you can save it to an SD card right from notepad, then
put it in your printer and select "print from sd"
Which is really just a generic label for "run program".
This is a great and easy way to make adjustments to your printer.
Alternatively, you can connect to your printer directly from your pc with
Pronterface and type the code directly into the text box at the bottom
right.
You don't need any comments if you do it this way but they don't hurt
either. Comments in any code are just notes to help humans figure out what
the code is doing.

Maybe you already know all that but maybe someone else will read through my
response and a keystone will fall in place on their mental bridge.

Cmos2112, 2023-05-05:
> Thanks for your message and I'm sorry for the whole situation.
>
> While you can count on our full support in the official Customer Support
> <https://help.prusa3d.com/article/customer-support_2287> channels, and
> freely carry discussions in our official Forum
> <https://forum.prusa3d.com/>, GitHub is the place where we keep track of
> bugs and feature requests along with the community.
>
> We appreciate you sharing your personal experience but this issue is quite
> generic or simply puzzling due to several undefined possible concerns about
> your printer. As a consequence, this issue could be closed as invalid.
> A proper troubleshooting session with our Customer Support can remove a
> few doubts and clarify the nature of the problem/s.
>
> On a side note, calibrating the filament loading distance may only make
> sense if you use a custom-length tube on the MMU2S, or if you are running
> on an obsolete MMU2 model. Interestingly enough, virtually all MMU2 have
> possibly been upgraded to MMU2S for free (more in blog post
> <https://blog.prusa3d.com/original-prusa-i3-mk3s-and-mmu2s-release-sl1-and-powder-coated-sheets-update_29599/>).
> Not sure if this is an exception.
>
> In case of concerns about the Thermal Model protection, please check the
> already existing issues. These are supposedly being labelled with a thermal
> model
> tag. If you are a Revo user, due to its popularity, we are collecting data
> [... 17 lines cut ...]
>
> Would you mind closing this issue?
>
> Michele Moramarco
> Prusa Research

Yes sir,
You may close this issue.
Long story short. I saw new firmware updates that took the thermal model
into consideration. Refreshed both boards, pid tuned, calibrated everything
once more, and still had an issue, however, I believe I have it solved now.

With the addition of an led to my filament sensor at the tool head, I found
I was getting multiple false positives for filament in the extruder. Once I
remedied that situation there seem to be no more troubles so far.

Thank you and thank everyone at Prusa for the fantastic support.
I love you all.(in the kindest most non-threatening way possible. )

Have a wonderful weekend.

Prusa-Support, 2023-05-06:
Thank _you_ for supporting our company and the Open Source.

Thank you for your kind update too.

Michele Moramarco
Prusa Research

closed 2023-05-06 by Prusa-Support