prusa3d/Prusa-Firmware #4110: [BUG] Firmware crash & M112 error during MMU2S tool change

simmons83, 2023-03-25:
**Printer type** - MK3S
**Printer firmware version** -  3.12.1 & 3.12.2

**MMU upgrade** -  MMU2S
**MMU upgrade firmware version** -1.0.6

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
Firmware crash leading to M112 error usually occurring during a filament tool change. Since updating to 3.12.x

 
2023-03-25 08:43:51,321 - Recv: Unloading finished 2
2023-03-25 08:43:51,530 - Recv: T:205.0 /205.0 B:59.7 /60.0 T0:205.0 /205.0 @:43 B@:62 P:35.0 A:40.9
2023-03-25 08:43:51,780 - Recv: echo:busy: processing
2023-03-25 08:43:53,360 - Recv: mmu_get_response - begin move: load
2023-03-25 08:43:56,366 - Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2023-03-25 08:43:56,369 - Send: N316782 T1*2
2023-03-25 08:43:57,899 - Recv: start
2023-03-25 08:43:57,900 - Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
2023-03-25 08:43:57,907 - Changing monitoring state from "Printing" to "Cancelling"
2023-03-25 08:43:59,673 - Recv: echo: 3.12.1-5686
2023-03-25 08:43:59,674 - Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2023-03-25 08:43:59,674 - Should resend line -2 but no sufficient history is available, can't resend
2023-03-25 08:43:59,675 - Changing monitoring state from "Cancelling" to "Error"
2023-03-25 08:43:59,692 - Send: M112
2023-03-25 08:43:59,694 - Send: N0 M112*33
2023-03-25 08:43:59,695 - Send: N1 M104 T0 S0*32
2023-03-25 08:43:59,696 - Send: N2 M104 T1 S0*34
2023-03-25 08:43:59,697 - Send: N3 M104 T2 S0*32
2023-03-25 08:43:59,698 - Send: N4 M104 T3 S0*38
2023-03-25 08:43:59,699 - Send: N5 M104 T4 S0*32
2023-03-25 08:43:59,700 - Send: N6 M140 S0*99
2023-03-25 08:43:59,708 - Changing monitoring state from "Error" to "Offline after error"
2023-03-25 08:43:59,720 - Connection closed, closing down monitor

[M112errors.log](https://github.com/prusa3d/Prusa-Firmware/files/11070001/M112errors.log)

Full log file attached

FlyingPete, 2023-04-02:
I am also receiving this error since updating to 3.12.2-5713, filament load starts then crashes and stops with filament part way down PTFE tube which requires manual removal before use or MMU2 errors as jammed (all LEDs flash orange).

gysnst2, 2023-04-08:
All 3 of my printers are doing this also kinda pia

Rob70ar, 2023-04-24:
I do have similar behavior with my mk3s/Mmu2s… It looks that downgrading to 3.11 is the current workaround so far. Yesterday support told me to downgrade and upgrade again, but it did not solve the problem. Today they told me to disconnect raspberry- I have not tried it yet.

Elcar903, 2023-05-01:
I am having a similar same issue with my MK3s MMU2 as well with Firmware 3.12.1 (1.0.6) and Firmware 3.12.2 (1.0.6)
If I use the buttons on the MMU2 to move or load filament it works fine but if the MK3s tries to load filament it will crash the MMU2 with flashing of all 5 green and red lights.

I did all the troubleshoot guides and check the voltage and stuff coming in and out of the machine.
it all checks out.
I even did a factory reset on both the MK3s and MMU2 and this didn't help it either.

I flashed back to Firmware 3.11.0 (1.0.6) and it doesn't have this issue.

PhLacoude, 2023-05-02:
Exact same problem as you guys. I could no longer print with Octoprint after a printer firmware upgrade.

I was able to print small multicolored models via the SD card. I did not try large models but I did try all 5 filaments. They all worked for prints whose GCode was coming from the SD Card (with the Octoprint unplugged.)

My MMU2S has been erratic since the latest printer firmware upgrade. In the MMU2S Single mode, the printer had trouble picking the filament. In MMU2S mode, the MMU2S unit was cashing regularly. It would flash 5 red+green lights. Manual loads of the filament via the LCD screen menu or via the MMU2S buttons were difficult or impossible. I had my very first “firmware crash” errors. I have 2 printers and I have at least 5000 hours on each one. Yet, again, I never had a “firmware crash” message before.

My Octoprint logs look exactly like the O.P. logs. Same codes. And my Octoprint sends an M112 code after each crash to prevent thermal issues (as expected, I guess...)

So I proceeded in the following order.

Verified I had the latest MMU firmware (which was the case).
Factory reset the MMU
https://help.prusa3d.com/guide/service-menu-factory-reset_86232
Retested with and without the Octoprint units. (Mine are Raspberry Pi 3B+ units connected via a USB cable.)
As a last resort, I went back to the previous printer firmware. This also fixed it for me.
Thank you very much for raising the issue and for suggesting reverting to firmware 3.11.

Elcar903, 2023-05-02:
So I have been messing with it for the last 2 days and I was able to get Firmware 3.12.2 (1.0.6) to work on my MK3 / MMU2
I am not sure why this fixed it or if will work with anyone else but I will post what I did to get it to work for me 
as well as some Pronterface logs that I got with this processes.

I will not claim that every step is necessary, I am just posting what I did to get it to work

I did try the factory reset option with Firmware 3.12.2 (1.0.6) and with the MMU2
before but it didn't fix it but I didn't try this sequence until today.

First I flashed Firmware 3.11.0 (1.0.6) and after it was done I ran the factory reset on the MK3s
Here is the guide to do this with.
[https://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216](https://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216)

I selected the "All" option on the reset.
When it finished and started back up it will ask to run the calibration guide , So I let it run and complete that.

After it complete the First run Calibration Guide I pressed the "X" button to restart the printer.
Then after it booted back up completely I then ran the factory reset on the MMU2 Unit.

You can find the guide here for that.
[https://help.prusa3d.com/article/factory-reset-mmu2s_1920](https://help.prusa3d.com/article/factory-reset-mmu2s_1920)

After the MMU2 finished the factory reset I then connected the MMU2 to my PC VIA USB cable
and as before I press the "X" button on the MK3s to restart it as well as the MMU2, after it completely restarted 
I  flashed the MMU2 firmware to 1.0.6 ( that comes with the Firmware 3.12.2 (1.0.6) zip)

After the flash complete I disconnected my PC from the MMU2 and then powered the printer off completely for about 2min
Then I turned the MK3s back on and after it booted completely I connected my PC to my MK3s VIA USB and then
flashed the MK3s to Firmware 3.12.2 (1.0.6)

When this finished I did another restart on the MK3s with the "X" button and after it booted back up 
I performed another Factory reset on the MK3s ( see previous guide) 
I again selected the "All" option.

I then proceeded to run the first time run guide and after it completed I ran a PID calibration 
( more of a personal option that I do after I do a firmware update)

When all that was done I turned off the printer again and waited a few minutes to turn it back on.

After that I started running test on the print like loading and unloading the filament from the MMU2 with an option on the MK3s printer to see if it would crash it like it did before, It seem to be working just fine.

So far it hasn't crashed and I have been able to make 3 prints with the MMU2 without issue so far.
I printed a MMU test cube and a Benchy and both came out looking good without issue.

I am about to start a 16 hour print with 5 filaments and around 1,100 filament changes , so this should be a good test for it.

I don't know what fixed it ( or if it is fixed completely  ) but these are the steps I did on my MK3s.

Maybe someone who knows more than me about this stuff can figure it out for everyone else, 
I know it seems to be a small number of people having this bug, but still its something I think need to be looked at.

[flash_data_dump.zip](https://github.com/prusa3d/Prusa-Firmware/files/11376517/flash_data_dump.zip)

Rob70ar, 2023-05-04:
I disconnected the raspberry pi zero W and the FW crashes during tool change persist. I was able to reproduce the error trying to change the nozzle temperature from the menu during tool change.. tried a couple of times and voala! FW crash error.

Prusa-Support, 2023-05-06:
Thank you for reaching out to us.

This seems to be related to a combination of
_FW 3.12.x + MMU + filament un/loading operations_
but we may not have enough evidence to confirm a bug.

We should possibly find a pattern to controllably reproduce the issue and confirm the bug.
The problem may seem to be seldom and apparently fixable in a few cases. It may sound like a temporary glitch.

I was under the impression that some users could track it back to an unreliable filament sensor signal, a failing SD card, and Octoprint connection problems. Sometimes a combined factory reset does the trick instead.
These possible causes and solutions are not confirmed either but the combination "_FW 3.12.x + MMU + filament un/loading operations_" is a constant that we can't ignore.

**Recommended actions.**
1. Downgrade and upgrade the firmware a few times to make sure the problem is only reproducible with FW 3.12.x.
2. Try a "combined" and complete factory reset ([Einsy ](https://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216)+ [MM-controller](https://help.prusa3d.com/article/factory-reset-mmu2s_1920)).
3. For debugging purposes, please also double-check the filament sensor signal, try other SD cards, and disconnect third-party parts (external boards e.g. Raspberry, cameras, UPS backup battery, LED stripes, other brand components e.g. motors, etc...)

**If the issue will persist**, possibly focus on the essential steps for reproducibility, take notes, and download the dump file.

1. Download the dump file (keep it on your computer, it may be useful later). The dump file (and warning message) will disappear after downloading. You can download it as per this article. https://help.prusa3d.com/article/crash-dump_364959
2. Possibly save a text file together with it and add relevant notes on it: FW version, date/time, how/when the FW crash occurred (last operation), personal notes...
3. If you can reproducible the problem, or it accidentally occurs multiple times, you will have multiple dump files and details for our developers to analyze. You can upload your files here.

Michele Moramarco
Prusa Research

simmons83, 2023-05-06:
> Thank you for reaching out to us.
>
> This seems to be related to a combination of _FW 3.12.x + MMU + filament un/loading operations_ but we may not have enough evidence to confirm a bug.
>
> We should [...]

Hi Michele

Does the flash dump data survive if the firmware has been downgraded to 3.11? I have run a D21 command and have copied the info. Is it safe to post here?

Prusa-Support, 2023-05-31:
Feel free to post logs here in GitHub, or, if you don't feel comfortable with that, you can email them to my attention if needed in the future. This may not seem to be necessary anymore though.

This issue seems duplicated several times (https://github.com/prusa3d/Prusa-Firmware/issues/3989, https://github.com/prusa3d/Prusa-Firmware/issues/3990, https://github.com/prusa3d/Prusa-Firmware/issues/4022, https://github.com/prusa3d/Prusa-Firmware/issues/4110, https://github.com/prusa3d/Prusa-Firmware/issues/4155, maybe more...) and that the cause if is known at this point - https://github.com/prusa3d/Prusa-Firmware/issues/3989#issuecomment-1427012689.

We are aware of the bug and it should no longer persist on the next firmware - https://github.com/prusa3d/Prusa-Firmware/issues/3989#issuecomment-1544300704

For now, if nothing seems to help, you can temporarily downgrade to [FW 3.11.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.11.0) (not ideal).
In many cases though, this glitch is fixable or traced back to other related problems - https://github.com/prusa3d/Prusa-Firmware/issues/4110#issuecomment-1537189239.

I hope you don't mind me closing this issue as a duplicate now.
Thanks for your collaboration and understanding.

Michele Moramarco
Prusa Research

closed 2023-05-31 by Prusa-Support