prusa3d/Prusa-Firmware #4115: [ENHANCEMENT] Do not move printhead up when loading filament is triggered by sensor

sarusani, 2023-03-27:
**Printer type** - MK3S

Currently the printhead is moved up when you insert filament and the filament sensor is active. I get that this can be an advantage if the menu "Load Filament" is used. But when you insert it by hand and it's triggered by the filament sensor this doesn't bring any advantage; The filament is already in the printhead, better clearance is not needed anymore.

The printhead moving on it's own while someones hand is touching it, scared anyone I introduced to this printer so far. They don't like that parts are moving without warning while their hands are in "harms way".

I deactivated it myself and compiled my own firmware, but I think it's probably worth discussing if this should happen or if it could only be triggered when "Load Filament" is used via the LCD menu.

Prusa-Support, 2023-05-10:
Thanks for your feedback.

As explained in issue https://github.com/prusa3d/Prusa-Firmware/issues/3532, there is a parameter in the firmware that defines the Z movements anytime this movement is required (for example for a filament change) and it will be possible to modify it in the future firmware version.

When you insert filament, you are probably about to load it, and thanks to the automatic movement you won't be bothered by rising the Z axis manually to clean the nozzle after purging the new filament. This movement is most likely required when you insert filament.

So, are you suggesting avoiding such movement in specific scenarios?
Maybe postponing it if the user is in the middle of specific interventions around the extruder?
Could you give us examples, please?

Michele Moramarco
Prusa Research

sarusani, 2023-07-24:
Sorry it took so long to respond, totally forgot about it :)

> As explained in issue #3532, there is a parameter in the firmware that defines the Z movements anytime this movement is required (for example for a filament change) and it will be possible to modify it in the future firmware version.

That's great, currently I modify MIN_Z_FOR_LOAD & MIN_Z_FOR_UNLOAD in Configuration.h to disable it. Would be easier to use some gcode instead 👍  :)

> When you insert filament, you are probably about to load it, and thanks to the automatic movement you won't be bothered by rising the Z axis manually to clean the nozzle after purging the new filament. This movement is most likely required when you insert filament.

First, I'm using a enclosure which has a ceiling about 5cm taller than the MK3S.
Examples:
Loading:
Normally I have to load filament after starting up the printer, since the printer isn't homed at that moment it doesn't know it's z position, if the last print was tall, it will crash the z axis into the top and pinch my hand between the extruder and the ceiling. (not hard, it can't insure anyone but it still scares new users).

I rarely encounter a case where the automatic raising would be beneficial to me. Even if you cancel a print on the first layer, the z-axis ends up about 4cm above the print bed. More than high enough to remove extruded filament. The same is for homing.

Unloading:
I normally unload filament after a print, so the z axis is already high enough. No raising needed.

> So, are you suggesting avoiding such movement in specific scenarios? Maybe postponing it if the user is in the middle of specific interventions around the extruder? Could you give us examples, please?

Personally I don't see the point of raising it automatically at all, for the few occasions it's needed, the long press on the dial & moving z-axis manually is fast enough for me. (That's my way of saying, please make configuring 0mm a option when you implement the gcode param :) )

**But in general I think it would be a good solution to wait with the raising movement after the filament is loaded and execute it just before the filament extrusion starts. That way the users hand would already be away from the print head when movement starts.**

Cheers!

closed 2024-07-11 by 3d-gussner