prusa3d/Prusa-Firmware #4120: [ENHANCEMENT]<Color Change M600>

KoProGnosis, 2023-03-31:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3S]

I have a print farm of 20 printers and I print an object that changes color manually from Jet Black to Prusa Orange. The default amount of filament extruded after the color change is not enough, so the filament is dark orange. 
  
It would be nice if there was an option in the slicer to control the amount of the extruded filament after the color change.

3d-gussner, 2023-03-31:
@KoProGnosis Thanks for your request.

During the `M600`  process you will be asked
- Press the knob to uload filament
  - which turns the extruder to pull it out of the hotend 
- to remove the the filament manually
- asked if it has been removed correctly
- to insert the filament
  - it turns the extruder to pull it in
- Get asked if filament changed correctly
  - Yes --> it will continue with printing
  - Filament not loaded --> will try to load again for a defined (100mm)
    - this is one option to flush more filament
  - Color not correct --> will load again filament (~35mm)
    - this is another option to flush more filament

Changing the `M600` to add extra "purge" value would need changes in the printer firmware and PrusaSlicer, and I doubt that will happen.

Another workaround would be that you use `Add custom G-code` in Prusa Slicer instead of `Add color change M600`
```
M600 ; change filament/color
G1 Exxx ; extrude extra xxx mm of filament
```

I hope that helps a bit.

Prusa-Support, 2023-05-12:
Thank you again for your suggestion, but, as explained above, there are enough options suitable for your purpose.
This said changes in the current printer behavior wouldn't really be legitimate.

Can this issue be closed?

Michele Moramarco
Prusa Research

KoProGnosis, 2023-05-12:
@3d-gussner Thank your for your help! 
@Prusa-Support Yes it can be closed.