prusa3d/Prusa-Firmware #4121: [BUG]MK3s+ will not complete z calibration.

wisonlau, 2023-04-01:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.2

**MMU upgrade** - no
**MMU upgrade firmware version** - not applicable

**SD card or USB/Octoprint**
SD card

**Describe the bug**
The Z calibration wizard stops when the nozzle drops to the bed. "Calibration failed! Check the axes and run again." I have done the temp model calibration.
  
**To Reproduce**
I select z calibration on the LCD.
  
**Expected behavior**
I expected the nozzle to drop to the bed and begin its calibration process.

**Video**
> http://43.132.168.109/93_1680357690.mp4

**Same question**
https://github.com/prusa3d/Prusa-Firmware/issues/4039#issuecomment-1492965895

michael-borck, 2023-04-15:
Video suggest some z-binding towards the top.   Try removing the bearings for the z-axis lead screws and see if you still have the problem.

wisonlau, 2023-04-18:
Thank you, it's this issue that caused it.

closed 2023-04-18 by wisonlau