prusa3d/Prusa-Firmware #4124: [ENHANCEMENT] Add "unload filmament" option if there is a mis-load during a manual color change.

thew1se, 2023-04-03:
MK3s+ no MMU

Description of Problem: 
If there is a misload during a manual color change, the prompt menu does not give an option to "unload and try again" if needed.   

In my color change problem experienced today, repeatedly trying to feed for filament forward did not work resulting in the filament slipping in the extruder gears.
  
**Describe the solution you'd like**
Add a fourth menu selection option to "No - unload filament" and then return to step 1 of loading filament.

Mbrinch, 2023-04-04:
Had the same problem yesterday and is an option I would love to see as well.

sarusani, 2023-04-08:
Duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/2415

Prusa-Support, 2023-05-12:
Thanks for your suggestion.
This issue can be closed as a duplicate of #2415 indeed.

Michele Moramarco
Prusa Research

closed 2023-05-12 by Prusa-Support