prusa3d/Prusa-Firmware #4125: [BUG] "PAUSED THERMAL ERROR" reappear in the new print after a previous aborted thermal protection print

SNDR0, 2023-04-04:
**Printer type** - original MK3S 
**Printer firmware version** -  3.12.2

**MMU upgrade** - [ MMU2S ]
**MMU upgrade firmware version** - [1.0.6]

**SD card or USB/Octoprint**
  Octoprint

**Describe the bug**
  during a print with a not perfect adesion on hot bed, the new feature noticed that the socket was ripped of and paused the printer fortunally... a 8 hours print and a lot of filament saved ... stop print via printer, no reset, no power off, preparing the same print with different sheet and restart from beginning via octoprint and prusa slicer. the printer displayed usually messages "Heating", Heating
Bed", "Heating done". After that and until the end of the print it appears "PAUSED THERMAL ERROR" 
**To Reproduce**
  thermal protection issue, abort print, reastart a new print.
  
**Expected behavior**
  Something goes to 1 but at the end doesn't return to 0
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
[image]
[image]

lowprize, 2023-04-20:
Same here, [PID calibration](https://help.prusa3d.com/article/pid-tuning_2265) not success too. 
Steps before Thermal error

- Firmware updated over Octoprint Octoprint, that failed, after that cannont change language from ENG to CZE
- Firmware again updated to 3.2.12 over USB and Prusaslicer - thermal error
- Firmware again donwgraded to 3.2.11 - same thermal error

Helpdesk: try update FW without connected all cables to motherboard except LCD and power.
Going to do that.

arekm, 2023-04-20:
(Note: Octoprint firmware upgrade plugin is not capable of properly upgrading languages part of firmware. https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/309)

lowprize, 2023-04-20:
Aha I think firmware was badly flashed. So now I can rule out that it is flashed incorrectly. Thank you for link

arekm, 2023-04-20:
And one more note. I always upgrade via octoprint plugin (because I don't need non English) and main firmware part is upgraded correctly :)

SNDR0, 2023-04-20:
Your problems are not referred to this bug...

lowprize, 2023-04-20:
I was write about Language problem as step before I got Thermal error for better reproduce. Now I know language it is not bug but feature :)

Thermal error is there still.
[image]
[https://photos.app.goo.gl/Pwe8PnTFHp8Yvq5g6](https://photos.app.goo.gl/Pwe8PnTFHp8Yvq5g6)

Prusa-Support, 2023-05-13:
Thanks for reporting.

Are you able to reproduce the error without Octoprint?
Of course, you don't need to sabotage a print but put a metal cold object onto the right side of the heater-block (away from the wires), or very carefully remove the silicon socket with pliers if you are using one (be very careful not to bend the hotend wires during the process).

Did you notice any other _thermal anomaly_ report at all, during other (non-failing) prints?

Michele Moramarco
Prusa Research

SNDR0, 2023-05-14:
tested with a sd file. same problem. 

stopped print after termal error. i noticed another issue: if you stop the print the filament still loaded and during purge in the new print it badly scratches..

first print after termal error
https://photos.app.goo.gl/FHqe87RHNKa4SWbW7

second print after previous stopped and not completed.
https://photos.app.goo.gl/KWy8eAfZVPUx1Awh8

JakubikD, 2023-05-17:
Same issue here. Btw, i jusst got my printer from prusa service ( We could say heis  like brand new printer ). After few successful prints, it has started showing PAUSED THERMAL ERROR or THERMAL ANOMALY during calibration or after few seconds of printing.

admin4life, 2023-05-24:
I started having the same issue today. Recently flashed to new firmware from prusaslicer via USB cable. It errors THERMAL ANOMALY but the temp graphs on octoprint don't show anything odd and if you set it to a temp it seems to hold stable.

SNDR0, 2023-05-24:
@admin4life your problem isn't what I mentioned in this issue.. do a thermal model calibration if you haven't done yet after the last flashing. After that if the printer continues to display that warning probably is better to do a double check to cables, connectors, wires, heater cartridge and hotend thermistor.

Prusa-Support, 2023-06-13:
I'm more concerned by the appearance of Thermal Anomaly than the persistence of the warning but I'm not sure this behavior is intentional.

Thanks for reporting. We will investigate this.

Michele Moramarco
Prusa Research

admin4life, 2023-06-13:
I ended up having to just do the Gcodes to ignore the error. I don't have anything "custom" on my printer other than aftermarket thermistor and heat tube. Printer works great otherwise

Prusa-Support, 2023-06-15:
We'd like to make sure about the testing scenario.
As anyone tried having Octorpint and third-party boards (Raspberry) totally disconnected?

Michele Moramarco
Prusa Research

JakubikD, 2023-06-16:
In the end I changed thermistor and I dont even know about this problem anymore

SNDR0, 2023-06-16:
@Prusa-Support 
In my previous reply I attached a video where you can see that I removed RPI zero, disabled rpi port and the issue stay there...  Before to begin I power off the printer, removed the rpi, power on, disabled the RPI port and forced the thermal issue untill the pause...

tideline3d, 2023-07-16:
FWIW I have 24 printers, all upgraded through the OctoPrint plugin and I ended up here because one of them refuses to print with an error like this.   It is in a paused state rather than canceled.   The hotend target was turned down to 0 but the bed target stayed at 80.    

I've had thermistors fail with this new firmware and it usually cancels the print similar to a M112 style cancel.   Not sure why you'd want to pause a print if there's a thermal anomaly, this seems like unexpected behavior.  

I'll reflash, re-run the thermal model calibration and try again.  If that doesn't work I'll throw an SD card in and try to get a reproduction sans RPi/OctoPrint

Prusa-Support, 2023-10-14:
@SNDR0 Interesting.
I'll make sure to share your result with the developers.

Michele Moramarco
Prusa Research

shutosg, 2023-12-30:
I have been having the same problem for a while now.

During printing, THERMAL ANNORMALY occurs frequently, causing the print to pause.
The frequent pauses make it difficult to continue printing, so PID Calibration and Thermal Model Calibration were performed, which were completed successfully, but the problem persists.
When I start printing, THERMAL ANNORMALY occurs during or shortly after the meshbed leveling calibration.

I assume this is the same symptom as this video uploaded by SNDR0.
https://photos.app.goo.gl/FHqe87RHNKa4SWbW7

Printer type - original MK3S
Printer firmware version - 3.12.2 & 3.13.2
NOTE: I forget what firmware I had when the problem occurred, but it was probably 3.13.1 or 3.13.0. After that, the calibration described above did not improve the problem, so I upgraded the firmware to 3.13.2, which did not improve the problem, and then reverted to 3.12.2, which did not improve the problem.

MMU upgrade - none
MMU upgrade firmware version - none

SD card or USB/Octoprint
SD card

shutosg, 2023-12-30:
Things seem to have gotten worse.

When I reverted the firmware back to 3.11.0 and tried printing again, the firmware restarted with sparks coming from the left side of the extruder (opposite side of the Super PINDA).

Then, when I run self test, the firmware reboots right after the heat bed test is completed (probably as soon as the hot end test starts).
Similarly, if I run preheat, it reboots as soon as preheat starts. 

This occurs even after updating the firmware to 3.13.2 again, and even after performing a factory reset (service prep).
[Video of the self-test](https://drive.google.com/file/d/1uQDftNGT8WzZRh7LskJ4af7zvxAYIGtp/view?usp=sharing)
[Video of the preheat](https://drive.google.com/file/d/1A-W3krMOOAF3wPFnKaa0NirC0asd7j4x/view?usp=sharing)

I am assuming that the hardware has been damaged, but if you can predict which hardware is the problem from a software perspective, could you please let me know?

SNDR0, 2023-12-30:
Replace the heater and you solve all the problems... Your cartridge is in
short circuit.

https://www.prusa3d.com/product/hotend-heater-cartridge-24v-40w-mk3-s/

shutosg, 2023-12-30:
@SNDR0 Thanks for the reply. I too had just read the forum and thought it was a hotend heater issue.

I removed the hotend heater cable from the Einsy as mentioned in the following thread and it no longer reboots. (Naturally, the temperature did not rise and a THERMAL ANNORMALY error was displayed.)
https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/hotend-or-bed-heating-causes-a- reboot/#post-143444

I will try replacing the hot end heater and see if the THERMAL ANNORMALY error continues to occur.
Thanks.

Prusa-Support, 2024-01-11:
In the first video, it seemed like the nozzle temperature never backed up to the target temperature so either the heater was out of specifics or the PID needs to be re-calibrated (followed by a new thermal model calibration).

However, in the video provided later, preheating was causing a reboot so the heater may actually be damaged or twisted, causing a short circuit indeed. In rare cases, cleaning and separating the heater's bare metal wires near the cartridge can solve the problem. A heater cartridge replacement would be the safest option though (recommended).

Our Customer Support can guide you through testing and evaluating the heater.
https://help.prusa3d.com/article/customer-support_2287

.

By the way, is this issue still reproducible on the most recent firmware 3.13.x?
Would the message remain on the screen on the second print, if and only if an external board is connected? 

Michele Moramarco
Prusa Research

SNDR0, 2024-01-11:
> In the first video, it seemed like the nozzle temperature never backed up to the target temperature so either the heater was out of specifics or the PID needs to be re-calibrated (followed by a new [...]

I just check if the last official firmare 3.13.2 solved the issue: something is changed, but not solved.. 
[image]

This is the video of the beginning of the new print After the issue recreated purposely.

 https://photos.app.goo.gl/zSCtpuXrk2rC6NhD7

I noticed another issue: when the error occurs the head move in park position but without retraction of the filament, so when it try to unload filament to mmu it fails. 

https://photos.app.goo.gl/cb7pZSKG4eJGYU5Y8

shutosg, 2024-01-21:
I would like to add to my post from 3 weeks ago
Since the issue has been resolved, there is no need to reply at all if it would interfere with the main discussion.

The rebooting problem [here](https://github.com/prusa3d/Prusa-Firmware/issues/4125#issuecomment-1872577268) was caused by a shorted heater as a result, and after replacing the heater, the rebooting problem was resolved.
However, even after replacing the heater and updating to the latest firmware, the problem of erratic THERMAL ANNORMALY and THERMAL RUNAWAY continued.
So I replaced the thermistor and the problem was solved perfectly.
As a result, it appears to have been a hardware issue.

Attached are photos of the heater and thermistor, which were probably damaged.

| damaged heater | damaged thermistor | damaged thermistor 2 |
|---|---|---|
|[image]|[image]|[image]|

3d-gussner, 2024-01-22:
@shutosg Thanks for the update and showing the real issue. This is one of the advantages of having the thermal model, it reports a potential issue with the hotend.

At which location have you found the broken isolation? Maybe this information can help others to prevent the issue.
- Can you please try to measure the distance from heater/thermistor to the breaks.
- Where in the cable loom are the breaks?

On the REVO we have seen that the connectors can cause issues in the loom.

3d-gussner, 2024-03-13:
Closing as no interaction

closed 2024-03-13 by 3d-gussner

peon374, 2024-07-31:
I have just upgraded to latest firmware and re-calibrated everything and now get thermal anomalies and pause thermal error. I can test there is nothing wrong with the hardware. Wouldn't mind knowing how to "resume" the pause.