prusa3d/Prusa-Firmware #4126: [BUG]MK3 PAT9125 sensor not recognized

emard, 2023-04-06:
on MK3 filament sensor PAT9125 is not longer
recognized as filament presence source however
its hardware readings are ok, seen in
support->sensor info

happened somewhere between v7089 and v7122

emard, 2023-04-06:
> is it autoload, runout or jam detection that isn't working?

printer "thinks" as no filament present, it constantly
asks to replace filament.

in Support->Sensor info there is only line
PINDA: 0/1
this line was at v7089, I tried v7022 and v7174
now this line is missing:
Fil.Sensor: 0/1

Next lines seem ok
B: 80   Xd: 0
S: 17   Yd: 20102

when I remove or insert filament they change correctly

leptun, 2023-04-06:
Oke. I'll check that issue out soon. Fixing now other worse bugs that were introduced in the latest merging spree (which you might have also noticed).

emard, 2023-04-06:
Yes :) I noticed knob press clicks but doesn't enter SD directory,
some obvious bug, should be fixed soon or already it is

leptun, 2023-04-06:
Didn't know about that one. Adding it to the list now :)

gudnimg, 2023-04-06:
> Yes :) I noticed knob press clicks but doesn't enter SD directory,
some obvious bug, should be fixed soon or already it is

It may be something is consuming the button event too soon.

To the SD directory work if you rotate the knob? If so that indicates a problem with `lcd_draw_update`. Should always be set if either clicked or rotated.

emard, 2023-04-06:
When I click 2nd time on the SD directory it enters it,
but then when I try to print file.gcode, no matter how
much I click knob, it won't start printing.

closed 2023-04-07 by leptun